Smoothieware/Smoothieware #871: Switch module not correcting setting PWM when M106 Sxxx is float

DouglasPearless, 2016-03-09:
I have come across a possible issue, the fan on my printer on accepts **M106 S255 or M106 S0** but any decimal value like **M106 S128.60** causes the fan to stop.
In reading **Switch.cpp** I noticed in **Switch::on_gcode_received** we have:
**int v = (gcode->get_int('S') * sigmadelta_pin->max_pwm()) / 255; // scale by max_pwm so input of 255 and max_pwm of 128 would set value to 128** which reading the way gcode.cpp implements get_int() it appears to return 0 unless an integer is parsed.  I believe the above line should be changed to use get_value('S') and convert the returned float to integer (simply truncate after the decimal point).

wolfmanjm, 2016-03-09:
sounds reasonable. try it and if it works submit a PR, thanks.

wolfmanjm, 2016-03-28:
actually I think I have fixed this.

closed 2016-03-28 by wolfmanjm