Smoothieware/Smoothieware #872: Printing slowly over Network drops the connection

dutchrazor, 2016-03-11:
Dear all,

We use Pronterface in TCP streaming mode to control Smoothie. I have found that when I print slowly, e.g. 200mm/min, lines of 20mm long, the connection drops during printing. It seems that if a Gcode takes longer than 1s to execute, this problem occurs.

Even though Smoothie has already received all Gcode at this point, it still stops printing shortly afterwards.

This only happens when printing in TCP streaming mode, without it and over USB the print works fine. I've attached a sample Gcode to illustrate the problem

Best regards,

Fabien Bruning

[0008_tnologo_demo_tracks.zip](https://github.com/Smoothieware/Smoothieware/files/168691/0008_tnologo_demo_tracks.zip)

After all gcode is sent, Pronterface says:

Print ended at: 15:55:51 and took 0:00:00
SENT: M105
RECV: ok
[ERROR] Can't read from printer (disconnected?) (Socket error 10054): An existing connection was forcibly closed by the remote host
SENT: M105
[ERROR] Can't write to printer (disconnected?) (Socket error 10054): An existing connection was forcibly closed by the remote host
SENT: M105
[ERROR] Can't write to printer (disconnected?) (Socket error 10054): An existing connection was forcibly closed by the remote host
SENT: M105
[ERROR] Can't write to printer (disconnected?) (Socket error 10054): An existing connection was forcibly closed by the remote host

wolfmanjm, 2016-03-11:
pronterfce has a socket timeout coded somewhere that will do this, you need to find it and set it to a long time. Smoothie does no ttimeout tcp connections.

closed 2016-03-11 by wolfmanjm

dutchrazor, 2016-03-15:
I set all socket timeouts in Pronterface to high, but it still occurs. 

I was wondering how the network interface is set up in Smoothie, does it stop responding at all during a Gcode move, and only update after a Gcode move is done?

triffid, 2016-03-15:
Smoothie says OK when the queue has a free slot. TCP connections  should
handle at least several minutes of silence without issue. I haven't dug
into smoothie's TCP implementation but it might set window size to zero
when the queue fills and reset it to a sensible value when there's space