Smoothieware/Smoothieware #873: Add support for a circular build plate

dcousens, 2016-03-15:
Just an issue,  so it can be more easily tracked.
Add the ability to use a circular build plate.

- [ ] Add constrained ZGrid feature
- [ ] Test?
- [ ] Add configuration example

I might be up to implement/test this over the next few months,  so I'll track what I need to do here also.

wolfmanjm, 2016-03-15:
circular build plates are supported in the delta grid strategy.
Please move this to the zgrid specific request issue.

closed 2016-03-15 by wolfmanjm

qharley, 2016-03-15:
Circular build beds are work in progress in zgrid at the moment, and not
ready yet.

[image: photo]
*Quentin Harley*
Founder, Morgan 3D Printers
p:+27120041883 | e:quentin@morgan3dp.com | w:www.morgan3dp.com | a:4 Burger
Ave, Lyttelton Manor, Centurion