Smoothieware/Smoothieware #874: spi brushless spindle driver

valpoder, 2016-03-15:
hello guys:

Would it be possible to interface a brushless motor driver chip such as allegro [A4963](http://www.allegromicro.com/~/media/Files/Datasheets/A4963-Datasheet.ashx?la=en) to the smoothieboard by the SPI interface ???  

Cheers ...

triffid, 2016-03-15:
Probably, but you'd have to write a driver for it

dandumit, 2016-03-15:
Unfortunately SPI it's not for long wires. You would have to place that
driver very close to your smoothie board.

DAniel

triffid, 2016-03-15:
That's fine, the motor wires can be long especially if you braid them

valpoder, 2016-03-15:
> Probably, but you'd have to write a driver for it

what do you mean by driver???  like SPI driver ???
i thought smmothie already has SPI libraries...
i have arduino code to drive the chip by spi by the way.. i guess this can be ported somehow

triffid, 2016-03-15:
Yeah we already have SPI driver but you need to add code to tell your chip
what to do.

Due to smoothie's modular nature it'll possibly be even simpler than your
arduino code, but certainly somewhat different

valpoder, 2016-03-15:
thanks so much...

What pins/connector should i use for my spi ???
where in the code should i write my "sauce' ?

Cheers :)

triffid, 2016-03-15:
There's a spare SPI port on smoothie board, and plenty of examples for
adding new modules in smoothie's existing codebase, probably some stuff on
the wiki too.

Basically your module should check config to see if it's enabled, then
listen for relevant gcode messages and issue commands to your chip as
appropriate

wolfmanjm, 2016-03-15:
There is already a motorcontrol module that sets up SPI based stepper controllers. Look at that one to see how to do it.
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/utils/motordrivercontrol/MotorDriverControl.cpp

wolfmanjm, 2016-03-15:
As this is a development question and not a bug report I am closing it.

closed 2016-03-15 by wolfmanjm