Smoothieware/Smoothieware #880: Zprobe grid issue with cartesian printer

cwaa, 2016-03-21:
I have set up Zprobe and I can't get it to work just right.  I have defined my bed as 200 x200, but during zprobe it goes out of that area.  When I redefined bed to 100 x 100 it stayed on the bed but only probed the lower left quarter of it.  When I first input G32 the nozzle does not go to the center of the bed.  So what have I set up wrong?

wolfmanjm, 2016-03-21:
this is not a bug, questions are not meant to be posted on the issues list. Yu can use IRC or the wiki forum to ask questions.

closed 2016-03-21 by wolfmanjm

cwaa, 2016-03-22:
Where is the right place???

qharley, 2016-03-22:
You can use IRC or the wiki forum to ask questions

>

cwaa, 2016-11-01:
This is not a complaint, but just seeking advice.  My printer quit two days
ago.  I found my laptop completely black and was about to toss it when I
took out the battery and then tried to restart it.  It did restart, but the
smoothie board would not.  I removed the Smoothie board and found the SD
card charred.  Also the positive side of the large moffet feeding the
heated bed was brown and the plastic was discolored.
Is there any value in this board or should I just keep it for parts?

I have ordered an new 5Xc with glcd kit from Robot seed to replace both the
board and the laptop.

This time I will solder the leads for the moffet.  Any other suggestions?

qharley, 2016-11-01:
Try the IRC for help. Does seem severely broken though