Smoothieware/Smoothieware #884: G32 Causes Z Offset on later G28 commands

tbrnn, 2016-03-24:
After the (successful) G32, there is now a calibration offset for X,Y,Z.

Running G28 after that causes a slight Z-move in the coordinate system. The endstop seems to be ignored and the error accumulates until the nozzle crashes into the bed.

The reason: During reset of the axes while homing, the compensation function is not considered. I implement a fix that does this. In order to ensure the correction move after G32 is executed, a G0 command is called (only implemented in ThreePointStrategy). 

Attached the patch.

```
diff --git a/src/modules/robot/Robot.cpp b/src/modules/robot/Robot.cpp
index aff08c2..4e752c2 100644
--- a/src/modules/robot/Robot.cpp
+++ b/src/modules/robot/Robot.cpp
@@ -771,6 +771,15 @@ void Robot::reset_axis_position(float x, float y, float z)
     // now set the actuator positions to match
     ActuatorCoordinates actuator_pos;
     arm_solution->cartesian_to_actuator(this->last_machine_position, actuator_pos);
+
+    // check function pointer and call if set to transform the target to compensate for bed
+    if(compensationTransform) {
+        // some compensation strategies can transform XYZ, some just change Z
+        compensationTransform(actuator_pos);
+        compensationTransform(this->transformed_last_milestone);
+    }
+
+
     for (size_t i = 0; i < actuators.size(); i++)
         actuators[i]->change_last_milestone(actuator_pos[i]);
 }
diff --git a/src/modules/tools/zprobe/ThreePointStrategy.cpp b/src/modules/tools/zprobe/ThreePointStrategy.cpp
index 9c3ddf9..202313e 100644
--- a/src/modules/tools/zprobe/ThreePointStrategy.cpp
+++ b/src/modules/tools/zprobe/ThreePointStrategy.cpp
@@ -143,6 +143,9 @@ bool ThreePointStrategy::handleGcode(Gcode *gcode)
             if(!doProbing(gcode->stream)) {
                 gcode->stream->printf("Probe failed to complete, probe not triggered or other error\n");
             } else {
+                // Move back to point of origin, this ensures the compansation move is executed after G32
+                zprobe->coordinated_move(10.0F, 10.0F, NAN, zprobe->getFastFeedrate(), false);
+                zprobe->home(); // home  at the end
                 gcode->stream->printf("Probe completed, bed plane defined\n");
             }
             return true;
```

qharley, 2016-03-24:
Something similar is already implemented in the ZGrid strategy.  It will
subtract the probe offset position from the ZGrid probe results in order to
repair the offset.  It is adjustable by entering the correct relative probe
position.

wolfmanjm, 2016-03-26:
This is technically incorrect. Homing sets the actuator position and does not use the compensation transform at all to move. On other types of arm solution like rotary delta or linear delta there is an FK done to get the cartesian coordinates.

In your case you are ignoring the instructions for three point leveling which explicitly states that the Z home position/endstop should be calibrated for the first probe position only. If you follow those directions then you would not see the error you are seeing.

What may be confusing you is that after homing Z the Z is set to the actuator position (in a cartesian it is the same as cartesian), however it is set with no compensation applied, so the first move after homing has to apply the compensation transform, which may cause a larger move than is expected as last-machine-position is actually incorrect directly after a home.
applying compensation transform to the actuator position is absolutely incorrect especially for non cartesians.

As was discussed in great detail here.. https://github.com/Smoothieware/Smoothieware/pull/725 the correct solution is to apply an inverse compensation transform to the actuator position set by homing.
I am still considering the correct fix, however what you have done may work for you but won't work for others. I'm closing this, but will work on an alternative strategy that will work for all arm solutions.

closed 2016-03-26 by wolfmanjm

tbrnn, 2016-03-29:
With active compensation function the compensation move is always applied after G28 during the back off move (this is done by a G0 move). It doesn't matter which axis is homed. The error accumulates with each G28 command until the nozzle crashes into the bed. The endstops are deactivated in this mode.

wolfmanjm, 2016-03-29:
No that is an incorrect assessment... back off home is done after the reset_axis is called, and there is no compensation applied when reset_axis is called.
The back off home is only called when limits are enabled, and that would then apply the compensation, so I do not see where anything gets accumulated, and FWIW I have used this for a long time and do not have the issue you are reporting.

wolfmanjm, 2016-03-29:
Here is the procedure that I use that works and does not exhibit the symptoms you are seeing...

1. G28 (Home XYZ)
2. Move Z up to about 30mm and attach probe
3. G32
4. remove probe.
5. M500 (to save probe results)
6. Move to the Probe P0 point (first probe point)
7. jog down to touch the plate
8. M306 Z0
9. M500 (to save homing offset)
10. G28
11. Move to 0,0 check Z height
12. Move to P0 check z height ditto for P1 and P2

Now it is set and no matter how many times I home it is correct.
I have a separate min z endstop and I do not use the probe to home.

What procedure exactly are you using that exhibits this behavior you report?