Smoothieware/Smoothieware #885: Homing sequence does Z even when it is not specified

MoonCactus, 2016-03-27:
My machine has no Z yet and my homing sequence is configured as "YX". But I had to write "nc" to the Z stops pins else G28 freezes the firmware (may be it tries to home Z indefinitely, and I wished there were a timeout & error instead!).

I find it counter-intuitive: homing should not add "Z" implicitly when it is not specified in the config. What if Z is a rotary axis that has no index (i.e. stop pin)? And it would be inconsistent with 4 or 5-axis machines (i.e. A, B additional axes).

IMHO, the sequence by itself should be enough to tell which axis to use when homing. No Z in sequence means no need to home Z. Well, I bet the issue is there.

wolfmanjm, 2016-03-27:
It is doing exactly what the gcode spec says it should do . this is not a bug. It maybe non intuitive but it is not a bug. In your case sounds like a misconfiguration.
The homing sequence setting is not used to eliminate what axis are homed, it sets the order of axis to home.

MoonCactus, 2016-03-27:
Understood. But it would be fine to know or to write it somewhere in the comments then. Two of us at the fablab almost thought the board was broken for a while! People not always refer to specs, especially with such a nicely configurable board :/

wolfmanjm, 2016-03-27:
looking at the wiki it could be improved, but that is why it is a wiki, you are welcome to update it and make it more understandable :)
However for your reference homing will home all configured axis, if an axis does not exist it should not be configured. You can do G28 X0 Y0 which will only home x and y (this is non standard but most reprap firmwares do that). the homing sequence config setting is an option in case you do not want the configured axis to home all at the same time.

wolfmanjm, 2016-03-27:
to be more clear if you configure a homing pin for an axis it will be homed. If it is not specified it will be NC by default. You must have had gamma_min_endstop configured in your config file.

thantik, 2016-03-27:
NIST RS274 says also: G28 goes to home thru the point given, in terms of the absolute coordinate system.  So G28 if following NIST should, for example in G28 X5, go to home coordinates through that position and then continue homing all axis.

MoonCactus, 2016-03-27:
Yep, learnt it the hard way. Sorry I tried but found no easy to tell about in the wiki without making it obscure.
I understand your points, but if "homing sequence" is a convenient hack to replace a sequence of G28 for each axis, I would keep on being intuitive imho, and sync only the provided axes. Or some error could be triggered while parsing the config, stating that an axis is forgotten in the optional sequence as it becomes a misconfiguration as you said (reminds me of the usefulness of --wall --wextra).

thantik, 2016-03-27:
Mach3 also does it the "RepRap" way.  If G28 is given with a single axis, it will home that axis alone and no other.

thantik, 2016-03-27:
Additionally, http://smoothieware.org/supported-g-codes says it in the way we expect it.  **_Home The given axis_**, or if no axis specified home all axis at the same time (edge)

wolfmanjm, 2016-03-27:
G28 in the NIST spec is NOT home, there is no G code for homing in the NIST spec. G28 is only homing in reprap. In the NIST spec G28 goes to a predefined position.  Linux cnc and Grbl do this and smoothie does it when in grbl-mode. (http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G28-G28_1)

To reiterate when you specify axis on the G28 command line (when in reprap mode) it will only home the axis you specify )or all of them if you do not specify an axis).

This issue is not about that it is about the fact that when you home all axis it will include any axis where the endstop is configured in the config file. If an axis is not homeable then do not specify it in the config.

MoonCactus, 2016-03-27:
Looking at the doc, would it make sense to add "none" as a third option to the "<axis>_homing_direction"?
This way,
1) it would make it clearer that homing is enabled on an axis as soon as the pin is set
2) it would still allow to specify end stop pin attribution without the implicit homing, so they can be used **only** as limit switches for example.

wolfmanjm, 2016-03-27:
@MoonCactus That seems like a good idea, I'll add it to my TODO list :)

wolfmanjm, 2016-03-27:
Also I'll add a note to the wiki saying pins must be set to NC if there is no endstop attached.

closed 2016-07-02 by wolfmanjm