Smoothieware/Smoothieware #886: ZGrid for PCBs

FreezeS, 2016-03-27:
Hi guys,

I have been trying to use the inbuilt zgrid probing for PCB milling but in it's current status I haven't been able to.

I have done the following:
- placed the etching bit 2 mm above the left bottom corner of the pcb and faked a homing (G28.3 X0 Y0 Z0) since my shapeoko doesn't have endstops.
- probed until it hit the pcb and set the offset (G38.3 Z-10; G92 Z0)
- lifted the bit 1 mm above the pcb and ran the scanning procedure (G32)

The result is this:
```
;Bed Level settings:
X0 A0.00 B-1.20 C-1.22 D-1.25 E-1.28
X1 A-1.05 B-1.11 C-1.14 D-1.17 E-1.20
X2 A-0.99 B-1.06 C-1.10 D-1.13 E-1.16
X3 A-0.94 B-1.01 C-1.05 D-1.09 E-1.12
X4 A-0.93 B-1.00 C-1.05 D-1.08 E-1.11
```

Good news is that it keeps the home position but the zgrid offsets are all bad.
Shouldn't the values be around zero? Instead they are around 1 (except for X0 A)

My config is:
```
zprobe.enable                true               # set to true to enable a zprobe
zprobe.probe_pin             1.28!^             # pin probe is attached to if NC remove the !, Azteeg X5 this is 1.29
zprobe.slow_feedrate         5                  # mm/sec probe feed rate
#zprobe.debounce_count       100                # set if noisy
zprobe.fast_feedrate         50                # move feedrate
#zprobe.probe_height          2                  # how much above bed to start probe NB only needed for G32 on delta

leveling-strategy.ZGrid-leveling.enable         true
leveling-strategy.ZGrid-leveling.bed_x           50
leveling-strategy.ZGrid-leveling.bed_y           50
leveling-strategy.ZGrid-leveling.bed_z           10
leveling-strategy.ZGrid-leveling.rows           5          # X divisions (Default 5)
leveling-strategy.ZGrid-leveling.cols           5         # Y divisions (Default 5)
leveling-strategy.ZGrid-leveling.probe_offsets  0,0,0
leveling-strategy.ZGrid-leveling.slow_feedrate  50
leveling-strategy.ZGrid-leveling.wait_for_probe  true
leveling-strategy.ZGrid-leveling.home_before_probe  nohome;    #  nohome homexy homexyz (default)
```

When I try to enable "zprobe.probe_height          2", the scan result is this:
```
;Bed Level settings:
X0 A-2.12 B-2.17 C-2.20 D-2.23 E-2.26
X1 A-2.03 B-2.09 C-2.12 D-2.15 E-2.18
X2 A-1.96 B-2.04 C-2.08 D-2.11 E-2.14
X3 A-1.91 B-1.98 C-2.02 D-2.06 E-2.09
X4 A-1.90 B-1.98 C-2.02 D-2.06 E-2.09
```

However, the old home position is modified, if I run G0 X0 Y0 Z0 it goes 2 more mm into the pcb.

Also, another question, I haven't tried it yet, I've broken enough bits already :)
Is it possible to set leveling-strategy.ZGrid-leveling.bed_x and y to be negative? This is absolutely needed to mill the backside of the PCBs, as the PCB2GCODE plugin generates them.

So, please let me know if I'm doing this incorrectly or if there is a bug in the code.

wolfmanjm, 2016-03-27:
ZGrid does not support pcb milling, please contact @qharley  for assistance as he is the only one that supports the zgrid stuff. I am closing this as the main devs cannot fix it. However I am linking it to the issue related to zgrid #649

closed 2016-03-27 by wolfmanjm

wolfmanjm, 2016-03-28:
I think the common knowledge says that it is best to do this offline. Now Smoothie supports G38.3 probing you can use offline tools like bCNC or http://www.autoleveller.co.uk/ or some other grid based offline tools.

FreezeS, 2016-03-28:
I was informed that @qharley would see this and it would be an easy fix for him as he's already familiar with the code. Having this will greatly simplify pcb etching on smoothie and I really hope it will be fixed at some point in the future.

qharley, 2016-03-29:
Hi @FreezeS 
I do not have a PCB etching machine, and as such would not be of much help unless you can give me a bit more info on your setup.
Please send me an image of your machine, complete with probe position, and perhaps a video of your homing arrangements currently.  I want to have a clear idea of what you have before I start changing anything.

FreezeS, 2016-03-29:
Hi @qharley !

This is my setup, just a shapeoko 1 with an etching endmill.
You can easily simulate this on a 3d printer by sticking a pcb to the print bed. Then just connect the probe to the pcb and the hotend.

[image]

cederom, 2016-03-29:
Hey :-) I am also interested to make my Zmorph3D intro PCB milling machine :-) I have no ZProbe.. do I need one? :-)

FreezeS, 2016-03-30:
@cederom : the smoothieboard already has a socket for zprobe, you can just use two crocodile clips like I used on the other end. Make sure to test it before trying a G38.3 :)

qharley, 2016-04-16:
The issue has to do with the way M306 reset the axis.  When homing, it did nit take the compensation transform into account.  This is normally not an issue since you always home again after setting, but in the event that you don't it will all the compensation after the fact, making it unsuable.

Pull request #910 fixes this issue.

wolfmanjm, 2016-04-16:
pull request #910 has been rejected as it is incorrect IMO.
M306 sets homing offsets so must be followed by a homing cycle to take effect. this is expected.
To set a dynamic instant Z offset you would use G92 Zxxx or more correctly G10 L2 P1 Znnn where nnn is the offset you want. This is the correct way to do it per the gcode spec, and will work instantly.
I do not know if this would fix the exact issue here, but confusing things is that M306/M206 is not to be used without a homing cycle. It is misusing them, and as they are not really specified in any GCode spec that is not surprising.

qharley, 2016-04-16:
I understand that it seems wrong to jump to the correct offset without
homeing, but what is the downside?