Smoothieware/Smoothieware #893: Panel displays incorrect position after G92

CapnBry, 2016-03-31:
I've noticed that sometime in December timeframe the panel LCD display started showing incorrect values for my Z axis. It looks like there were some internal changes made to how G92 position adjustments are used and these values are not used to calculate the position displayed on the panel. This may be for every axis but I only use G92 to calibrate my Z offset.

* Home Z axis: `G28 Z0`
* LCD Panel shows Z0.00. M114 reports Z0.00
* Enter a G92 value: `G92 Z1.0`
* LCD Panel still shows Z0.00. M114 reports proper Z1.00

Note that the robot works as expected, just the LCD display doesn't display the right value.

wolfmanjm, 2016-03-31:
fixed in PR #898 

TODO still need to make it an option to display machine coordinates.

closed 2016-03-31 by wolfmanjm

CapnBry, 2016-04-01:
Confirmed, working great now, thanks!