Smoothieware/Smoothieware #894: max31855: potential fire hazard

greyltc, 2016-03-31:
Maybe the issue title is an over reaction, maybe not, but [these lines](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/max31855.cpp#L69-L73) especially when taken in conjunction with [these other lines](https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/temperaturecontrol/max31855.cpp#L99-L104) seem to me to mean that thermocouple temperature measurement errors get silently ignored.

In real life, this means if your thermocouple becomes disconnected/open circuit (a very real possibility when you're heating things that are also moving) when you're by chance in a heating cycle, then your heater will run continuously. That's a fire hazard.

We should do our very best to consider all the potential (temperature measurement) failure modes the firmware can possibly detect and make sure they result in the heaters being shut off.

greyltc, 2016-03-31:
@hakalan, do you agree with my analysis here? or am I missing something?
When I unplug my thermocouple while using the latest released firmware.bin my temperature readings are frozen to the last value measured, so something clearly needs to change. I hope https://github.com/Smoothieware/Smoothieware/pull/895 fixes it. It will mean the ring buffer will get "poisoned" by an inf value whenever there is a read error. Why and how often were you seeing read errors?

hakalan, 2016-03-31:
I agree that the error handling seems to be broken and that this is bad. Your test confirms this. I distantly remember doing more work on this which unfortunately never got mature enough for a pull request. 

In my setup there were occasional errors which I blamed on switching of the heater (due to cables running alongside). So some robustness is needed, but several consecutive errors are not to be expected.

Grogyan, 2016-06-24:
Recently we have fixed a similar bug in Marlin firmware, my diagnostic code was to ignore n over temps to find a solution to the random over temps.
This was fixed by reducing the SPI baud rate only for the MAX31855 from 1/16 to 1/64th of the 16MHz Oscillator, i'd expect a similar approach could be done here. Changing SPI bit SPR0 to SPR1.

In almost all printing areas, a thermocouple is more reliable and consistent over a thermistor, and safer.

At 1/16 Osc this is the limit of the clock for the MAX31855 where the max transition time for the CS pin seems to be around 200ns

hakalan, 2016-06-24:
For me, the issue was never spi comms, but rather that the max31855 signalled various measurement errors via its status bits. Of course it makes sense to  robust against both.

Grogyan, 2016-06-24:
@hakalan my experimentation with Marlin leads me to believe that those status bits get triggered if something goes wrong with the SPI comms.
As I don't have a Smoothieboard compatible board I cannot test this on this firmware, however I noticed a dramatic stability on the SPI comm lines with the reduced baud with Marlin.

It is just a suggestion

hakalan, 2016-06-24:
@grogyan interesting! Thanks for your input. I have the max31855 connected to a beaglebone running machinekit at the moment. Will make some diagnostics to see how many errors I get there.

greyltc, 2016-06-24:
I've experimented with both slowing down the SPI clock and with properly handling the status bits. The bits are set when the TC is open circuit, shorted to ground or shorted to VCC. I've never actually seen any of them set in my setup. I don't think the status bits are causing any problem for me and I noticed no changes when messing with the clock rate. Here's my code for handling the status bits:
https://github.com/greyltc/Smoothieware/blob/useThis/src/modules/tools/temperaturecontrol/max31855.cpp#L68-L100

I put an oscilloscope (with SPI decoding capabilities) on my SPI lines and saw significant noise when my stepper motors were on. I think that was driving any reliability issues I was having. To address this, I switched all my wiring to shielded, twisted pair cables. Properly grounding the cable shield on the thermocouple was super important for me. My thermocouple readings have been vastly more reliable since, but I can't say they're perfect, my readings jump by ~1.5 degrees when my motors go from sleep --> powered.

Grogyan, 2016-06-24:
@greyltc thanks for that, I had long suspected EMI on the signal lines and thermocouple side, though without an oscilloscope on handle, it is hard to diagnose.
Thanks

DjDemonD, 2017-07-15:
+1 for shielded cables with pt100 sensor too, can bus cable is very suitable.

newt80ns, 2017-10-26:
I'm wrapping up my M31856 support, and wanted to confirm something that has been bothering me about the 31855 code, and some others I've seen.

This:
// Discard occasional errors...
    if(!isinf(temperature))
    {
        readings.push_back(temperature);
    }

Seems like it should be this:
// Discard occasional errors...
    if(!isinf(temperature))
    {
        readings.push_back(temperature);
    }
    else if(readings.size() > 0) { //And start subtracting from valid readings so we aren't stuck with the last 'good' samples
        readings.delete_tail();
    }

As it stands now it looks like if the sensor drops into a complete error state, we will simply report the last values and never react to a dead sensor/board/wire/connector/etc.

wolfmanjm, 2023-07-19:
Added the suggestion above so if there are too many errors the last good reading is not maintained

closed 2023-07-19 by wolfmanjm