Smoothieware/Smoothieware #914: Websocket Support

raykholo, 2016-04-17:
What would it take to get a websocket server integrated into Smoothie?  We'd like a more reliable/ lower overhead/ sleeker way than HTTP to be able to communicate commands to the smoothie (including writing a job file to the microsd card) from something like Chilipeppr or another browser based controller (wink wink).

Thanks,
Ray

openhardwarecoza, 2016-04-17:
Seconding the need for this.  

@laut3kr may also appreciate websocket support.   

What would be nice is just transpose the current serial as a websocket server (dont add complexity of extra / new commands) -  that way any host Currently written for the serial commands (cough,  LaserWeb,  cough)  can seamlessly switch to websocket directly to smoothie ethernet.

wolfmanjm, 2016-04-17:
TBH I have looked at this in depth, and given the lack of Flash storage and memory and the simplicity of the UiP stack I don't think it is feasible. But someone could prove me wrong.

wolfmanjm, 2016-04-17:
You can use plain old TCP/IP for 99% of what you want to do anyway, especially if it is a custom app.

closed 2016-04-17 by wolfmanjm

triffid, 2016-04-17:
Websockets aren't a magical mechanism for browsers to open raw TCP links,
rather they use a http header and a packetised wrapper around the data of
some sort. It's fairly complex, requiring the http server to negotiate
protocol upgrades and stuff.

Not the sort of thing I'd expect smoothie to handle well, if it could even
be made to fit

wolfmanjm, 2016-04-18:
FWIW what I have done in the past in these cases is use a javascript to java bridge (I know it is horrible but it works). And use raw UDP to communicate with the embedded client, (in this case it would be smoothie). The advantage of UDP is it is lightweight and fast, the downside is you need to do your own error checking and resend. You could also use the javascript/java bridge to create a TCP/IP communication layer. I have also done that. IMO they work much better than websockets, despite having to use ugly bridges.

openhardwarecoza, 2016-04-20:
Just pointing out the conversation in https://github.com/Links2004/arduinoWebSockets/issues/61 - has a lot of valid explanations as tho why websockets in this application is awesome.  I wouldnt want to swop the lan8720 for the esp8266 (;