Smoothieware/Smoothieware #918: Temperatureswitch suspends working when hitting print

dintid, 2016-04-25:
Posted here after suggestions by wolfmanjm

I have learned this is expected behaviour as the code is right now, as M109/190 is supossed to suspend/stop the execution of Gcodes.
Since the switch can execute Gcodes it is under this umbrella.

**My request** is for the switches to be either exempt from the above restrictions, or the addition of a new switchtype where gcodes are not possible.

**Example:** many other system critical fan usages I'd like to use
**Desired effect by using temperature switch:** Heatsink hotend fan (Calling it HHF from now on) turns on and off when hotend thermistor reaches 50c.
**Reason:** To avoid noisy HHF unless the hotend is actually hot and to avoid forgetting to turn it on if controlled manually.

**Symptoms - unwanted behaviour**
If temperature is below 50c on hotend when I hit print, the HHF does not start untill the print itself starts. Same for FAN.
If temperature is above 50c on hotend, when I hit print, the HHF shuts off and does not start untill the print itself starts. Same for FAN.
If print is stopped, the HHF fan stops as well!

Google-Group discussion over here:
https://groups.google.com/forum/#!topic/smoothieware-support/GQOyoaneDqw

wolfmanjm, 2016-04-25:
What I plan to fix is the temperature switch will turn on the fan (or off) when the set temperature is reached regardless of if there is an M109 active.
I do not plan on adding any new switch types or changing the way switches currently work.
This is solely focused on the one use case of temperatureswitch (which is already processed in on_idle) to be able to toggle the switch pin on/off as required in on_idle.
If the referenced switch  is set to issue a gcode on toggle, then it will still have to wait for M109 to finish.
The current issue is that switches are always toggled in on_main, which is blocked by M109/M190. This would not be an issue if there were no M109/M190 in the gcode file.
FWIW M109/M190 work exactly as they are supposed to work. So there will be no changes there.

qharley, 2016-04-25:
I like this. Currently a workaround is to manually turn on the fan before
heating...

dintid, 2016-04-25:
Awesome. The alternative ideas were just if they were easier to implement.
Solution would Work perfect in the usecase specificed.

@qharley one problem is that if you turn on the fan specified in a switch, the fan will be turned off once you hit print.
Usecase for me in this instance is to get the fans to Work autonomously without need to use gcodes at all.

wolfmanjm, 2016-04-25:
I'm not sure why your fans turn off when you print. That is nothing to do with smoothie I think. You must have a fan off M code in your gcode. look at the m codes at the start of the gcode file.
Some slicers issue fan off M codes as the startup gcode.

dintid, 2016-04-25:
When I say fans turn off, I mean the fans orginally started by switches are turned off when I hit PRINT the fans turn off. Once the print starts the Switches "reactivates" and the fans turn on Again.

qharley, 2016-04-25:
Do you perhaps have a line of gcode in your slicing config (Start-up
G-code) that turns off the fans?

dintid, 2016-04-25:
@qharley my startup code is untouched. 
I HAD the  default M107 turn off fans in there (cura), but removed it  it for testing. Didn't matter.
Learned startupcode is not processed untill the job itself start though, so it wouldn't matter.

Process so far as I've seen (without looking at the coding as I'm not competent for that job):
**When hitting PRINT**
<ol><li>Printer stops any switches from working (my switches to turn on fans)</li>
<li>Starts heating to the set temperatures from gcode file (heatsink fan does not start, since switch is deactivated)</li>
<li>Process startup codes once temperatures are reached.</li>
<li>Starts processing the rest of the gcode file (print job)</li>
<li>Switches are reactived and my temperature-controlled fans starts up Again. Which is rather late to have cooling on heatsink fan on hotend.</li>
</ol>

Might be other steps or stuff to it, but the important part in this aspect is how the fans controleld by switches are turned off once you hit PRINT and does not turn off untill the actuall printing starts after startup codes.

dintid, 2016-04-25:
Guys, I might fail to convey what I mean since you obviously keep having questions for me, but the solution presented by @wolfmanjm IS going to solve my issue :)

wolfmanjm, 2016-04-25:
There is no such thing as print in smoothie... basically the host just starts streaming gcode. The firmware therefore cannot turn off fans as there is no concept of print. So there must be a gcode in the gcode you stream to smoothie that tells it to turn off fans at the start of your gcode file. That particular issue is not going to chnage by my fix. If you tell the fans to turn off they will. You would be overriding the temperatureswitch at that point. I recommend you have no mcode on the hotend fan that allows it to be turned on or off. I think the issue was the incorrect config you posted. basically the same mcode for both print cooling fans and hotend fan. so when a print starts the gcode sends a code to turn off the print cooling fan, which would also turn off the hotend file , because you have misconfigured the switch for that.

dintid, 2016-04-25:
You KNOW what I mean... Print = hit print on display.  or just when you START the print.

Many programmers or Engineers has really lost the ability to just view stuff as regular people do. Meaning see it as it looks and not what happens behind the sceene.

No need to argue away at nothing :)

dintid, 2016-04-25:
I have tried removing the code on heatsink hotend fan. Made no difference.
I do not have any fan codes in my start code.
Can we please stop arguing here?

I don't know why Smoothie stops my temperature switch everytime I start a print job (hit print on my lcd).. the switch start up Again after the print itself starts.
I have tried multiple files and also a different slizer.

Mikk36, 2016-04-25:
@dintid There is nothing to argue, @wolfmanjm already told You that he will make it so that the temperatureswitch module will be able to toggle the fans.

Hotaman, 2016-05-11:
I had the same problem initially. Most slicers add code to turn off the cooling fan in the startup gcode. sometimes this can be difficult to edit but every slicer has a way to define the codes inserted at various stages. Once you stop the slicer from including the offending codes, temp switches work as expected.

My setup is exactly as you describe. My hotend cooling fan is set to turn on/off when crossing 40C. It works like a champ for me now that my slicer is doing things the way I want it to! Since I only have one fan on my effector, I also had to set the minimum allowed speed in the slicer so it wouldn't turn the fan off when part cooling was not needed (according to the slicer). That way, the fan speed can be increased for say bridging when needed, then it slows down to normal speed when done.
The firmware only does what it is told to do. Make sure your slicer is telling it what you want to happen!

The purposed changes will make setup easier and more consistent, even though they aren't really needed to get the results you seek.

dintid, 2016-05-12:
@Mikk36 what do you mean? I wasn't told anything off the sort.
 "already told You that he will make it so that the temperatureswitch module will be able to toggle the fans."..
**In fact, I was told by everyone to comment away the on and off codes....**
I don't ever manually control the fans.. I just mentioend it to show that they Work.

@Hotaman if that is the case, then the developers do not know how their firmware Works. They told me switches worked as intended, which was: being disabled as soon as a heatup code was initialized.. once the temperature was reached, the switches was reactivated.
So in effect, you say one thing, while all Devs says something else?

Hotaman, 2016-05-12:
@dintid, Interesting, I'm getting into the code now so I will see for myself how it works. I believe your description is accurate and consistent with what I've seen. I was just sharing my experience as a new user of this firmware as I had similar issues at first.

With a combination of slicer and smoothie settings I was able to accomplish my goal. However, I wish I could set this up in the firmware alone and remove the need for special slicer settings.

For example, if the fan is turned off while it is hot, the temp switch in firmware does not turn it back on! Changing the temp does not reset the switch either. I see uses for a simple one-shot threshold switch, which this appears to be but for safety reasons there should be an option to enforce the switch setting as a minimum. This would allow the fan to run at higher speeds and ensure it always runs at a minimum setting no matter what gcode is being received from a host when the switch is set.

Perhaps we are ultimately after the same thing. The only negative press I've seen for Smoothie revolves around the lockup bug and safety features. I will push and help until both are resolved because I like this project. Sorry if you thought I was putting you down, that is never my intent. I only wish to help and enlighten fellow travelers with my experience.

tiaanv, 2016-11-22:
Has this been addressed yet in one way or another... Still seems to be a problem in latest build.  Not to rehash, but there are workarounds, but they are not great...

kenoro, 2016-12-14:
I'm having the same issue here.

My only solution is to make a M-code subcode, so the M-code commands are M106.1 and respectively M107.1 to turn on and off the fan for the cooling of the hotend, manually or by a command in the file *.gcode. So that way you dont have a problem with what the slicer is generating as a Gcode.

As by my testing, the Switch and temperatureswitch work ok unless you stop them.
If the temperature is above the temperatureswitch.hotend.threshold_temp, mine is set to 35 degrees, the fan will start. But if I issue a command to stop it M107.1, the fan should start by it self after 3 seconds, when he pulls the temperature and sees that its above the threshold value.

It doesnt. And its not nice to ruin a E3DV6 hotend every time that happens.

What am I missing?

My config part on this is:

## Switch module for fanhotend control
switch.fanhotend.enable                      true             #
switch.fanhotend.input_on_command            M106             #
switch.fanhotend.input_off_command           M107             #
switch.fanhotend.subcode	             1	     	      # the subcode that the input on or input off commands respond to M106.1
switch.fanhotend.output_pin                  2.6              #
switch.fanhotend.output_type                 digital          # pwm output settable with S parameter in the input_on_comand
#switch.fanhotend.max_pwm
switch.fanhotend.fail_safe_set_to	     1                # 0 or 1 what to set the output pin to in case of a crash or HALT condition
switch.fanhotend.ignore_on_halt	             false	      # set to true to not set the fail safe value when a HALT condition is triggered

#switch.misc.enable                           true            #
#switch.misc.input_on_command                 M42             #
#switch.misc.input_off_command                M43             #
#switch.misc.output_pin                       2.4             #
#switch.misc.output_type                      digital         # just an on or off pin

# Switch module for spindle control
#switch.spindle.enable                        false           #

## Temperatureswitch :
# automatically toggle a switch at a specified temperature. Different ones of these may be defined to monitor different temperatures and switch different switches, useful to turn on a fan or water pump to cool the hotend
temperatureswitch.hotend.enable               true            #
temperatureswitch.hotend.designator           T               # first character of the temperature control designator to use as the temperature sensor to monitor
temperatureswitch.hotend.switch               fanhotend       # select which switch to use, matches the name of the defined switch
temperatureswitch.hotend.threshold_temp       35.0            # temperature to turn on (if rising) or off the switch
temperatureswitch.hotend.heatup_poll          3               # poll heatup at 15 sec intervals
temperatureswitch.hotend.cooldown_poll        3               # poll cooldown at 60 sec intervals
temperatureswitch.hotend.trigger

wolfmanjm, 2016-12-14:
@kenoro that is a completely different issue and does not belong here. FWIW it is working as designed. Overriding the fan on/off is a separate function and not involved with temperature switch at all.

wolfmanjm, 2016-12-14:
@tiaanv I think you misunderstand the issue and proposed solution. It has nothing to do with gcodes being defined on a switch. they can and do still work to allow you to manually turn the fan on/off. It would override the temperature switch if you want to do that.

tiaanv, 2016-12-14:
@wolfmanjm There is an excellent chance I don't understand the details ;)  But it seems to me that the temperature switch does not fire whist a "wait for temperature" gcode command is issued?  Is that at least correct?

kenoro, 2016-12-14:
I there a middle way to this? What if by bad luck the Gcode generated/issued contains such a command and the fan stops. shouldn't be a fail mechanism that would prevent that? I mean what if during print I send a command to switch off the fan of the exhaust, but in stead stop the fan of the hotend? After 3 seconds that should start back on. Or am I missing something?

dintid, 2016-12-14:
I've been using my printer as is since I posted this. Meaning I still use the switch to turn my hotend heatsink fan on and off at 50c.
While I still find how it works really annoying I really havn't had any issues.On MY machine.

I can see how people might get a problem if they have hotend at 200c then hit print - > heatsink fan will stop working and heated bed will start up -> problem is, if people use slow DC bed.. it might take 15 minutes or so to reach 90c... this is really bad when hotend is at 200c with no fan on it!!

I'm using a 500w AC heater under my heated bed, so it doesn't take a minute to get ready and the plastic in the hotend doesn't get at chance at being cooked in my machine.

tiaanv, 2016-12-14:
@dintid.  You are right.   It's happened a couple of times I've started a print from cold,  and then forgot about the hotend fan.   I then just hit the kill switch and start from scratch,  giving the temperature switch a chance to trigger.   I've only once completely forgotten about it,  but I also have a 500w 220V heated bed,  so total time us a few minutes.

However this shortcoming/bug rendered the temperature switch close to useless for me,  and I think for it to be safe,  it has to be addressed or removed.

dintid, 2016-12-14:
Yea, another thing is how any given "designator" can only be used for 1 switch.

I have fans on my motors since I use a heated chamber, and wanted motors to start alongside my heatsink fan.... just a no go. Only one of the switches would start.

I ended up using G21 as trigger for the motor fans, as I always have that in startcode of my prints, and then turn them off at end of print .... but it's really annoying.

wolfmanjm, 2016-12-15:
Fixed with PR #1072

closed 2016-12-15 by wolfmanjm