Smoothieware/Smoothieware #922: Failure when printing longer than 2.5 hours

Hotaman, 2016-05-09:
This failure normally happens after 6+ hours, but one test failed at 2.5 hours. I've tried 3 different models sliced with both S3D and Slic3r. expected print times are 10.5-14 hours to complete. It has failed 8 times in a row so far. The point at which it fails is always different. The main model I'm testing with now is a 150mm tall spiral vase so every layer is basically the same once you get past layer 6. A circle with spikes on the outside.

Failure mode: All communications on both serial ports is dead (USB and Aux). All movement stops. Temp and fan regulation is functional (maintains current temps and fan speed) so the PWM ISRs are still running. Never resets. Steppers are all enabled and holding position. The extruder stepper is whining like crazy but not moving at all. steppers are at their normal luke warm temp (running @ 1amp). LEDs are flashing.

I'm running the latest disablemsd.bin from the bin folder on Edge and the SD mounting is disabled. (Apr 23 10:11)

I've used both S3D on my Mac and a MakerBase TFT32 to feed the code to the controller. The controller is the latest MakerBase SBase controller.

There are no temp problems, power problems or anything else that I have control of. I'm 99% sure the whine is coming from the extruder stepper and it feels like it is vibrating, but does not move.

Hitting the reset button makes everything wonderful again. It is a new Kossel Mini Delta that is printing fantastic, when it is printing! In other words, it works perfectly right up to the point of failure.

I'm setting up another test run, when it fails this time I'll put some ear plugs in and scope the stepper signals on the extruder to try to understand why it is whining so bad. (Had to make some room on my bench to get it next to my test gear now that I have eliminated the environmentals (heat, PS, player, etc.))

I can and will help. I'm an EE/SE and have been building embedded systems for the past 30+ years. I'm not currently set up to compile/debug this project but I'm getting there.

Let me know if there is something I can look at for you or if there are any known causes of this behavior. I'll update this as I learn more.

arthurwolf, 2016-05-09:
Note: Extruder whining has already been reported, months ago, as one of the failure modes for the S3D bug.

Do you have a FTDI cable ? If so we could try doing a remote debugging session when the bug occurs.

Could you try the following two things : 
* Pass the gcode file through the online tool linked to at http://smoothieware.org/simplify3d ( even if sliced by slic3r ) and try printing
* Setting the resolution in slic3r to something high like 0.1mm and try printing again.

Thanks for reporting this.

Cheers.

Hotaman, 2016-05-09:
Ya I saw the S3D bug that was reported which is why I tried S3r. I will run the gcode through the preprocessor and give it a whirl.

Yes, I have a 3v3 FTDI so we can set that up if needed.

I'll let you know how the first two test go, we can go from there based on the results.

Thanks for the blazing fast response! I must have caught you at just the right time. :)

BTW I love what you are doing with this firmware! Excellent work!

arthurwolf, 2016-05-09:
> Yes, I have a 3v3 FTDI so we can set that up if needed.

Great, then we can probably use http://smoothieware.org/mri-debugging to look inside the microcontroller while the bug is happening. Will require you put aside some time, and come to IRC. No hurry, but would be nice to do.

Hotaman, 2016-05-09:
Sounds good Arthur, I ran the S3D gcode through the preprocessor, it dropped about 100K (19.2MB - 19.1MB). I have it running that gcode now. Should complete in about 10 hours. I'll hop on your IRC channel in a bit.

I was just about to setup the network connection and test that out. If the ethernet port works on this board, which it should, I can get you a telnet connection to it no prob. Otherwise I'll get setup on the serial port. Then we can work out a good time for both of us. I'm in the US Central time zone. What timezone are you in?

arthurwolf, 2016-05-09:
In in France. I generally find overlapping hours with US people after a
while.

dcousens, 2016-05-10:
@arthurwolf would it possible for Smoothie to scan the GCode and maybe output an error if it detects the S3D bug?

qharley, 2016-05-10:
Do anyone know what exactly happens in Smoothie when it hits? Surely we
should just be executing the gcode, no matter how senseless it may be.

dcousens, 2016-05-10:
@qharley assuming this a result of that bug (may not be),  this was the last I saw of it: https://github.com/Smoothieware/Smoothieware/pull/676

qharley, 2016-05-10:
Is there a sequence of gcode available that will always trip up
smoothieware? If so, we can test the countermeasures more effectively

triffid, 2016-05-10:
Sure, and if smoothie recieves G1 ... F0 then the "correct" execution for
this senseless gcode is to stop forever!

arthurwolf, 2016-05-10:
There is no way to reliably reproduce the bug, just ways to increase it's
likeliness to occur.

There is some work on smoothie's step generation upcoming that may solve
this issue.

Hotaman, 2016-05-11:
OK, running the gcode through the pre processor allowed the print to complete, however, there was still a ~40 second pause in the middle of the print, which I have seen smoothie do before. It continues just fine but leaves an ooze pile which is also discolored from the constant heat.

This bug is not an S3d problem, it IS a smoothie ware problem and needs to be addressed as such. S3D just exposed it. I can print S3D gcode on every other printer controller I have access to, including a makerbot gen5 and 8bit controllers without any issues. No pauses, no lockups, just excellent output.

Let's get to the bottom of the problem and get it fixed! This appears to be a major issue holding the adoptance of Smoothieware back. Like most, I will switch to a RAMPS before I run every file through a preprocessor, what a pain!

I'll get setup for Arthur to do a remote debug session, I'm also setting up a dev env so I can poke around myself. I haven't looked at the code specifically yet, just the overall structure and design of the project.

I'm currently using 1/32 micro stepping and I noticed that the smoothie board supports 1/16th. Has anyone explored how this effects the bug? Is it less likely to occur at 1/4th vs 1/16th?

My first impression based on the failure mode, is that the stepper ISRs are getting into a race condition (perhaps just the extruder ISR) which ends in an ISR infinite loop (It continuously interrupts itself). I'm going to scope the stepper drive signals which should shed some light on what it is doing to the extruder cold end when it fails.

The other Idea I have, since I have only seen it fail after hours of operation, is memory corruption. Not the normal kind, more like accessing shared memory when it's not really my turn so the values are not in the proper state at time of access. Again, a timing problem.

Is the problem Delta specific, or does it effect XY as well?

The default move slice for a delta is 10ms, how does changing this setting effect the problem?

@triffid I agree G0/G1 ... F0 is an invalid move, I would ignore all F0 params in firmware as a safety measure and output an error in the log.

@arthurwolf I'll let you know when I'm ready for you to connect. Probably take me a few days before I'm setup. Can't wait to see the new code, that's exactly where I expect the problem to be found :)

arthurwolf, 2016-05-11:
Hey.

> OK, running the gcode through the pre processor allowed the print to
> complete, however, there was still a ~40 second pause in the middle of the
> print, which I have seen smoothie do before. It continues just fine but
> leaves an ooze pile which is also discolored from the constant heat.

Good news.

> This bug is not an S3d problem, it IS a smoothie ware problem and needs to
> be addressed as such. S3D just exposed it. I can print S3D gcode on every
> other printer controller I have access to, [...]
8-bit firmwares take a lot of shortcuts to keep up, which is likely why
they don't meet that problem. It's likely if you took a 8-bit firmware and
just gave it 10x the processing speed, it's fair in more ways than Smoothie
does now ...

> Let's get to the bottom of the problem and get it fixed!
Yes, there's work going on right now to get there.

> This appears to be a major issue holding the adoptance of Smoothieware
> back. Like most, I will switch to a RAMPS before I run every file through a
> preprocessor, what a pain!
>
> I'll get setup for Arthur to do a remote debug session, I'm also setting
> up a dev env so I can poke around myself.

Great !
You need to be executing a firmware you compiled yourself in order to be
able to use MRI.

> I haven't looked at the code specifically yet, just the overall structure
> and design of the project.
>
> I'm currently using 1/32 micro stepping and I noticed that the smoothie
> board supports 1/16th. Has anyone explored how this effects the bug? Is it
> less likely to occur at 1/4th vs 1/16th?

It'd be less likely to occur yes. This happens at very high step rates.

> My first impression based on the failure mode, is that the stepper ISRs
> are getting into a race condition (perhaps just the extruder ISR) which
> ends in an ISR infinite loop (It continuously [...]

Actually, it essentially thinks it's not done moving, when it is. So it
misses the move to the next block, and just keeps executing the same block
forever.
We got that far debugging with another user, but if we can reproduce it on
your machine, we'd be able to confirm this, and/or to explore a slightly
different occurence of the problem.

> The other Idea I have, since I have only seen it fail after hours of
> operation, is memory corruption. Not the normal kind, more like accessing
> shared memory when it's not really my turn so the [...]
The only thing it's specific to is : high speeds, very high gcode density.
We excluded pretty much everything else.

There was *a lot* of work, by many people, on this, some months ago. That
ultimately resulted in the current understanding of the bug, and in the
gcode post-processor. There is a massive refactor of the smoothie motion
code in the works too, that is believed to fix this issue ultimately.

Cheers.

The default move slice for a delta is 10ms, how does changing this setting
> effect the problem?
>
> @triffid <https://github.com/triffid> I agree G0/G1 ... F0 is an invalid
> move, I would ignore all F0 params in firmware as a safety measure and
> output an error in the log. [...]

Hotaman, 2016-05-11:
Thanks for the info and the heads up on MRI! I just compiled the latest pull without error so I'm getting there :)

I jumped on the Smoothie bandwagon knowing it is fresh and new. (the $50 _available_ hardware didn't hurt!) It's obvious that this project has gotten a lot of TLC!

I'm looking forward to helping push Smoothie to the top of the heap! The results I'm getting out-of-the-box are amazing. I've been tweaking my i3 for almost two years now. It took me months to get the same results I had on day one with Smoothie and the Kossel Mini and my Mini is non-standard (larger foot print and 600mm carriage rails).

By the way, if anyone is interested, I put my 3v3 retractable optical Z-probe design on PinShape.com I use it to set bed height before each print. Now my prints always start correctly at least ;) Just need to get them finishing as reliably as the old i3. Soon, very soon, I can feel it!

wolfmanjm, 2016-05-11:
@Hotaman I'd just like to point out that by buying the MKS Sbase you are doing smoothieware no favors... they are parasites that do not support the open hardware initiative or support the smoothieware S/W development in any way. They use the firmware but provide no support, bug fixes, patches or money. Just FYI before you push the "cheap" smoothie clones.

dcousens, 2016-05-12:
@Hotaman I understand where you are coming from, but it is worth nothing that development on Smoothieware [by the core contributors] is more or less funded by the Smoothieboard sales,  so while its entirely up to you to save a dime and contribute in other ways,  for some,  it was probably worth the mention above.

Anyway,  this is probably de-railing the issue.

Hotaman, 2016-05-12:
Agreed, I would have bought two of them if they were available to purchase and I will purchase them in the future if they are available to purchase. Heck, once the firmware is stable I'll give the devs a night on the town if there is a donate link somewhere cuz I don't think I'll make it to France any time soon :(

Sorry if I stepped on a toe or two, I do that sometimes on accident :(

Now, back to the debugging :)

arthurwolf, 2016-05-12:
Hey.

On Thu, May 12, 2016 at 5:00 PM, Hotaman <notifications@github.com> wrote:

>  I tried to purchase an 'official' board but was unable to acquire one.
>
Yeah, we have a hard time being in stock all of the time, but shortages are
generally short.

> Did I see somewhere on the smoothie site that the firmware started with
> the awesome GRBL code base? Isn't that what 'Open' is about? The smoothie
> board is just a more advanced revision of RAMPS which is just a more
> advanced version of the CNC shield. I think the attitude you expressed was
> a contributing factor in the demise of Makerbot, which also was based on
> other open projects.
>
In this scenario, MKS is Makerbot ...
Both took from an open-source project/community, both didn't give back,
both took an open-source project and made it closed-source.
Only difference I can see is makerbot faked being community-friendly for a
while, and MKS outright says they don't care about contributing or helping
in any way, and will never be open.

I hope that helps see why there is a bit of friction when MKS is
mentionned. This is not about "you didn't get a smoothieboard" : anything
that's open-source ( like an azteeg, a sunbeam, etc ) is fine ... this is
really specifically about MKS as a company.

Cheers.

-- 
Courage et bonne humeur.

Hotaman, 2016-05-13:
Thanks for setting me straight @arthurwolf. I was curious exactly where the friction was. I really wanted the azteeg board, but alas they were also no where to be found when I needed it either. I'll be sure to pass the good word through my circle and on the net.

Back to the task at hand... I have compiled up the latest in Debug mode. I've disable all the modules I think I can do without but still can't quite get the network support in there (3K too big). I'll get you a dump with debug on then turn networking on and expose a telnet connection for you to use.

arthurwolf, 2016-05-13:
Hey.

Nah, you don't need to disable modules, and telnet has nothing to do with
this.

The plan is
* The bug occurs
* You enter MRI mode ( http://smoothieware.org/mri-debugging )
* You plug a serial cable to the uart serial port ( via a ftdi cable )
* You connect to mri using gdb ( very similar to what would happen with a
jtag cable )
* You come on IRC, and I give you commands to send, and you give me the
answers ( that or we do something fancy with ssh )

Cheers :)

Hotaman, 2016-05-13:
OK, I can do that no problem. I'll start it printing tonight so it will be crashed tomorrow. I'll make sure I can get into MRI before I start the print to crash it. I'll be in touch.

triffid, 2016-05-15:
If you press the ISP button after the board has started it should crash
into MRI debug mode.. There may be a config setting for this behaviour now
though; check the docs. Quick way to verify your ability to debug :)

The serial port acts as a regular command channel that accepts gcode when
the board is running normally; gdb won't talk to it in this mode.

Note that the USB connection will drop while in debug mode as MRI prevents
USB code from running, this is normal and unavoidable.

Hotaman, 2016-05-17:
Well here is how the weekend went. I started a test and it locked up after about 5 min. Unfortunately, I was unable to get into MRI with a break on the serial port (it doesn't crash into MRI) and the effector was right on the bed so I couldn't get under it to hit iboot. So I reset it and hooked up an external iboot button. Three prints later, it locked up in the middle of the night. Unfortunately my lovely wife was woke up by the noise and she unplugged it to 'help' me. I'm working on getting it locked up now at my office.

In the mean time, I spent a lot of time watching it work because I noticed under extrusion problems with the firmware I'm currently running. The firmware is defiantly having problems in the extruder step generation. It will randomly retract the filament at an extremely high speed between ~.5-2mm and then continue from there? It's like a sudden burst of spurious steps in the reverse direction. (I am currently running a build of edge from last Friday or so) This might be expected behavior for the current code base, I'm not sure. Just reporting what I see happening. It does it at random times, sometimes just once or twice, sometimes it does it every second or more for 10-20 seconds straight at various speeds of repeat.
I'm using the internal player with the gcode on the sd card, serial port connected with a 3v3 FTDI @115200 and nothing connected to the USB port. This spurious retract behavior is new from the nomsd.bin I was using (end of last month or so).

One question, how do I compile the nomsd version? I know the setting is right in front of my face, I just can't seem to see it right now.

Thanks for all the tips guys, I have my dev env all setup and working great on both my WS and laptop. GDB is working as expected now that I can get into MRI with a press of the button. Breaks on the serial port however do not invoke MRI mode as documented.

arthurwolf, 2016-05-17:
Hey.

Thanks for all the effort so far !
I don't know how to enable non-msd unfortunately, maybe somebody who knows
will tell us.

Cheers.

Hotaman, 2016-05-17:
@arthurwolf I don't think I was clear on that, sorry, I was looking for DISABLEMSD. I see that I need to define it = 1. Where is the best place to make the define?

arthurwolf, 2016-05-17:
I think the makefile at the root of the project is the right place. if not,
it's the one in src/

Hotaman, 2016-05-17:
Cool, thanks. I put it in the main make when I found it this afternoon, but haven't had a chance to try it yet.

Hotaman, 2016-05-25:
@arthurwolf, I'm closing this issue since it no longer represents the real problems with Smoothieware that I am having.

In conclusion, switching from 1/32 to 1/16th micro stepping has pretty much solved the lockup problem with S3D gcode for me. Post processed or not the code runs fine @1/16th with the exception of the extruder.

After reading through all of the code and issues relating to movement and my own testing and scoping, there are some fundamental design/timing flaws in step generation that rear their ugly heads when Smoothie is pushed to high stepping rates. I currently feel the root of the problem (for 3d printers at least) lies in how the extruder is implemented.

The extruder is treated as a normal tool by the 'Robot' but it is not a typical CNC tool which is just a chunk of steal or an on/off type tool that is moved around by the robot. The extruder has it's own continuous movement that must be accurately synchronized with the rest of the robot movement to work correctly. It's like an after thought that was just bolted on and as such is never really in sync with the robot movement which is why it is almost always the last thing to finish moving and it usually finishes late. So the head stops moving and the extruder is still squirting plastic. As the name 'follow mode' implies, it is always behind and never really in sync.

I'm not sure how @wolfmanjm step generation works since it is apparently locked up inside his private firmware and he is still 'working up' to implementing it in Smoothie. I don't see any activity on this front in his fork or anywhere else so I have no clue when it might come to fruition. To me the best solution (without going crazy) is to move the extruder step handling into the robot/planner so it is handled as a first class citizen along with all the other movement and can be synchronized correctly. I know this would kinda break the design mold currently in use but it seems like the best approach to me at this time to iron out the kinks so to speak. Of course I'm probably miss guided at this time, but time will tell.  

I'm planning to take a shot at this approach when I have some time late June/July unless someone else fixes these issues prior to then or something more interesting pops up ;) I'm currently involved in a very large project that needs to be completed before I take on anything new in ernest.

Here are some current results, the rim on the left is from my i3 running Marlin, the ones on the right are successive prints with Smoothie on my Kossel Mini using different retract settings (10mm,5mm,7mm). Both printers use identical Bowden setups and are currently running at the same speed. I use 10mm retraction on the i3. 
[image]

I'd love to here any other ideas floating around in the mean time, and of course, my thoughts will probably change a bit once I get more involved in the code.

closed 2016-05-25 by Hotaman

Claymonkey, 2016-05-25:
I thought the stepper drivers on the smoothieboard were A4982's and they were hard coded in the smoothieboard at 1/16 microsteps. Where you using external stepper drivers? The A4982 doesn't do 1/32 or 1/8th, only 1/1-2-4-16.  If I had to guess the design choice was 1/16th was the cnc standard and it had a larger footprint for better heat dissipation.

Mikk36, 2016-05-25:
He's not using a Smoothieboard, but an MKS Sbase instead, which has DRV8825 drivers.

Hotaman, 2016-05-26:
Yes I'm using the bad clone. I really feel bad for the Azteeg users with their unusable 1/128 micro stepping. Here are some prints @1/32 micro steps. The eggbot failed due to the lockup bug. If you look closely you can see the effects of the slight over extrusion at the ends of all the little line segments. It leaves an interesting texture on the print surface. I suspect the under extrusion on the fish was caused by the spurious reverse movement but I didn't watch it print so I don't know for sure. You can also see that the robots positional movement is spot on and very smooth. It moves really nice and is very quiet @1/32 but I have seen the spurious extruder  reverse movement when simply sending a G1 E10 F300 so I consider anything past 1/16 is just too unstable to be usable at this time for the Extruder module. Robot movement seems to be rock solid @1/32 so it probably works fine for other CNC machines at higher micro stepping as long as you don't try to move too fast.

[image]

wolfmanjm, 2016-05-26:
FWIW I use azteeg at 128 and 64 and have NEVER had a problem... then I don't use S3D and never would use such crap s/w. o this is a problem that is only affecting a few people, so don't go painting it with a large brush.

Hotaman, 2016-05-27:
@wolfmanjm Thats great news Jim! I should have an X5V3 in a couple weeks to test with. Now I'm looking forward to it's arrival again!

It only effects a few people because all the rest of them have moved to other controllers, waiting for Smoothieware to mature. It is a problem if you push the board to high stepping rates. I have no problems @ 30mm/sec. The faster I go, the more problems I have.

One question, I use 4 different firmwares at this point. Smoothie is the only one that has problems with S3D. How do you propose to correct the problem if you never use S3D?

Just because YOU don't use S3D, many others DO. When it works fine with the other firmwares, how do you justify blaming S3D?

I want to help, but you keep pushing me away. It's like AA, the first step to a solution is admitting you have a problem.

wolfmanjm, 2016-05-27:
@Hotaman that is total BS, and you know it, there are literally hundreds if not thousands of users of azteegs having no problems at all. So this thread is closed if you insist on bullshiting.