Smoothieware/Smoothieware #924: Maximum speeds not checked when moving axes from the internal web interface

dethegeek, 2016-05-16:
Hi

I'm setting up my printer with the smoothieboard.

I noticed that I can move X axis at higher speed that the speed limit defined in the config file.

Is it a wanted behavior ?

arthurwolf, 2016-05-16:
There is no reason why this would happen. How did you specify the movement
speed, and your maximum speed ?

wolfmanjm, 2016-05-16:
Jogging speed is limited by whatever is set in M203. This works. I suspect you have either the config-overide set or the config is wrong.
Check M203 with M503 see what you have it set to.

closed 2016-05-16 by wolfmanjm

dethegeek, 2016-05-16:
I think I have to understand the difference between x_axis_max_speed and  alpha_max_rate

Setting my speed limit in x_axis_max_speed solved the issue.

what is the diference between max_rate and max_speed ?

dcousens, 2016-05-17:
@dethegeek at a guess, you could probably put `x_axis_max_speed` at close to infinity, and instead just limit the `alpha_max_rate`?
Relevant file seems to be Smoothieware/src/modules/robot/Robot.cpp

dethegeek, 2016-05-17:
@dcousens 

x_axis_max_speed was 30000 and alpha_max_rate 12000

The issue has been solved by setting x_axis_max_speed to 12000 too

My understanding of the code you mention seems confirm these new settings. However, I don't deduce setting x_axis_max_speed no a near infinite value is a good choice.