Smoothieware/Smoothieware #927: Arcs should be interpolated to max allowable error, not just divided into a fixed length

Zaaphod, 2016-05-19:
The current scheme of breaking arcs into line segments of a maximum length works for relatively small arcs, but arc segments can have huge radii, it's not uncommon to have radii of  20,000mm.  If you encounter these large radii arcs on a large machine, you will end up with thousands of tiny segments, and this is not necessary.    What has been done by other systems to prevent this is to do a true interpolation of the arcs.  This method normally asks for the "Maximum Arc Error", instead of a segment length.  For any arc-segment, if you draw a straight line across it, the distance from the center of the line to the highest point of the arc is the Arc Error.  Typically this arc error would be set to a very small value like 0.01mm.  for example on a 12,700mm radius arc-segment 2286mm long, if you just break it into 0.5mm pieces you will have 8192 segments, but if you interpolate it and allow a max error of 0.01mm,  you end up with only 128 segments, each 17.8mm long, that's quite a huge difference. having more segments doesn't help anything because even with these long segments, the maximum error from a true arc-segment is only 0.01mm. 

Even on a much smaller arc, say 120mm radius 50mm long,  if you interpolate it with a maximum error of 0.01mm, you get 32 segments, each 1.6mm long, but if you just break it into 0.5mm pieces you get 128 of them.  What we really care about is how far from a true arc the line segments end up being, and that's what interpolation is, it allows for the longest segments possible that will produce no more than the allowable error.

on the other end of the scale, interpolation also works better,  if you have a 90 degree arc-segment with a radius of 2mm, if you just divide it into 0.5mm pieces, you only get 4 of them, these 4 pieces will have an arc error of .019mm...  But if you interpolate it with an max error of 0.01mm, you get 8 segments each about .2mm long

I'm not sure what the formula is to calculate this, it's something my CAD system does, but I think it would be beneficial to everyone using smoothieboard if it did the same thing internally. It would process most arcs much faster but still maintain precision on small arcs.   I would be happy to help work out the formula for this if it would be helpful. 

I have done some thinking about it from a programming perspective, and basically to implement this, all you need to do is ask for "max arc error" then when you encounter an arc, you get the radius of it, which is simple...  `SQRT(SQR(X-I)+SQR(Y-J))` if you have absolute I and J, or `SQRT(SQR(I)+SQR(J))` if I and J are relative.    From there you calculate the longest line segment you can have across a fictional arc of that radius...   I haven't worked out the formula for this yet,  but it can't be that difficult..  Wait a minute.. Actually as I was typing this, I realized the formula for it...  basically imagine you are at the top of a circle with radius R,  Draw a horizontal line across it, now draw one radius from the center of the circle to the point where the right end of the line segment will intersect with it.  Next draw another radius from the center of the circle to the top of the circle, this divides the line segment in half. the line above the horizontal line is the arc error, we'll call that E..  now it's easy..  the radius is the hypotenuse of a right triangle, the radius minus the error is one side and the length of the right half of our line segment is the other side, we'll call that L.  the length of the line segment from the center to the right side is simply `L=sqrt((R²) - ((R-E)²))` multiply by 2 to get the entire line segment length, making the formula  `mm_per_arc_segment=L=(sqrt((R²) - ((R-E)²)))*2` 

Once you have this max segment length, you just plug this number into your existing segment length variable "mm_per_arc_segment" and let it process as it does now because that part will work fine if we dynamically re-calculate mm_per_arc_segment for each radii.  so all that is really needed is to do these two formulas to get the new mm_per_arc_segment before it processes the arc into line segments.

I hope this is helpful   :)   
[image]

arthurwolf, 2016-05-19:
Hey.

I realize that true interpolation is more "ideal", but it makes *a lot* of
things ( including planning and stepping ) insanely more complicated, and
has no real benefit that I can see for the end user ...
It'd also potentially require much more processing power, that we can't
afford in step generation.

What's wrong with segments, other than they are maybe not as "elegant" as
doing actual live interpolation ?

Cheers.

Zaaphod, 2016-05-19:
this method would greatly **reduce** the processing needed for most arcs..  you only need to do 2 calculations at the beginning of the arc, and then instead of generating hundreds or thousands of segments, you only generate what is needed for any given arc.  most arcs have a drastically reduced number of segments because of this and therefore much less processing power needed.   I guarantee that the vast majority of arcs will have fewer segments and process much faster.  It's not insanely more complicated, I even gave you the formulas.. just figure out the radius of the arc, then figure out the new  mm_per_arc_segment then do what you are already doing...   except your mm_per_arc_segment will be small when needed but normally, much much larger, but the arcs will still conform to the max allowable error.  any arc larger than 13mm will have fewer segments

the problem with fixed length segments is there are way to many of them on large arcs, but even on medium arcs there are still way more then you need so you are wasting processing power calculating all these unnecessary segments.  less segments can calculate and run much much faster

dcousens, 2016-05-19:
Concept ACK,  this is how I've done this in the past when creating 2D geometry... shame I can't seem to find the code...

Zaaphod, 2016-05-19:
I gave the code already.. 

find the radius of the arc with:
` SQRT(SQR(X-I)+SQR(Y-J)) `if you have absolute I and J, or `SQRT(SQR(I)+SQR(J))` if I and J are relative.
find the new mm_per_arc_segment with:
`(sqrt((R²) - ((R-E)²)))*2`
then process it the same as now.
attached is a spreadsheet proving the calculation time will be  reduced on all arcs larger than 3.5mm radius and drastically reduced on all arcs larger than 13mm by drastically I mean less than HALF the processing a 13mm radius arc going from 82 segments down to 40
spreadsheet is attached for your analysis.

[interpolate.xlsx](https://github.com/Smoothieware/Smoothieware/files/272596/interpolate.xlsx)  

you do two simple calculations to save generating way too many points on most radii, and small radii have improved resolution.   there is no downside to this..  it's simple, and overall it will save processing time.. that's why it's the normal way arcs are segmented in industry.   this was a NECESSITY back in the early 80s when we had only 2MHz processors.

dcousens, 2016-05-19:
@Zaaphod there is a difference between giving the algorithm,  and giving the code.  I'll see if I can write this up,  but,  if you had provided the code,  you could simply submit a pull request and this could be merged soon after.

Zaaphod, 2016-05-19:
I can write the code.. it's just 2 formula... What language is this written in and where are the arcs processed?

Zaaphod, 2016-05-19:
we just need to define a new variable called max_arc_error then then right before you segment the arcs put in something like this:
 
```
If G91
mm_per_arc_segment=(sqrt(((SQRT(I**2+J**2))**2)-(((SQRT(I**2+J**2))-max_arc_error)**2)))*2
else
mm_per_arc_segment=(sqrt(((SQRT((X-I)**2+(Y-J)**2))**2)-(((SQRT((X-I)**2+(Y-J)**2))-max_arc_error)**2)))*2
```

dcousens, 2016-05-19:
@Zaaphod the relevant code is [here](https://github.com/Smoothieware/Smoothieware/blob/3332442f94b68f409962239aebd9130102d36e1a/src/modules/robot/Robot.cpp#L1022)

Zaaphod, 2016-05-19:
@arthurwolf 
>  What you are talking about, is a different segmentation method, which is based on a maximum allowable error, instead of a fixed segment length, correct ?

Correct

>I think that can probably make sense ( though it seems like only a very minor improvement ). 

It depends on the application.  If you are cutting on a mill, router, or laser, parts that are drawn on a CAD system, it will greatly reduce the line segments needed but keep within the allowable error, but if you are 3D printing something made out of true type text, the arc segments and line segments are are all so tiny anyway, that it wouldn't make much difference at all.

>I think if somebody coded this, it'd get accepted, but I'll let others comment on it too.

I will volunteer to contribute code and testing, but I may need some guidance as I am not familiar with the procedure on how to compile this.   I'm willing to learn and to put in the time to help in any way I can.

Zaaphod, 2016-05-19:
@dcousens  thanks!  I will work on it today

Zaaphod, 2016-05-19:
This looks like C++ to me, but can someone please verify or correct me if it's something else?

dcousens, 2016-05-19:
@Zaaphod indeed it is C++.

Zaaphod, 2016-05-19:
It's been a while since I did any C++ programming, so my syntax might be slightly off, but what's needed is the following code on inserted at line 1020

```
 float mm_per_arc_segment=(sqrt((radius**2)-((radius-max_arc_error)**2)))*2
```
I believe in c++  variable**2 means to square the variable, but please correct me if I'm mistaken.

the radius is already calculated on line 1105, and is used on line 1011, so I should be good to use it on line 1020 as well.   Since the radius is already known this only requires one more calculation.

we would also need to define max_arc_error and allow it to be set in the config file.  Not sure where to do those things

I am new to github so if someone could instruct me on the correct way to compile and test this as well as submit a change request it would be appreciated

Zaaphod, 2016-05-19:
I've been thinking about this..  It should only re-calculate mm_per_arc_segment if max_arc_error is not 0.  That way if it is not defined it will not try to run with a zero error, which would cause a mm_per_arc_segment to be zero which would cause a divide by zero on line 1022

```
if(this->max_arc_error != 0.0F) {
     float mm_per_arc_segment=(sqrt((radius**2)-((radius-max_arc_error)**2)))*2      // Calculate largest line segment length which will keep within the alowable maximum arc error
} 
 
```

wolfmanjm, 2016-05-19:
@Zaaphod FYI none of that is valid C++ syntax :)

Zaaphod, 2016-05-19:
@wolfmanjm Yeah, I thought it would be off a bit..  I have done very little C++ and it was years ago.   For one I left off the semicolon at the end.  

Could someone who knows what they are doing please take over and perhaps I can follow along and learn something :)    I have been going around github and I'm still not sure how to go about compiling this to test it, or how to make a request.. or actually do anything at all if I'm going to be honest.

Zaaphod, 2016-05-19:
oh I think variable**2 for powers is pascal... how's this?  please feel free to correct me :)
```
if(this->max_arc_error > 0.0F) {
     float mm_per_arc_segment=(sqrt((pow(radius,2))-((pow((radius-max_arc_error),2)))))*2;    // Calculate largest line segment length which will keep within the alowable maximum arc error
} 

```

I still don' know where I define my max_arc_error variable or how to read it from the config file.. I'm lost

dcousens, 2016-05-20:
@Zaaphod and thus lies the difference between the code,  and the algorithm :+1: .  I'm happy to look into this for my own purposes,  but,  it could be a while.

rene-aguirre, 2016-06-03:
@Zaaphod , take a look to my patch to get you started (or maybe done, I don't have any smoothieboard): https://github.com/rene-aguirre/Smoothieware/commit/83c6e067ce69357a55b56d864fde068452551617

I mapped a new configuration for the arc error. Also simplified the mm_per_arc_segment estimation.

(got to this issue just looking for the state of smoothieboard).

Zaaphod, 2016-06-05:
@rene-aguirre Thanks for getting this started!  I really don't know much about github.

I checked your simplified formula and it works perfectly.  Great job coming up with that!   Here's an updated spreadsheet showing the proof the new simplified formula produces the same results.   I have also made the spreadsheet calculate the actual error for the existing method as well as this interpolation method, and made it dynamic so the variables for the fixed distance and max error can be changed.
[interpolate2.xlsx](https://github.com/Smoothieware/Smoothieware/files/299472/interpolate2.xlsx)

On line 1027 it looks like you are replacing mm_per_arc_segment if the new value is larger.    I've been thinking about this and would like to discuss it a little.  My initial opinion is that you should just use the new value all the time, because if you check out my analysis, the only time there will be more segments with this formula, is when the Arc Radius is less than 3mm,  but for those small arcs, you really want more segments, otherwise you are making it into too much of a polygon instead of keeping within the specified arc error.    This may not seem important, but it could cause problems on something like a PCB router where the engraving is extremely small and there isn't necessarily room to cut off the corner of an arc with an error many times larger than specified.

On the other hand however, if you have something like a large marching cutting large parts,  Then those tiny arcs are close enough to the same point that having fewer segments would be better, and possibly run smoother, even though the error would be larger than specified, the error is still negligible given the size of the final part.

So I was thinking, maybe we need one more variable to determine if we do the check or just always use the interpolated value, but then I realized another variable is not needed at all.. all one would need to do is set mm_per_arc_segment in the config file to zero, and then the interpolated value we calculate would always be larger and it would use that.. in fact after playing with my spread sheet a little.. I think I would like to set my large CNC router with a mm_per_arc_segment to 1mm and mm_max_arc_error to 0.01, but my milling machine where I want precision, but runs much slower,  I would set mm_per_arc_segment to 0 and mm_max_arc_error to 0.001 

The reason I am bringing this up is that I think the code as is is actually a great solution, It allows the user to use either method, or a combination of both for whatever the purpose is,  however I think it needs to be clearly documented, so users know exactly what these two variables do and how they work with each other... if you are doing very small, intricate work, set mm_per_arc_segment = 0 and just use mm_max_arc_error, but for cutting large parts perhaps it would be best to use both.  
Here's another spreadsheet showing a combination method as well where it only uses the new interpolated value for mm_per_arc_segment if it is larger than the specified mm_per_arc_segment.
[interpolate3.xlsx](https://github.com/Smoothieware/Smoothieware/files/299482/interpolate3.xlsx)

rene-aguirre, 2016-06-05:
@Zaaphod , you got the point, the idea was to use the less segments using a combo of former method and your new proposal.

If this is confusing or not expected then I'd propose removing mm_per_arc_segment altogether, and only keep the mm_max_arc_error new setting.

@dcousens , Please feel free to comment.

wolfmanjm, 2016-06-05:
Any solution needs to maintain backward compatibility.

dcousens, 2016-06-06:
@wolfmanjm in what way?  The `mm_per_arc_segment` concept doesn't make sense when you can simply define `mm_per_arc_segment` in terms of it.

`mm_per_arc_segment` is derived from `mm_max_arc_error`,  so if it remains defined,  it should probably just be ignored.

What is the deprecation cycle?

wolfmanjm, 2016-06-06:
There is no deprecation cycle, if there is a meaning to a config entry it must be maintained. It is not an option to ignore it.

dcousens, 2016-06-06:
@wolfmanjm what do you mean?  Obviously in this case we can make a new meaning that is backwards compatible,  but it certainly isn't exactly the same behaviour.
Ideally you'd deprecate `mm_per_arc_segment` and make something new `min_mm_per_arc_segment` or some such.

How do you deprecate features that have replaced by *better* features?

wolfmanjm, 2016-06-06:
I mean that if a config entry has an expected result and someone has changed it from the default and expects it to have a specific effect, it is not acceptable to have the functionality suddenly change.
We maintain backwards compatibility. With no surprises.
We have not deprecated stuff, usually new features need to be activated and configured, we have not changed existing features in a way that would cause a surprise.
It sounds like the original suggestion was to only activate this if it was set in config, that  sounds like it would be backward compatible. As there is no PR anyway it is academic, I was merely pointing out that if there were a PR and it wanted to be accepted it needed to maintain backward compatibility, or not change expected behavior if the user has an old config and they flashed a new edge.

dcousens, 2016-06-06:
:+1: no worries.

I guess then the default should be that `mm_max_arc_error` remains undefined (say infinite),  and if it is defined,  then `mm_per_arc_segment` can act as a minimum?

Zaaphod, 2016-06-06:
I agree with backward compatibility, and the code would need a slight modification to assure this.    
To make this backward compatible with no change in functionality, we should make the default of mm_max_arc_error = 0 and change the IF statement on line 1025 to also only be true if mm_max_arc_error > 0   as I had indicated in this previous post:
 
> I've been thinking about this.. It should only re-calculate mm_per_arc_segment if max_arc_error is not 0. That way if it is not defined it will not try to run with a zero error, which would cause a mm_per_arc_segment to be zero which would cause a divide by zero on line 1022
> 
> **if(this->max_arc_error > 0.0F)** {
>      float mm_per_arc_segment=(sqrt((radius**2)-((radius-max_arc_error)**2)))*2      // Calculate largest line segment length which will keep within the allowable maximum arc error
> } 
> 
> *edit changed != 0 to > 0 there isn't any use for a negative arc error either.

setting an mm_max_arc_error = 0 is not technically valid, as you would end up with infinite segments, so we should be checking for this anyway, and since mm_max_arc_error = 0 is invalid for computational purposes, it can safely be used as an easy way to disable the new feature should it be required for some application.

I'm curious if a sample config file should be distributed with this,  the sample config could have recommended settings for mm_per_arc_segment and mm_max_arc_error, as well as come comments explaining them, and explaining how setting mm_max_arc_error to zero would disable the new feature and setting mm_per_arc_segment to zero would use nothing but the new method.  It would be invalid for both to be zero

Zaaphod, 2016-06-08:
I've been testing this and made my own fork to make it easier to show things I've changed 
[Zaaphod Fork](https://github.com/rene-aguirre/Smoothieware/compare/edge...Zaaphod:Zaaphod--Disable-Arc-Error-Method-&-output-diagnostic-information?expand=1)

As discussed this allows setting mm_max_arc_error to zero having the effect of disabling this method.  I also did a printf command to show the contents of the variables for testing purposes.

I'm running into an issue where I can manually change the default mm_max_arc_error on line 185 here: [rene-aguirre@83c6e06](https://github.com/rene-aguirre/Smoothieware/commit/83c6e067ce69357a55b56d864fde068452551617) -re compile it and then it will use the new default value I put in,  but adding the line
`mm_max_arc_error                                          0.02 `
 in the config file seems to have no effect.  I'm not sure where to look to correct this.  It seems like line 185 should make it get the value for mm_max_arc_error from the config file.. but it doesn't seem to be working

Mikk36, 2016-06-08:
mm_max_arc_error_checksum vs mm_max_arc_error

Zaaphod, 2016-06-08:
Thank you, that fixed it!

Zaaphod, 2016-06-08:
Here are the current changes,  the mm_max_arc_error is now working in the config.
[Zaaphod - testing](https://github.com/Zaaphod/Smoothieware/commit/4d0f60a9a1aaa34e23e2ba6cc50af574550bd56b)

Zaaphod, 2016-06-09:
If anyone following this is curious why I changed line 1032, the answer is in this spreadsheet:  
[interpolate4.xlsx](https://github.com/Smoothieware/Smoothieware/files/305954/interpolate4.xlsx)

My original spreadsheet was rounding the distance to travel divided by the interpolated distance to get an integer number of divisions, however after analyzing my spreadsheet closer, I realized this caused the error to sometimes be a little lower than the target error and sometimes a little higher.. but since it's supposed to be the MAX arc error, the actual error should never come out higher than what is specified.  By adding 1 and truncating it gives you the fewest number of divisions that guarantees you are always lower than the max arc error specified.... so now it truly is the max arc error.. I know it's splitting hairs, but it might as well be correct.  This also improves the accuracy on small arcs when using the original method, and the on larger arcs the change is so insignificant no one could ever notice it.

I have been rigorously testing the calculations in this to see that the results are always as expected, so far,  I have not found any problems yet.  There is defiantly a bottleneck on the larger arcs that prevent them from running at the same speed I can run them at with this new method.  This makes sense since the new method does way fewer line segment calculations.  I will do a more detailed analysis of this as well.

Zaaphod, 2016-06-09:
While testing this I realized that the original calculation on line 1032 could easily result in a zero which would cause divide by zero errors on lines 1035 and 1036.   Any time mm_per_arc_segment < the circumference of the circle, you would get zero,   The updated calculation I submitted in a [Pull Request](https://github.com/rene-aguirre/Smoothieware/pull/3) back to rene-aguirre is not not only is more technically correct for both methods, but it solves this issue being the minimum number this can ever result in is a 1, not 0.    The reason it is more technically correct is this will never produce an error greater than mm_max_arc_error when the new method is used, and it will never produce line segments longer than mm_per_arc_segment when that method is used..  The numbers are such that weather you are slightly above or slightly below the error specified or above or below the segment length specified is extremely minuscule and insignificant, however removing the possibility of a divide by zero makes this adjustment much more important.

Zaaphod, 2016-06-11:
Update: 
I've been testing this extensively,  The calculations are now flawless and always produce the expected result.  Now that I am satisfied the calculations are correct, I have connected my smoothieboard back to my my cnc router for testing.  I've been running circles all over the place using both the fixed length method and the new interpolation method and I'm happy to report that the new interpolation method is working perfectly.   My CNC router uses half step drives, but even with the low step rate associated with half step,  I do notice I am now able to run large arcs at full speed, where as with the old method my speed was limited on large arcs.  I am not sure if the limitation was due to the thousands of points it was trying to reach, or if it was because it was so busy with calculations that I could not move any faster, but the new method solves this issue completely. 

Here is an example of this:  I was running a test arc segment 1350mm long with radius 1850mm.  I run two programs, one to position at the beginning of the arc segment, and one to run the arc segment.  that way I can time only the arc without the positioning move.  With the old method, I could not run this faster than 12 seconds, this works out to about  6500mm/min (F6500).  I am just using the pronterface run time, but I did put M400 at the end of my program so it would give an accurate (to the nearest second) run time of the arc itself.  If I put in F7000, or F10000, even F20000 it made no difference, it would complete the arc in 12 seconds.  

With the new method, I am now able to run it very fast, and it is noticeably smoother as well. With the new method if I run it with F6500 it takes 12 seconds, F7500 it takes 11 seconds and F10000 it takes 8 seconds, and F20000 takes 5 seconds.   I even tried setting mm_max_arc_error = 0.001 and I am still able to complete the test arc in 5 seconds at F20000, however I do notice with this arc error, its running a bit rougher,  not as bad as with the 0.5mm fixed segment method, but noticeable.  This seems to confirm that if you have too many line segments it's having trouble hitting all the points in the allowable time.. there is absolutely no reason for all those segments anyway.

I am completely satisfied with the effectiveness of this new method on my CNC router.  

I have a milling machine with microstep drives I can test this on. It has a DOS computer on it, so I'm currently building a win 7 computer to test smoothie on it.  I expect that because of the microstepping requiring a higher step rate, that I will have an even more noticeable difference between the fixed length method and the new interpolation method.

Zaaphod, 2016-06-13:
I've completed testing my my milling machine with microstep drives. I experienced the same performance improvements on that machine as well. Even with microstepping, I the machine runs large arcs much smoother with this new interpolation method than it does with the fixed segment length method.
I decided to perform a test to see what causes the performance limitation of the fixed length method,  and I used my CAD system to do the interpolation with both methods.  My thought was that if I 'pre interpolated' the arc into line segments, then smoothie would not need to use processor time to divide the arc.  However when I ran my file of 0.5mm line segments with G01 commands, it had nearly the same limitations that it had letting smoothie do the divisions into arc segments, in fact my top speed was actually slightly slower than letting smoothie do the divisions.  This seems to indicate that it is not a calculation time issue causing the limitation but rather the attempt to go hit thousands of points around the arc when only a few hundred are needed.  
I also 'pre-interpolated' the same arc setting a maximum arc error of 0.01mm and I was able to run this file at the same high speeds as when I used the new mm_max_arc_errror = 0.01 on smoothie.    With my test on a microstep drive machine I believe I have tested every possible condition, and I am VERY happy with the performance and accuracy of smoothie with this new interpolation method, in fact my CNC milling machine actually ran better and smoother with smoothieboard using this new interpolation method than it has run on with ANY other controller I have tried.  I was going around 200mm diameter circles at 7000mm/min and still never stalled once, which is an amazing speed for that machine.  Just a note, with the fixed length method, my top speed of a 200mm diameter circle was only 2400mm/min and it was noticeably rougher.   

I have sent a final [Pull Request](https://github.com/rene-aguirre/Smoothieware/pull/4) back to rene-aguirre which adds the mm_max_arc_error variable to the sample config files and also sets the defaults as well.   

@rene-aguirre Thank you for getting this started! It would have taken me a very long time to figure out everything.  Your help on this is very much appreciated!   I am happy with the final pull request I sent back to you, you can go ahead and send this upstream.

wolfmanjm, 2016-06-13:
Actually the problem with performance of segmented things (lines in deltas and arcs is that when a segment is so short that the steps are finished before the next segment can be added and planned, the planner queue never has more than one entry which means the planner is forced to decelerate all new segments to zero as the exit speed. Speeding up the planning would help, but I don't think the arc calculations are the bottleneck here. I think the move needs to take more than about 50ms to execute to allow the planner queue to fill up. I think the reason this new ting works better is that the segments are longer therefor take longer to execute, giving the system time to fill and plan the rest of the segments.
Looking forward to the PR on this one.

Zaaphod, 2016-06-13:
@wolfmanjm I have some test results that seem to confirm what you are saying.  One thing I noticed is that if I accelerate faster, I can run faster with the tiny segments, but slow accelerations can't run as fast....  Since slower accelerations run slower, but due to the slower speed there should be more time for calculations,  I don't think there is any calculation bottleneck at all nor do I think there is a problem adding new segments to the queue fast enough.    
I think it comes down to this.. If your queue is full of tiny segments, the maximum speed you can run, is the speed at which you could decelerate to a full stop by the end of the queue... because you don't know what's coming as the next queue entry.. it could be a new line segment pretty much in line with the direction you are going, but it could also be a sudden direction reversal, or a sudden right angle, either of these would require a full stop before the new segment.   So Smoothie is handling things perfectly, by not assuming anything about future queue entries, it only allows enough speed to be able to decelerate to a full stop by the end of the queue.  When the segments are very small and a slow acceleration, you need to go slower to be able to stop by the end of the queue, if your acceleration is faster, then you can go faster and still stop by the end of the queue.  with long segments you can get it going pretty fast, in fact full speed on my machines, because you can always get to a full stop by the end of the queue.    

This also explains why my machine runs noticeably rougher on short segments, as new segment is added, it now has a little longer distance to be able to achieve a full stop, but as that segment is completed and just before the new segment is added, the total distance is shorter so it has to decelerate a little so that it could stop if the new queue entry requires it, then once the segment is completed, and a new segment is added to the queue, it now has the longer distance to go again, so it accelerates..so the short segments cause this acceleration / deceleration oscillation cycle because the segment length is so short it needs to always be prepared to stop.   As long as you could stop somewhere before the end of the queue with the longer segments, it can go ahead and accelerate to full speed, because it can see far enough ahead that if and... or should I say, WHEN then new segment requires a full stop, it will have enough distance to make a smooth controlled deceleration.

So.. what this all means for me,  is that smoothie is doing EXACTLY what I wanted it to do.    I have problems with my old controllers where if there is a small radius at the end of a long line, it tries to maintain a constant velocity around that little radius and of course there is not enough distance to slow down the one axis nor to speed up the other axis during this radius, so I'll have stall issues.  So I have the situation where I want to run a program cutting some foam or something at 40,000mm/min but I have to run the entire thing at 6,000mm/min so I can guarantee going around that little stupid radius safely.  The problem is the old software does not consider the acceleration of just what the individual axes will see and it does not compensate for over-running the acceleration capability of the machine.  I also can have problems with angles as well.. if I'm going say up the Y axis fast, and then I have a 30 degree angle, the old software would not consider the acceleration requirements of the X axis and try to just bring it up while the Y slowed down just enough so the new combined movement was the required federate... but that was not slow enough for the X by itself to accelerate from a dead stop to the target speed... and stall..... ugh  My previous software did not allow F commands during the queue, and the queue also was linear, not circular, so I could read in 125 lines, run those.. stop, read in another 125 lines, stop... etc... yea, it's 80's technology... but then again.. it was the 80's :)    The old queue limitations are where this interpolation thing originally came from, We did not have to limit the speed to be able to stop at the end of the queue... all queues were required to stop after 125 entries... so trying to run in little short line segments would take forever, especially since most of the machines I've built have tables 3meters long on them.

Now that I have my arcs running fast and smooth with this new method, I can move on to my next series of tests to see if smoothie will in fact handle problem areas correctly.  It seems that smoothie is always taking into account that it might need to stop at then end of the queue and if so, I'm hoping it will handle the problem areas better than the old motion control.   At least with smoothie, it seems that I can stick in an F command in the middle of a contour and force a slowdown, but I'm not sure I even need to bother.. my quick preliminary tests seem to show smoothie will take care of these problems for me.

wolfmanjm, 2016-06-13:
basically yes, it will merge a queue full of moves into a single trapezoid, the final entry has an exit speed of zero always. Any new entry forces recalculation of the trapezoid for  the entire queue. This is what the planner does, The problem is when the queue cannot fill up because the segments take less time to execute than it does to receive and process them.

triffid, 2016-06-14:
There's also a conundrum when the moves are so short that it would take the
entire length of the queue to decelerate to zero, when this happens there's
a race condition between adding new segments and recalculating the
trapezoid, and finishing the current segment, leading to rough running as
the head constantly accelerates and decelerates around the max speed from
which we could decelerate.

This isn't an issue of segment processing time, rather it hinges entirely
on max speed (deceleration) < commanded speed, and the race between adding
segments and completing segments.

Note that a segment that's currently being executed can't have any of its
parameters changed, so if it starts immediately before trapezoid
recalculation, its exit speed will be lower than expected because the
planner can't update it. The following move will then have acceleration
back to the critical speed, and this cycle can and will repeat as long as
being able to decelerate by the end of the queue is the limiting factor on
segment velocity.

I don't believe this is "fixable" and I'm not sure we should try, a little
rough running is unavoidable in this situation without making the queue
longer, and smoothie doesn't have enough ram to extend the queue.
Furthermore, an extended queue would simply cause this to happen at higher
speeds and/or smaller segment sizes, it doesn't "solve" it at all.

Zaaphod, 2016-06-14:
That explains what I think is happening when trying to run at high speeds through tiny segments.  I agree that this is not fixable with smoothie, because it cannot know what future elements will be added. 
I have two thoughts here.
One is, the host computer issuing commands to smoothie could perhaps pre-process the entire file and put in F commands to speeds slightly slower than the critical speed where we get these oscillations.  I could test if running just below the critical speed would run smoother.  To have the host pre-process the file I would need to know how many segments the queue holds, and the deceleration rate does planner_queue_size  = 32 mean 32 line segments?

I have another idea on how maybe smoothie could take care of this....  
If instead of always bringing it back to the critical speed which would allow deceleration by the end of the queue, could there perhaps be some kind of check to only adjust the speed faster if it is going to be an increase greater than say 5%?  So you always must decrease the speed immediately to the speed at which you could decelerate to zero by the end of the queue, because you never know when the stop, direction reversal, or sharp angle will be introduced,  but increase it only if it will be greater than a certain %.. this would make it go to the lower speed in the oscillation cycle and stay at a the slightly lower, but constant speed until it perhaps gets a long line, where it would then be greater than than the set % so it would accelerate, or an event that requires a stop in which case it would decelerate to a stop.   It would run just slightly slower because it would be at the slower speed all the time, but it would be more consistent and smoother.    

Any thoughts on this concept?

triffid, 2016-06-15:
Neither idea will work.

The first idea requires that the host knows smoothie's acceleration
settings and length of planner queue, and also requires that smoothie
absolutely trust the host to do the math right, and also trust the host to
never have a single moment of lag long enough for the queue to drain. We're
simply not willing to put that much trust in the host, and it kinda defeats
the purpose of doing acceleration planning in the firmware in the first
place.

The second idea simply shifts the critical speed slightly, and I think
would go completely pear shaped if the length of the segments is changing
as this would radically alter the 'critical' speed from one segment to the
next

wolfmanjm, 2016-07-02:
fixed in PR #944

closed 2016-07-02 by wolfmanjm