Smoothieware/Smoothieware #930: upload_command via SerialConsole crash smoothie

zhulin1573, 2016-05-26:
when  use console command (upload) to send a file to SD Card via uart (RX/TX  P0.3/P0.2). smoothie crashes.  But  it is  working properly when use USBSerial. 
The possible reason for the problem is:    when use uart (RX/TX  P0.3/P0.2) to send upload command, This method（ready()） has not been implemented in the SerialConsole.cpp or Serial.cpp.
So stream->ready() will work abnormally. it is can not call  THEKERNEL->call_event(ON_IDLE); in the while(uploading){}. then it is crashed.
Can someone fix it?

wolfmanjm, 2016-05-26:
This is correct it is ONLY supported in the USB serial. If you want it fixed I am afraid you will need to do it yourself. Actually it is not a bug, it is by design as the serial uart is not fully supported for 100% use, and was never intended to be.
I suspect if you look hard enough at what needs to be done to support it you will find it relatively easy to implement a solution. However I have no intention of doing so, very sorry about that.

closed 2016-05-26 by wolfmanjm

wolfmanjm, 2016-05-26:
If you used M28 to upload, and finish with M29 it will work for both USB serial and uart serial. So use that, not the upload command which was written for a very specific purpose, and not the one you are trying to use it for.

dcousens, 2016-05-27:
Isn't any crash a bug?

wolfmanjm, 2016-05-27:
It isn't crashng it is just getc doesn;t work as expected on the uart serial.
Feel free to fix it!