Smoothieware/Smoothieware #931: Hotend happily keeps heating up even though wrong thermistor defined

dintid, 2016-05-27:
Noticed the problem as I've been transitioning to using Thermocouple and was at the end of testing.

When everything is set up to use Hotend as normal, but the Thermistor pin is not defined as the one actually in the hotend.

Problem is that command to heat up hotend results in the hotend just keeps heating untill it melts or is manually turned off.

Firmware keeps reading the defined thermistor and thinks it still needs to heat up.

I thought there was some algorithm checking every few seconds to see if a heating had any effect?
Meaning checking if temperature was at the expected level. If temperature stays at 23c after heating at full power for 5 seconds, then something is wrong.

If there is such a feature, it isn't working.
If there is no such thing, I really hope and urge someone to implement it under basic safety features.

Having 2 thermistors is not a solution to this problem.

triffid, 2016-05-27:
That isn't unique to smoothie, almost all heater controllers targeting a
massive range of applications will do this.

Firmware can't be expected to automagically compensate for user error.

Having said that, I believe such a feature is being developed for smoothie
at the moment - I wrote it for Teacup firmware many years ago
<https://github.com/Traumflug/Teacup_Firmware/commit/2b1fbe38db5f> but
couldn't find anyone interested in testing it so it languished.

dintid, 2016-05-27:
Marlin would catch it.

And yes, in THIS particular instance it was user error, but this check is an absolut MUST have. 
I only used this example to showcase how I noticed the problem.

If this safety feature is properly implemented It would catch a whole host of problems I can think off. 
A loose thermistor is one such.

triffid, 2016-05-27:
Marlin won't catch it unless specially configured to do so. A default
installation would burn the hotend without blinking.

And if it's such an important feature, why have I only seen like 3-4 people
talking about experimenting with it over the past 6 years? I thought it
would be quite important too, but apparently it's simply not..

Feel free to find the smoothie users talking about heater sanity checks and
runaway detection, I believe there's an open issue which you could watch or
contribute to, and test any commits that folks post.

arthurwolf, 2016-05-27:
There is an open issue about this here : https://github.com/Smoothieware/Smoothieware/pull/782 and work was done on it recently, I believe it's not far from being ready to merge.
Like Marlin, it'll be active only if enabled though, it's impossible to have defaults that will fit everybody, and it's likely users need to configure other things before enabling it, or it'll just trigger for no reason while configuring/tuning the machine.

closed 2016-05-27 by arthurwolf

dintid, 2016-05-29:
@triffid  why do you keep talking about Marlin and what everyone else is or is not doing?
Saying it's a needless option because no one else has it is not an argument I'll accept.

I find it very much needed and see it as a critical basic safe-fail. So basic that I thought it might be a bug that made it not working though it surely must have been made.
Guess I was wrong.

@arthurwolf  sounds good :) I don't much mind if it is on default or not, as long as the option is there.

People not wanting it in there are most likely the people allready very familiar with the Smoothieware and capeable of digging through the technical documentation, so I'd still suggest putting it in the default config file to make it more readily available for the future general average user.

Lots of safety features are enabled by default in Marlin.. most annoyingly the "cold extrusion prevention", but really, I don't see why we even need to compare?

I'd personally rather have too much enabled by default than nothing.. then I could go in and disable it with a note to self on remembering to turn it on.. or my favorite: just have it preconfigured in default config file but commented out.

:)

triffid, 2016-05-29:
I never said it's needless, and I mentioned marlin because you claimed
marlin would do the right thing but it won't without special
configuration.. In fact like I said earlier I wrote this feature myself
several years ago for another firmware, but couldn't find anyone who cared
enough to test it, and eventually concluded that most people don't want it
despite how sensible it is

qharley, 2016-05-29:
Debating how useful it would be will not implement it. Do it and see if
others want it. If they don't (and sometimes they wont) you at least have
it for yourself.

If I have time I will look into this myself. My hotends goes to 650degrees
C if it keeps heating. It turns PLA into smoke...

arthurwolf, 2016-05-29:
> Debating how useful it would be will not implement it. Do it and see if
> others want it. If they don't (and sometimes they wont) you at least have
> it for yourself.
>
> If I have time I will look into this myself. My hotends goes to 650degrees
> C if it keeps heating. It turns PLA into smoke...

Completely unrelated to the current github issue, but I strongly recommend
you change heaters :) That sounds dangerous, and it sounds like it'd be way
safer with a less powerful heater.

>
>
> > I never said it's needless, and I mentioned marlin because you claimed
> > marlin would do the right thing but it won't without special
> > configuration.. In fact like I said earlier I wrote this feature myself
> > several years ago for another firmware, but couldn't find anyone who
> cared
> > enough to test it, and eventually concluded that most people don't want
> it
> > despite how sensible it is
> >
> > > @triffid <https://github.com/triffid> why do you keep talking about
> > > Marlin and what everyone else is or is not doing?
> > > Saying it's a needless option because no one else has it is not an
> > > argument I'll accept.
> > >
> > > I find it very much needed and see it as a critical basic safe-fail. So
> > > basic that I thought it might be a bug that made it not working though
> it
> > > surely must have been made.
> > > Guess I was wrong.
> > >
> > > @arthurwolf <https://github.com/arthurwolf> sounds good :) I don't
> much
> > > mind if it is on default or not, as long as the option is there.
> > >
> > > People not wanting it in there are most likely the people allready very
> > > familiar with the Smoothieware and capeable of digging through the
> > > technical documentation, so I'd still suggest putting it in the default
> > > config file to make it more readily available for the future general
> > > average user.
> > >
> > > Lots of safety features are enabled by default in Marlin.. most
> > annoyingly
> > > the "cold extrusion prevention", but really, I don't see why we even
> need
> > > to compare?
> > >
> > > I'd personally rather have too much enabled by default than nothing..
> > then
> > > I could go in and disable it with a note to self on remembering to turn
> > it
> > > on.. or my favorite: just have it preconfigured in default config file
> > but
> > > commented out.
> > >
> > > :)
> > >
> > > <
> >
> https://github.com/Smoothieware/Smoothieware/issues/931#issuecomment-222354350
> > >,
> > > <
> >
> > >
> > >
> >
> > <
> https://github.com/Smoothieware/Smoothieware/issues/931#issuecomment-222355936
> >,
> > <

triffid, 2016-05-29:
I drop 45w into my nozzle and it'll definitely smoke anything if left on,
however I find that having so much power is very useful for effortlessly
maintaining temperature in a wide range of conditions including fan running
full speed in winter. I think sanity checking the heater is a grand idea

qharley, 2016-05-29:
More power does give better control, and users generally do not enjoy
waiting.
I had to see what happens if a fatal error would occur.  At least it does
not turn into a fireball.  You just get a very elaborate caramel smoke
machine...

Still not what I had in mind for my machines, so it would be better if we
can prevent something like this to happen altogether.