Smoothieware/Smoothieware #936: Minimum actuator rate is hardcoded

dutchrazor, 2016-06-03:
I saw something interesting in StepperMotor.cpp:

"
// in steps/sec the default minimum speed (was 20steps/sec hardcoded)
float StepperMotor::default_minimum_actuator_rate=  20.0F;
"

I set it to 1.0 since I wanted to move a high step-rate axis slowly, and it seems to work. Is there still a reason this is hardcoded?

Best regards,

Fabien

triffid, 2016-06-03:
Yes, some slicers issue moves with F0 and we get frequent complaints that
smoothie locks up as a result. Feel free to keep this change in your
private fork and rebase as necessary if it works well for you :)

wolfmanjm, 2016-06-03:
yea but this is no longer used in the upcoming tick generator rewrite.

closed 2016-06-03 by wolfmanjm

dutchrazor, 2016-06-03:
Ah thanks for the explanation. Is there any ETA on the tick generator rewrite?

I stumbled into this because I hooked up a printhead which required longer pulses than the normal stepper motor pulses (longer than desired for the XYZ axis), so I wrote in a little hack to unstep the pulse of one of the extruder axes every x'th unstep pulse.