Smoothieware/Smoothieware #938: Movements are stopping short then being corrected.

Zaaphod, 2016-06-06:
I've been having this issue since I first started trying to use smoothieboard, and it's preventing me from using it at all,  So I'm submitting this as a bug report here to hopefully get to the bottom of it and find a solution.  

The problem is smoothie is decelerating and stopping short of it's intended position, then correcting the error.  I do not know where or how this is happening, I just know that it is defiantly what is going on. 
I decided to buy a logic analyzer to eliminate all possibilities of hardware compatibility issues, I'm using the latest edge firmware (as of 6/6/2016) and the sample 3d printer config file.  I'm going to attempt to illustrate this issue.

I'm using pronterface to control smoothie,  and only issuing a single move command to demonstrate this issue. I am using 4 channels of a logic analyzer connected to X-Step, X-Dir, Y-Step, and Y-Dir. 

Below is a screenshot of the Acceleration of the following command: G0 X1250 Y750 F30000  This is a completely valid command on my machines.  You can see this is a perfect, nice smooth acceleration, it takes 4.1538ms between the first and second pulse on the X axis, and each pulse is closer than the last
[image]

now lets look at the Deceleration of the same command.. scrolling near the end of the sample,  here we see what appears to be a perfect, nice smooth deceleration 4.863ms occurring at just after 4.9s... now this would be great, but unfortunately, It's only completed 99,867 steps on X, but it should have been 100,000 exactly since I am using the default of 80steps per mm and I told it to go 1250 steps... so it's 133 steps short (this is over 1/8" on my machine)
[image]

So if we scroll some more, we see that smoothie is correcting for this shortfall.  you can see my mark I made at just after 4.9s, and the correction steps to the right.   The steps are always at a very slow frequency in comparison to the movement speed.  Also notice that both X and Y completed at about the same time, and started the correction pulses around the same time.
[image]

Since the time between the first and second step was 4.1538ms and the time between the 99,866th and 99,867th was 4.863, this is way outside of what would be considered the end speed.  also notice the correction steps are NOT decelerating, the first one and the last one are all exactly 39ms  The next two screen shots show the first and last of these correction pulses
[image]
[image]
now here is a screenshot showing the whole thing.... notice my mark at 4.9seconds, these corrections are continuing for over 5 full seconds on the X axis, adding 133 steps on X, but Y stopped much before then.. 
[image]

so this presents a problem... because the corrections are NOT completing at the same time, it's not positioning the machine correctly.  Here is another example.. doing a G0 X0 Y700 from the previous move which should move X-100000 and Y-4000 steps: you can see here that Y reached it's final correct location at about 4.6seconds, while X continued this until 9.6seconds... which means Y reached it's location more than 4 seconds before X and the shape is also wrong because X moved over 1/8" after Y stopped... this is the real problem.. it doesn't matter if it happens so fast  you can't see it,  it's still happening, even on microstep drives.. it's mispositioning then correcting.   After these adjustment steps complete X has output exactly 100000 and Y has output exactly 4000 steps.
[image]

I have been unable to stop this from happening.. There is something being calculated wrong that is causing the movement to ALWAYS come up short, then require a correction. Those 133 steps at the end of the X movement should have happened during the rapid part of pulse train, not after the end of the deceleration.  It seems to require a longer correction for longer moves and a shorter correction for short moves, and it seems that the acceleration has something to do with the magnitude of the error, with fast accelerations requiring less adjustment than slow accelerations.  With the super fast accelerations in the default config file, it's still doing this, but there are only 9 adjustments, which you would never see on a microstep drive, the problem is, I cannot possibly accelerate my machines that fast... and just because you don't see it, doesn't mean it's working right.  There is something fundamental causing this to happen.

I'm happy to assist in any way I can in diagnosing and trying to resolve this issue.  I'll be happy to run tests and provide logic analysis, and I can even try to help figure out what's happening in the code, but I'm not familiar with this project enough to even know where to look for things like step generation and acceleration control.   If someone can point me in the right direction of the relevant code, I'll be happy to see if I can figure out what's happening.    Perhaps I can learn how to trace through the code during execution and put in a few breakpoints, analyze the results of some variables to figure out what's happening

For further reading on this issue,  please see my post [here]( https://groups.google.com/forum/#!topic/smoothieware-support/nZZsi_qJQCg)

and I also have videos here and here of the machine demonstrating this issue.
[Straight line test](https://www.youtube.com/watch?v=WEIu9AZFOOs)
[Rectangle program test](https://www.youtube.com/watch?v=eRnPhxIt0Tk)

arthurwolf, 2016-06-06:
This is extremely weird.
I have never seen a Smoothieboard do that, and never had any reports of
something like that.
We definitely need to figure out what's going on here.

Two questions just to have that out of the way before going further : are
you using the edge branch, and are you using the latest version of it ?

wolfmanjm, 2016-06-06:
FYI I believe he is not using a smoothieboard. and I also do not see anything like this. There are usually one or two steps at the end which can be slow due to the acceleration ticker being too slow (this is fixed in the new stepticker code). But I have never seen so many delayed pulses. I suspect there is something odd in the config or other issue that causes this. it is definitively not something many other people see. This may well be an artifact of the non smoothieboard and different drivers.

wolfmanjm, 2016-06-06:
I am closing this as it is a one off issue that only affects this user, I cannot reproduce it and neither can anyone else. I also think this will be resolved by the upcoming stepticker release.
I have spent a lot of time trying to reproduce this issue with no luck at all. I have tried 1/2 step similar steps/mm and speeds.
Also there is no "catch up" code in the current stepticker, so that is not what is happening here.
I think what happens for whatever reason is that the deceleration falls below the minimum 20steps/sec that is hard coded, so the following steps are issued at that rate until they are completed.
As I cannot reproduce this I cannot confirm that, and so cannot fix it.

closed 2016-06-06 by wolfmanjm

Zaaphod, 2016-06-06:
I am using smootieboard 5axis.  it says v1.0a in the corner. I updated the firmware today
>>> @version
SENDING:version
Build version: edge-564f4d4, Build date: Jun  3 2016 04:05:31, MCU: LPC1769, System Clock: 120MHz
I'm using the 3d printer config file found here: 
https://github.com/Smoothieware/Smoothieware/raw/edge/ConfigSamples/Smoothieboard/config
the only change I made to this config was to set acceleration on line 18 to 250

I DO get these adjustment pulses at faster accelerations as well, and I can do an analysis to show how acceleration values affect this issues if it would be helpful.

I have it hooked up to only USB and wires from ST1, DIR1, ST2, and DIR2 going to logic analyzer.. getting ground for the logic analyzer from one of the limit switch ground pins.

Just a note,  I have also tried setting steps per mm to 1, thinking that if there is some error being caused by converting to and from mm, that having it set to 1 step per mm would eliminate any conversion issue.. so I gave commands in numbers of steps I wished.  I ended up with the same results.

I will be more than happy to test any configuration that would be helpful and post the results.

wolfmanjm, 2016-06-06:
There are no such thing as adjustment pulses, so the title and suggestion mischaracterize this issue and confuse things.
I think this is some odd artifact caused by your very unusual setup. I also thing this may be solved with the new stepticker code.
If you want you can run the same test with the WIP of the the stepticker code. NOTE that the code currently only works for single issued commands like you are doing above so should work as a good demonstration. do not try to stream gcode to this version of the code.
Let me know if it solves the issue, if it does then case closed :)

you will need to grab this fork and compile it.   https://github.com/wolfmanjm/Smoothie/tree/feature/acceleration-per-tick
you will also need to copy this into a file named src/default_excludes.mk

```
export CNC=1
export NONETWORK=1
export EXCLUDE_MODULES = tools/laser tools/filamentdetector tools/scaracal tools/temperaturecontrol tools/extruder tools/zprobe tools/endstops
```

As this modules are not yet rewritten so will not compile. Note there are no endstops or homing available in this branch.

Zaaphod, 2016-06-06:
I'll be more than happy to test the new code and will report back on my findings.   I'm hopeful that it will help because I would really like to use smoothie board, as I really like everything else about it..  I just can't figure this issue out.

I don't have an unusual setup at all..  It's just a cnc router with 1/2 step drives that I would like to accelerate at 250 mm/s/s.  This is why I'm testing it with the stock config file and only changing the acceleration.. I'm not even changing the steps per mm, just to make sure.... so literally changing no other numbers.   

The thing that is odd is it is doing a perfect deceleration, but it's doing it too soon.  whatever is causing this is always happening, 

I just did a test with a completely unchanged config file with the default 4000 acceleration,   This is from a completely un-touched config file and giving only a single command of G0 X1000 Y50 F15000 and not specifying the federate whatever causes those pulses out past my mark which is the point at which the timing matches the initial two pulses is demonstrating this issue..  It always happens, I can't prevent it no matter what I do.  Even though you would not see these slow steps on a microstep driver, they are still causing an unnecessarily slow delay before the next movement can happen.

[image]

arthurwolf, 2016-06-06:
> I'll be more than happy to test the new code and will report back on my
> findings. I'm hopeful that it will help because I would really like to use
> smoothie board, as I really like everything else [...]

In the Smoothie community, that qualifies as "unusual". I believe most CNC
routers used in the community right now use microstepping of 1/16 or
higher, and accelerate significantly faster than 250mm/s ( though not
always ).
Not that there's anything wrong with your setup, but it's definitely
unusual, and something that has likely not been tested much before.

> This is why I'm testing it with the stock config file and only changing
> the acceleration.. I'm not even changing the steps per mm, just to make
> sure.... so literally changing no other numbers.
>
> The [...]

wolfmanjm, 2016-06-06:
have you made sure you do not have a co fig-override? post the results of M503

Zaaphod, 2016-06-06:
m503
SENDING:M503
; No config override
;Steps per unit:
M92 X80.00000 Y80.00000 Z1600.00000
;Acceleration mm/sec^2:
M204 S4000.00000 Z0.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max feedrates in mm/sec, XYZ cartesian, ABC actuator:
M203 X500.00000 Y500.00000 Z5.00000 A1250.00000 B1250.00000 C62.00000
;WCS settings
G54
;Digipot Motor currents:
M907 X1.50000 Y1.50000 Z1.50000 E1.50000
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00
;E Steps per mm:
M92 E140.0000 P57988
;E Filament diameter:
M200 D0.0000 P57988
;E retract length, feedrate:
M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
;E retract recover length, feedrate:
M208 S0.0000 F480.0000 P57988
;E acceleration mm/secÂ²:
M204 E500.0000 P57988
;E max feed rate mm/sec:
M203 E50.0000 P57988
;PID settings:
M301 S0 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S0 P300.0000
;PID settings:
M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
;Max temperature setting:
M143 S1 P300.0000

wolfmanjm, 2016-06-06:
just to make sure... but you do not have the external drivers hooked up here right?

Zaaphod, 2016-06-06:
no I do not,  I am bench testing with nothing other than the usb cable and the logic analyzer.    

I have downloaded the firmware from the link and I'm trying to follow the directions here: 
http://smoothieware.org/compiling-smoothie
to compile it,  but  I'm not finding win_install.cmd on the smoothie drive.  can I get some advice on finding this please?

wolfmanjm, 2016-06-06:
it is in the repo. I can build a firmware for you hold on.

wolfmanjm, 2016-06-06:
Unzip this and rename the mian.bin file to firmware.bin and flash it.

[firmware.zip](https://github.com/Smoothieware/Smoothieware/files/301365/firmware.zip)

Zaaphod, 2016-06-06:
Great news!!  
I can't find any evidence of this issue with the test firmware!  Here's my results for your review,

Stock config with acceleration set to 250.   G0 X1250 Y50 F30000  beginning of test:
[image]
End of test:
[image]

Accel set to 50.   G0 X1250 Y50 F50000  Beginning of test: width of 1st and 2nd pulses:9.297
[image]
End of test - pulse width between 99,999th and 100,000th pulses:14.16ms
[image]

I really appreciate your help with this and I'm glad the solution is in the works.  I have a lot of work to do to make my program send commands over serial interface to smoothie, so that will keep me busy while this new firmware is developed.

qharley, 2016-06-07:
What needs to be done on the scaracal module to get it in line.  I am working on that module at the moment, and I might as well help to get this release out