Smoothieware/Smoothieware #939: Triggering an event from a temperature reading

DarklyLabs, 2016-06-08:
I am trying to trigger few events from a temperature sensor reading. My goal is to issue a few GCode commands when a thermistor reaches a certain temperature.

I can see the temperature changing with the M105 command but I am having problems with triggering the event. 
I have attached the config code I am using:

# define roomTemp temperaturecontrol
temperature_control.roomTemp.enable             true             
temperature_control.roomTemp.thermistor_pin     0.23            
temperature_control.roomTemp.heater_pin         nc               
temperature_control.roomTemp.thermistor         EPCOS100K        
temperature_control.roomTemp.designator         T                
temperature_control.roomTemp.readings_per_second 10

switch.toohot.enable                                  true            
switch.toohot.input_on_command             G28X0Y0             
switch.toohot.input_off_command             G0X100Y100         

temperatureswitch.roomTemp.enable               true            
temperatureswitch.roomTemp.switch               toohot     
temperatureswitch.roomTemp.designator           T               
temperatureswitch.roomTemp.threshold_temp       22.0  
temperatureswitch.roomTemp.heatup_poll          2    
temperatureswitch.roomTemp.cooldown_poll        5   

Any ideas or suggestions on why this is not working?

Thank you,
Domenic

wolfmanjm, 2016-06-08:
Please **DO NOT** use issues for asking questions. There are plenty of other forums to ask this type of question... the Forum, google groups or IRC.

closed 2016-06-08 by wolfmanjm

dcousens, 2016-06-08:
This is *the* place for Smoothieware,  and will forever be where people who search for anything related to the above,  will find themselves.
Why not just tag it as a question?

DarklyLabs, 2016-06-08:
I am unfamiliar with the issue / question protocols for this project. I will take Jim's advise and search for the answer elsewhere.

wolfmanjm, 2016-06-08:
http://smoothieware.org/ look at the community link at the top