Smoothieware/Smoothieware #943: G32 ignores Zprobe during moves, leading to possible head crashes

dumle29, 2016-06-11:
Hey there. I've recently built a delta, and decided I'd calibrate radius and endstop offsets with the G32 command, so I didn't care for the accuracy of the endstops in relation to each other, and I only set a rough radius in the config.

These things obviously led to a highly concave motion by the effector, which I hadn't thought would be an issue with the G32 command, but as it turns out, it was.

After probing the bed center, and homing, the effector moved out to a corner to probe, but on the way, the effector dug into the bed, and broke. During this, the probe obviously triggered, but the algorythm didn't care.

So a feature request here, is to use the probe even when doing X/Y moves, and either

1. Cancel the whole process if the sensor is activated during an X/Y move

or

2. Adjust the Z probe height until the desired X/Y movement can be completed.

wolfmanjm, 2016-06-11:
Yes this is by design, the probe is only looked at when it is actually probing not moving. This is not likely to be changed as it is how I believe it needs to be done.

There are several reasons for this, but the main one is there is currently no way to monitor the probe while it is moving, only when it probing. Also noise can be an issue if the probe would stop non movements.
If you feel very strongly about this then you are welcome to attempt to add this "feature" but it needs to be optional and off by default.

wolfmanjm, 2016-06-11:
In addition there are options that you need to set to avoid this situation, which I think are well documented in the wiki. Simply set Zprobe height to a value higher that the bed at all positions.
Adding complicated features to avoid proper configuration is not a good design choice.

closed 2016-06-11 by wolfmanjm