Smoothieware/Smoothieware #945: Spindle does not work on P1.23 with Network enabled

dutchrazor, 2016-06-17:
I had the following config which did not work:

-Spindle module enabled, with PWM pin on P1.23
-Pronterface connected via Telnet

It seems that enabling telnet interferes with the PWM out capability in 1.23. It does work fine on pins in the 2.x range.

wolfmanjm, 2016-06-17:
This did not conform to the issue submission guidelines.  https://github.com/Smoothieware/Smoothieware/blob/edge/HowToFileIssues.md

dutchrazor, 2016-06-17:
You don't want to fix it , fine by me.

arthurwolf, 2016-06-17:
That's not what that meant.
We recently added a set of rules people should follow when they create new
issues, and he pointed you at those, that's it.

If there is a problem in Smoothie, we want to fix it, make no mistake.

Thanks.

arthurwolf, 2016-06-17:
@dutchrazor Just to be clear : closing the issue is just a way for us to sort issues. 
If you follow the issue posting rules, it will be re-opened no problem.

dutchrazor, 2016-06-17:
I hope you understand the tone may be confusing, especially since the guidelines are new.

Let's try again :) 

What were you doing?

Enable the spindle via M3 S200 (set Spindle to 200RPM) while connected to Smoothieboard via telnet.
-Spindle PWM is connected to 1.23

What did you expect to happen?

Spindle should go to 200 RPM.
-> Expected PWM output to go to 1.0, spindle to accelerate

What happened instead?

-PWM output from M957 indeed goes to 1.0
-PWM output pin 1.23 stays low

Branch & Commit or Version of Smoothieware
Tried both Edge and Master from http://triffid-hunter.no-ip.info/Smoothie.html. 

Hardware model you are using (Make, and version)
Smoothie board 5X V1a

Host and Operating System running the Host
Windows 7, Pronterface

Link to and console logs
Will upload Monday. I take it this means a log from Pronterface communication?

wolfmanjm, 2016-06-17:
Thank you, Any logs on pronterface if relevant, (Showing what you are sending and results), but also config file and result of M503 is also required.

wolfmanjm, 2016-07-07:
Closing as the requested additional information was not given.

closed 2016-07-07 by wolfmanjm