Smoothieware/Smoothieware #969: laser_module_maximum_power config setting doesn't do anything

cilynx, 2016-07-07:
I need to limit my maximum duty cycle to 0.75 and setting the config value doesn't appear to have any effect.  Looking in [Laser.cpp](/Smoothieware/Smoothieware/blob/edge/src/modules/tools/laser/Laser.cpp), I'm also not seeing any logic that uses this setting.  Should there possibly be something like `if (requested_power > this->laser_maximum_power) requested_power = this->laser_maximum_power` around [line 141](/Smoothieware/Smoothieware/blob/edge/src/modules/tools/laser/Laser.cpp#L141) or am I missing some logic going on behind the scenes?

wolfmanjm, 2016-07-07:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/HowToFileIssues.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

closed 2016-07-07 by wolfmanjm

pelrun, 2016-07-07:
@cilynx, you appear to misunderstand how the power settings work. Your S values are always between 0 and 1 (unless you change `laser_module_maximum_s_value`), which is then automatically remapped to the range `laser_module_minimum_power` to `laser_module_maximum_power`. So if you've changed max power to 0.75, and you actually want to cut at that power, then you should be sending `S1`, *not* `S0.75`.

triffid, 2016-07-07:
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/laser/Laser.cpp#L160-L161
looking for this?

cilynx, 2016-07-07:
Thanks for the responses.

@triffid, yes, that's exactly what I was looking for, which is hilarious being that it's only a few lines later in a relatively small and straightforward file.  My search-fu was apparently pretty weak last night.

@pelrun, here's my scenario:  I'm testing with a scope hooked up to P3.25 and ground.  My initial laser-relevant config settings are:

`laser_module_enable                          true`
`laser_module_pin                             3.25`
`laser_module_pwm_period                      500`

(I'm setting the period higher because the signal will be feeding a pwm->voltage converter that expects 1-3kHz.)

When I do some test "cuts", the scope shows the pulse trains I would expect: S0 is always low, S0.5 is a 50% duty cycle, and S1 is always high.

Next, I update my config to limit max power like so:

`laser_module_enable                          true`
`laser_module_pin                             3.25`
`laser_module_pwm_period                      500`
`laser_module_maximum_power                   0.75`

Save the config, unmout the SD card, reboot the Smoothieboard and do some more test "cuts".  The results on the scope are exactly the same as before.  When I send S1, I get a 100% duty cycle.  When I send S0.5, I get a 50% duty cycle.

I know the config change is saving because if I change other settings as well, they take effect as expected.  Reading over the logic with fresher eyes this morning, everything does indeed look sound and the behavior I'm seeing doesn't make any sense.  I'll do some more poking around after work and see if I can gather any more usable datapoints or sort it myself since it does appear to be application specific.

arthurwolf, 2019-03-17:
Out of curiosity, did you ever figure this out ? This sounds like formatting the SD card might help, I've seen cases in the past with similar problems where it did.