Smoothieware/Smoothieware #977: Motion glitch or stumble

DavidMFlynn, 2016-07-09:
Some files are still causing motion problems.  When two line segments are nearly parallel to each other and one axis with a jog in between the motor moving mostly in a straight line bounces (acceleration goes too high) sometimes causing loss of reference.  I can change the part slightly and re-slice and the stumble will go away, but I shouldn’t have to.  Please fix this.

This code caused a stumble
G1 X129.351 Y109.931 E8.20025
G1 X129.399 Y109.026 E8.25830
G1 X129.453 Y108.683 E8.28052
G1 X129.453 Y63.764 E11.15839
G1 X128.977 Y63.228 F3000.000

this code does not. Mirrored part same segments.
G1 X95.800 Y64.081 F3000.000
G1 E0.70000 F900.00000
G1 X95.800 Y109.060 E3.58172 F1800.000
G1 X95.855 Y109.403 E3.60394
G1 X95.902 Y110.307 E3.66199
G1 X95.947 Y126.651 E4.70907

Just tried it with the latest version. It’s still there and maybe worse.  It lost reference on layer 2.  Last time it got to layer 23 before losing reference.

Here is the code that caused loss of reference:
G1 X129.555 Y126.307 E7.13709
G1 X129.306 Y126.274 E7.15315
G1 X129.351 Y109.931 E8.20023
G1 X129.399 Y109.026 E8.25828<
G1 X129.453 Y108.683 E8.28050<
G1 X129.453 Y63.764 E11.15837<

The motor stalled moving from Y109.931 to Y63.764.

[config.txt](https://github.com/Smoothieware/Smoothieware/files/355757/config.txt)

wolfmanjm, 2016-07-09:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/HowToFileIssues.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

closed 2016-07-09 by wolfmanjm

wolfmanjm, 2016-07-09:
FWIW I am told you have a leadscrew driven axis. I have a lot of experience with these. The skipping steps are not a firmware issue. You need to use more powerful stepper motors (I had to use triple stack nema17, but nema23 maybe needed).

You need to reduce the Junction deviation to 0.01 or lower. You need to reduce your speeds and your acceleration. and beware of resonance.

Nothing we can do in firmware would help you avoid this.

Also in your examples above (which are insufficient to do a decent analysis) there are missing speeds on the first set so what speed is it being asked to run at?

DavidMFlynn, 2016-07-09:
This is not a hardware problem a different brand controller works without any other changes.  The speed is 40mm/s and accel is 240mm/s/s. It runs smooth and nice most of the time.  I have run large complex multi-hour prints without a problem.  Only certain files have issues.  It is the particular GCode file that causes the issue.  The only way to make one of these bad files work is to set the speed so slow it can start in 1 step or increase the acceleration so that the short move in between the long moves has a different sign on the accel at transition.

BTW: I have been building and doing controller retrofits on machine tools for a long time.

Dave Flynn
Oxford V.U.E., Inc.
office (626) 256-6557
cell (626) 893-6654
dflynn@oxfordvue.com