Smoothieware/Smoothieware #978: Motion glitch or stumble

DavidMFlynn, 2016-07-09:
What were you doing? 3D Printing
What did you expect to happen? Part to print to completion without error.
What happened instead? Lost reference part way into layer 2.
Branch & Commit or Version of Smoothieware: Smoothie edge-70349
Hardware model you are using (Make, and version): Azteeg X5 Mini V2
Host and Operating System running the Host: stand-alone running from SD card
Link to running config and M503 results:
Send: M503
Recv: ; No config override
Recv: ;Steps per unit:
Recv: M92 X314.96060 Y314.96060 Z2519.68506
Recv: ;Acceleration mm/sec^2:
Recv: M204 S240.00000
Recv: ;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
Recv: M205 X0.05000 Z-1.00000 S0.00000
Recv: ;Max cartesian feedrates in mm/sec:
Recv: M203 X83.33334 Y83.33334 Z16.66667
Recv: ;Max actuator feedrates in mm/sec:
Recv: M203.1 X83.33334 Y83.33334 Z16.66667
Recv: ;WCS settings
Recv: G54
Recv: ;Digipot Motor currents:
Recv: M907 X1.00000 Y1.00000 Z1.20000 E0.60000
Recv: ;Home offset (mm):
Recv: M206 X0.00 Y0.00 Z0.00
Recv: ;E Steps per mm:
Recv: M92 E648.0000 P57988
Recv: ;E Filament diameter:
Recv: M200 D0.0000 P57988
Recv: ;E retract length, feedrate:
Recv: M207 S3.0000 F2700.0000 Z0.0000 Q6000.0000 P57988
Recv: ;E retract recover length, feedrate:
Recv: M208 S0.0000 F480.0000 P57988
Recv: ;E acceleration mm/sec��:
Recv: M204 E500.0000 P57988
Recv: ;E max feed rate mm/sec:
Recv: M203 E50.0000 P57988
Recv: ;PID settings:
Recv: M301 S0 P13.7000 I0.0970 D24.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S0 P300.0000
Recv: ;PID settings:
Recv: M301 S1 P10.0000 I0.3000 D200.0000 X255.0000 Y255
Recv: ;Max temperature setting:
Recv: M143 S1 P300.0000
Recv: 
Recv: ok

Some files are still causing motion problems. When two line segments are nearly parallel to each other and one axis with a jog in between the motor moving mostly in a straight line bounces (acceleration goes too high) sometimes causing loss of reference. I can change the part slightly and re-slice and the stumble will go away, but I shouldn’t have to. Please fix this.

This code caused a stumble
G1 X129.351 Y109.931 E8.20025
G1 X129.399 Y109.026 E8.25830
G1 X129.453 Y108.683 E8.28052
G1 X129.453 Y63.764 E11.15839
G1 X128.977 Y63.228 F3000.000

this code does not. Mirrored part same segments.
G1 X95.800 Y64.081 F3000.000
G1 E0.70000 F900.00000
G1 X95.800 Y109.060 E3.58172 F1800.000
G1 X95.855 Y109.403 E3.60394
G1 X95.902 Y110.307 E3.66199
G1 X95.947 Y126.651 E4.70907

Just tried it with the latest version. It’s still there and maybe worse. It lost reference on layer 2. Last time it got to layer 23 before losing reference.

Here is the code that caused loss of reference:
G1 X129.555 Y126.307 E7.13709
G1 X129.306 Y126.274 E7.15315
G1 X129.351 Y109.931 E8.20023
G1 X129.399 Y109.026 E8.25828<
G1 X129.453 Y108.683 E8.28050<
G1 X129.453 Y63.764 E11.15837<

The motor stalled moving from Y109.931 to Y63.764.

wolfmanjm, 2016-07-09:
As I said in your previous issue thi sis not a firmware issue IMO. Please read my comments there.

closed 2016-07-09 by wolfmanjm

DavidMFlynn, 2016-07-09:
You’re opinion doesn’t matter.  My logic analyzer does.  Fix your software.

Dave Flynn
Oxford V.U.E., Inc.
office (626) 256-6557
cell (626) 893-6654
dflynn@oxfordvue.com

arthurwolf, 2016-07-09:
Hey Dave.

The folks working on the Smoothie firmware are volunteers, giving you a lot
of work for free.
You really don't want to be saying things like "your opinion doesn't
matter", or to order things like "fix your software".

If you think you have good reasons why this is indeed caused by the
firmware, please provide more information if you have more, and explain in
detail why you think it is a firmware issue.

If a developer is not convinced it is caused by the firmware, they will not
try to fix something they do not believe is broken.

Cheers.

wolfmanjm, 2016-07-09:
That kind of attitude will get you nowhere. AFAIK there is no bug so nothing to fix.
Try setting junction deviation to 0.001. also try using better stepper motors. Skipped steps are due to steppers being insufficient.

Please go use the firmware you find works for you if you feel that way.