Smoothieware/Smoothieware #982: SCARA homing broken

qharley, 2016-07-11:
#### What were you doing?

    Testing new motion code: homing.

#### What did you expect to happen?

X, Y and Z actuator motors should all start moving towards home, halting at the home hall effect switch position, and then doing a homing retake together.

I suspect it has to do with the new way of using the arm-solution to move the actuators in stead of using raw homing.  Is there a way to still force it to home raw?

#### What happened instead?
Only the Z axis homed

#### Branch & Commit or Version of Smoothieware
Smoothieware Edge 000dccb

#### Hardware model you are using (Make, and version)

    Speedyboard V1,2 (100% smoothie compatible)

#### Host and Operating System running the Host

    Pronterface, linux x64

#### Link to and console logs

   no output.  Homing only
    
#### Link to running config and M503 results
https://gist.github.com/qharley/4e01ac488a07033a706a37a87455cf67
   
I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Copy-paste this template **completely**. Do not skip any lines!

It might happen that you are asked to provide a more thorough crash log which you can find out how to do [here](http://smoothieware.org/mri-debugging
)

wolfmanjm, 2016-07-11:
No there is no way to do homing the way it used to, it must be a planned move. I am afraid you will need to figure out how to do this. I suspect scara should not be treated as a delta in this case, but I have no way of knowing how to do this or what it needs to do. Look at how corexy works maybe that will give you some clues.

I'll close this as I cannot fix it. But please continue to use this to ask questions and I'll try to help as best I can. Thanks.

closed 2016-07-11 by wolfmanjm

wolfmanjm, 2016-07-11:
You should be able to home using a planned move, you may need to home XY then Z though. Depending on how the kinematics work it may not be possible to home all three at the same time.

I'd think this is more like corexy where XY are related and Z is separate, in which case you need to home all three axis separately as in the corexy.

qharley, 2016-07-11:
Is there a way for cartesian_to_actuator to see the current homing status?  In that case I could add something in the arm_solution that can counter the problem

qharley, 2016-07-11:
The issue is that the homing movement is a circular path around the driveshafts.  This is needed because the position of the head is not known when homing starts.

I could use a G3 move to do this come to think of it.  Would that be possible during homing?

wolfmanjm, 2016-07-11:
there is no way really for an arm solution to see homing status, and there is not a way I currently see to use a G3 to home.

Can't you simply home X Y Z separately?

If you explain exactly what the sequence needs to be I can try to see what solutions are possible.

An alternative is you can write a new module scarahoming to do what you need. There is a new call in steppermotor that does manual steps (see simpleshell test raw) so you could do a raw manual stepping and detect the endstops etc. this is the brute force approach.

wolfmanjm, 2016-07-11:
so all you need to do is issue a movement that moves both your actuators in the direction of the homing switch. each actuator will stop when it hits its respective endstop, the actuator position is then the home position. This works for delta and rotary delta so I think it should work for scara. You will have to home Z separately though.

qharley, 2016-07-11:
This is what needs to happen in theory, but there is a singularity point in the middle of on side of the bed.  There are points from where the home position is not a straight line in cartesian space.

Thanks for the hints so far.  Will make it work...