Smoothieware/Smoothieware #985: Very fast steps on only the first move attempted after a halt and reset by M999

Zaaphod, 2016-07-17:
#### What were you doing?
Manually moving an axis to get to a position where home could be initiated safely after recovering from a halt state with M999.  

#### What did you expect to happen?
The machine to move at a normal speed, and distance with reference to where it thinks it is, even if that is wrong due to the halt. I understand fully that axis position is no longer reliable after the halt.  this does not affect a relative movement, and sometimes you need to move the machine before you can home, and some machines simply cannot be move by hand.   Example... if  you have a bullnose bit engaged in and undercutting material or a keyhole cutter.   You can't move Z until you get the bit out from under the material in a direction that is clear

#### What happened instead?
The axis attempted to move but stalled because it was running SUPER fast, so fast my stepper motors sounded like they were going ultrasonic... also way beyond the limits set in config.    This always happens on the first move after a halt state which is then reset by M999, the second and subsequent moves are all fine, only the very first move is affected.

#### Branch & Commit or Version of Smoothieware
Build version: edge-fceb3a0, Build date: Jul 17 2016 11:20:36, MCU: LPC1769, System Clock: 120MHz

#### Hardware model you are using (Make, and version)
Smoothieboard v1.0a 5XC

#### Host and Operating System running the Host
PuTTY terminal on  USB Serial  - Windows 7

#### Link to and console logs
I do not know about logs... instructions?
    
#### Link to running config and M503 results
Config - http://pastebin.com/AwKNBVPQ
M503 - http://pastebin.com/Z3pYC5s3

#### More information:
The M999 itself does not cause this error... If I am moving around and just do an M999 even though there was no halt condition, then following moves are at the correct speed.

Logic Analyzer Results:
Overview  Please note that the Y move (green) before the halt condition and the Y move after the halt are both the SAME distance (1000mm)  The second move attempted to produce the correct number of steps, but they were just very very fast.
[image]

Close up inspection of Y move before the limit was hit showing frequency of : 5.247kHz
[image]

Close up inspection of Y move after M999 showing frequency of: 48.6kHz (I was right it is ultrasonic)
[image]

Close up inspection of second move after M999 showing frequency of 5.246kHz once again
[image]

wolfmanjm, 2016-07-17:
your assumption that delta moves are different is incorrect. All moves are in machine coordinates, and delta moves just add to the current position to determine how far to move. 
There is code to try to predict where the actuators are after a HALT but it is not guaranteed to work, hence the requirement to Home after a halt/kill. I am not quite sure why this happens but it may just be something you need to accept (unless i can find a bug). 

You can also manually home with G28.3. So please try this after an kill... send G28.3 Xnnn Ynnn Znn to what you think the position is then try to jog and see what happens, this will tell me if there is a bug or just as expected the system no longer knows where it is.

I do not think the motion control can be changed to treat delta moves as only delta moves ignoring the current machine position.

wolfmanjm, 2016-07-17:
I also need exact steps to reproduce this as it works fine on my tests. This may also be a side effect of your unusual H/W and setup.
I would need to know what the last speed and destination was (from where) and what the next G0/G1 was that caused the issue.

wolfmanjm, 2016-07-17:
I also need to know what caused the HALT?

Zaaphod, 2016-07-17:
To reproduce:

turn on smoothie
access with USB serial terminal
send: G0 X500
trip X limit while it's moving
send: M999
Send G91 G0 Y-300
Y tries to move very fast

My bench-tests above were done on a smoothieboard with a minimally modified stock config file.. all I did was enable limits and invert the limit pins to make it more convenient to test.  The default in this stock config was 4000mm/min and 80steps/mm

wolfmanjm, 2016-07-17:
Should be fixed by PR #986

closed 2016-07-17 by wolfmanjm

Zaaphod, 2016-07-17:
Confirmed, issue is now resolved,  Thank You @wolfmanjm