Smoothieware/Smoothieware #987: Zgrid compensation not functional in latest edge (19 july 2016) works fine in latest stable (4 july-16)

roomooe, 2016-07-20:
#### What were you doing?
Trying to run zgrid compensation after update to latest edge (19/7-16)

#### What did you expect to happen?
Zgrid compensation...

#### What happened instead?
Error message "No strategy found to handle G32"
After installing latest stable version it works fine. Iäve attached my config it it helps..

#### Branch & Commit or Version of Smoothieware
Recv: Build version: master-27197b7, Build date: Jul  4 2016 04:10:42, MCU: LPC1768, System Clock: 100MHz

#### Hardware model you are using (Make, and version)
    MKS SBASE 1.2

#### Host and Operating System running the Host

    Octoprint and web host

[config.txt](https://github.com/Smoothieware/Smoothieware/files/373460/config.txt)

lordofhyphens, 2016-07-20:
If you're using it on a cartesian, suggest using deltagrid instead: http://smoothieware.org/zprobe#toc11 
Tagging @qharley because it's his module.

wolfmanjm, 2016-07-20:
works fine if y uuse delta grid. Zgrid is no longer supported unless @qharley fixes it.

closed 2016-07-20 by wolfmanjm

wolfmanjm, 2016-07-20:
@qharley FYI it is commented out in zprobe.cpp, however I did get it to compile, it just needs to be tested. 
If it tests out ok you can simply uncomment the lines in zprobe.cpp. to reactivate it. - Thanks