Smoothieware/Smoothieware #1009: Improve menus for the RDGLCD

ThisIsJustARandomGuy, 2016-08-24:
As this panel is cheap and easy to get I think the interface could get some improvements. This issue should serve as a general discussion for the particular improvements that could be made.

One section that has much more potential (and that I want to do work on) is the probe section. I don't know what the guidelines for interdependency between modules are. I suggest that at least the appropriate menu entries for doing the probe commands (g32 and so on depending on the enabled modules) and saving them afterwards or not (M500,...) should be added. What I would like to know before having a go at it is what is up with  the repository by Arthur wolf describing the "new menu-describing system" is: https://github.com/arthurwolf/Panel-Menu

Another example for easy an to implement but missing feature the fan speed.

If someone can think of other easy to implement features then I would like to have a go at it and create a pull request at some time. Suggestions on how to handle changing commands depending on the  enabled leveling strategies (e.g. G29 [P1]) are welcome.  I also really want to implement a confirmation for some of the options because I use a probe I need to attach manually and running a probe command by accident may damage the printer. Suggestions on that are quite welcome as well. Tomorrow I will push my slightly modified version and hope someone could have a look Aramco changes and tell me if I  am going on the right direction.

wolfmanjm, 2016-08-24:
there are very strict rules and methods for handling dependencies, you will need to read and understand the code to understand this. look at where the publicdata requests are used.
also make sure you read and understand the contribution guidelines in the wiki before submitting anything.

wolfmanjm, 2016-08-24:
Also it cannot be just for the rd glcd, it has to work with all supported displays.

wolfmanjm, 2016-08-24:
As a personal note I would change your name. no one is going to take you seriously with that name. I almost deleted this based on that name alone :)

ThisIsJustARandomGuy, 2016-08-24:
Yes of course. Most of it would probably just be click, click again and send a gcode. I don't think that's going to be a big problem. The bigger problem is going to be what to display. But I will read the documentation and come back tomorrow or in a few days.

wolfmanjm, 2016-08-25:
You are aware that you could add the G32 or G31 to the custom menu options? That is what I do. Also the Save (M500).

wolfmanjm, 2016-08-25:
How about using your real name like most of us do, with a Nickname which is displayed (that is optional). I mean this is not really facebook or twitter :)

ThisIsJustARandomGuy, 2016-08-25:
Of course I am aware of the custom menu and it is what I have used up until now. Another solution could be to add a few example configs to the documentation - it really should provide these features out of the box though in some way.

wolfmanjm, 2016-09-08:
closing this as it is not really an issue.

closed 2016-09-08 by wolfmanjm