Smoothieware/Smoothieware #1013: Documentation: Setting pins to Open Drain "o" appended onto pin number for running external driver.

Cromaglious, 2016-08-29:
Setting pin to open drain in the config is not discussed in config or manual.  There is mention in switch.  Also need mention on any voltage limits in using a pin as open drain.  TB6600 driver such as this one requires an open drain:

. http://www.ebay.com/itm/Single-Axis-TB6600-DC12-45V-Two-Phase-Hybrid-Stepper-Motor-Driver-Controller-B1-/111873859036

While this one does not:

http://www.ebay.com/itm/Single-Pulse-4A-TB6600-Stepper-Motor-Driver-Controller-9-42V-Micro-Step-CNC-/282116676829?hash=item41af76dcdd:g:xo4AAOSwH3NXmyVy

I don't want to fry my board from not knowing if the 5v into the step and dir pins need to be limited to 3.3v or less than so much mA.  I'm I going to be needing to build an opto isolator board to handle the 5v in and run the step and dir pins in source to drive the LED in the optoisolator.

There also needs to be an example config files for using external drivers using source and open drain modes.

wolfmanjm, 2016-08-29:
The wiki is open add the documentation if you feel it is inadequate.

closed 2016-08-29 by wolfmanjm

arthurwolf, 2016-08-29:
> Two-Phase-Hybrid-Stepper-Motor-Driver-Controller-B1-/111873859036
> Controller-9-42V-Micro-Step-CNC-/282116676829?hash=item41af76dcdd:g:
> xo4AAOSwH3NXmyVy

Hey.

For discution about the documentation, you generally want to go to the
forums or mailing lists ( or just edit the wiki directly, it's open to all
), the issues on github is just for code.
There is a mention of open-drain here :
http://smoothieware.org/configuring-smoothie, in the external drivers
example, and in the switch page. I believe there are a few other ones.
There is no page dedicated to Open-Drain but you are welcome to write one.

Cheers.

Cromaglious, 2016-08-29:
I don't know the hardware.  I don't know the limits on the pins.  All I can do in the wiki is make notes on where there is information lacking.

arthurwolf, 2016-08-29:
I don't know the limit either, they can be found in the LPC1769 datasheet I
believe.

Zaaphod, 2016-08-29:
The LPC1769 datasheet here: http://www.nxp.com/documents/data_sheet/LPC1769_68_67_66_65_64_63.pdf   Page 45 lists the open drain pins as having an absolute maximum of 5.5V and it calls them '5V tolerant Open-Drain' throughout the datasheet.  If you need more than 5V you will need an external buffer or transistor.