Smoothieware/Smoothieware #1021: Delta grid compensation: M375 gives "error:Failed to read grid radius"

courcirc8, 2016-09-11:
I have a problem with the delta grid compensation:
G31 probes the points, M375.1 displays the grid, M374 saves it correctly, but M375 (reload) always gives:
"error:Failed to read grid radius"
######################################################
...
... not all the probing point displayed here: ...
DEBUG: X90.0000, Y0.0000, Z0.1969
DEBUG: X60.0000, Y30.0000, Z0.1641
DEBUG: X30.0000, Y30.0000, Z0.0391
DEBUG: X0.0000, Y30.0000, Z-0.0234
DEBUG: X-30.0000, Y30.0000, Z-0.0797
DEBUG: X-60.0000, Y30.0000, Z-0.1234
DEBUG: X-60.0000, Y60.0000, Z-0.1578
DEBUG: X-30.0000, Y60.0000, Z-0.1203
DEBUG: X0.0000, Y60.0000, Z-0.0656
DEBUG: X30.0000, Y60.0000, Z0.0469
DEBUG: X60.0000, Y60.0000, Z0.1828
DEBUG: X0.0000, Y90.0000, Z-0.1234
 0.1734  0.1859  0.1922  0.1859  0.1906  0.1953  0.2000
 0.1000  0.1047  0.1141  0.1000  0.1391  0.1594  0.1719
 0.0391  0.0375  0.0359  0.0422  0.0875  0.1234  0.1594
 0.0750  0.1516  0.1969
-0.1672 -0.1234 -0.0797 -0.0234  0.0391  0.1641  0.2891
-0.2219 -0.1578 -0.1203 -0.0656  0.0469  0.1828  0.3187
-0.2359 -0.1984 -0.1609 -0.1234  0.0547  0.2016  0.3484
Probe completed
>>> M374
SENDING:M374
grid saved to /sd/delta.grid
>>> M500
SENDING:M500
Settings Stored to /sd/config-override
>>> M375.1
SENDING:M375.1
 0.1734  0.1859  0.1922  0.1859  0.1906  0.1953  0.2000
 0.1000  0.1047  0.1141  0.1000  0.1391  0.1594  0.1719
 0.0391  0.0375  0.0359  0.0422  0.0875  0.1234  0.1594
 0.1516  0.1969
-0.1672 -0.1234 -0.0797 -0.0234  0.0391  0.1641  0.2891
-0.2219 -0.1578 -0.1203 -0.0656  0.0469  0.1828  0.3187
-0.2359 -0.1984 -0.1609 -0.1234  0.0547  0.2016  0.3484
>>> M375
SENDING:M375
error:Failed to read grid radius

wolfmanjm, 2016-09-11:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/HowToFileIssues.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2016-09-11:
FWIW it sounds like you have a bad sdcard.

closed 2016-09-11 by wolfmanjm