Smoothieware/Smoothieware #1027: Errors build

rdacomp, 2016-09-16:
Smothieware -> Master
Windows 10 64 bit.

Error Build.
Logging install results to D:\Smoothieware-master\Smoothieware-master\win_instal
l.log
Downloading GNU Tools for ARM Embedded Processors...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100  100M  100  100M    0     0   654k      0  0:02:37  0:02:37 --:--:-- 1088k
Validating md5 signature of GNU Tools for ARM Embedded Processors...
Extracting GNU Tools for ARM Embedded Processors...
Creating helper scripts...
Performing a clean build of the gcc4mbed samples...
16.09.2016 23:25:16,56  Starting D:\Smoothieware-master\Smoothieware-master\win_
install.cmd
16.09.2016 23:25:16,57  Executing build\win32\curl -kL0 https://launchpad.net/gc
c-arm-embedded/4.8/4.8-2014-q1-update/+download/gcc-arm-none-eabi-4_8-2014q1-201
40314-win32.zip
16.09.2016 23:27:53,92  Executing build\win32\md5sum --check D:\Smoothieware-mas
ter\Smoothieware-master\gcc-arm-none-eabi.md5
gcc-arm-none-eabi-4_8-2014q1-20140314-win32.zip: OK
16.09.2016 23:27:54,54  Executing rd /s /q D:\Smoothieware-master\Smoothieware-m
aster\gcc-arm-none-eabi
16.09.2016 23:27:56,45  Executing md D:\Smoothieware-master\Smoothieware-master\
gcc-arm-none-eabi
16.09.2016 23:27:56,46  Executing cd D:\Smoothieware-master\Smoothieware-master\
gcc-arm-none-eabi
16.09.2016 23:27:56,47  Executing ..\build\win32\bsdtar xf D:\Smoothieware-maste
r\Smoothieware-master\gcc-arm-none-eabi-4_8-2014q1-20140314-win32.zip
16.09.2016 23:29:02,31  Executing cd ..
16.09.2016 23:29:02,38  Executing D:\Smoothieware-master\Smoothieware-master\bui
ld\win32\make.exe clean
Cleaning mbed
make[1]: Entering directory `D:/Smoothieware-master/Smoothieware-master/mbed'
Cleaning src
make[2]: Entering directory `D:/Smoothieware-master/Smoothieware-master/mbed/src
'
Cleaning mbed-lpc1768.mk
make[3]: Entering directory `D:/Smoothieware-master/Smoothieware-master/mbed/src
'
make[3]: Leaving directory `D:/Smoothieware-master/Smoothieware-master/mbed/src'

Cleaning mbed-lpc11u24.mk
make[3]: Entering directory `D:/Smoothieware-master/Smoothieware-master/mbed/src
'
make[3]: Leaving directory `D:/Smoothieware-master/Smoothieware-master/mbed/src'

Cleaning ../drop
make[2]: Leaving directory `D:/Smoothieware-master/Smoothieware-master/mbed/src'

make[1]: Leaving directory `D:/Smoothieware-master/Smoothieware-master/mbed'
      0 [main] sh 1568 sync_with_child: child 8608(0x18C) died before initializa
tion with status code 0xC0000142
  68622 [main] sh 1568 sync_with_child: *** child state waiting for longjmp
/usr/bin/sh: fork: Resource temporarily unavailable
make[2]: *** [clean_drop] Error 128
make[1]: *** [src.clean] Error 2
make: *** [mbed.clean] Error 2
16.09.2016 23:29:03,70  Failure forced early exit
Для продолжения нажмите любую клавишу . . .

wolfmanjm, 2016-09-16:
This is still a problem with your environment it is not reproducible and works fine here.

wolfmanjm, 2016-09-16:
and please stop filing the same issue over and over again.

closed 2016-09-16 by wolfmanjm

rdacomp, 2016-09-16:
It does not work and so I write.

rdacomp, 2016-09-16:
And I do not know what has caused.

rdacomp, 2016-09-16:
I took branches with Master. Unpack. Launched wines installs get an error. What could be the reason?

wolfmanjm, 2016-09-16:
You have something in your PATH that is using a different shell, read the wiki there are some trouble shooting techniques. This is for issues and bugs and this is not an issue or a bug and this is not a place to ask questions.

rdacomp, 2016-09-16:
I do not quite understand what it was about.
I have written since. What do you have in mind?

C:\Python27\;
C:\Python27\Scripts;
C:\Python27;
C:\WinAVR-20100110\bin;C:\WinAVR0100110\utils\bin;
C:\ProgramData\Oracle\Java\javapath;
%SystemRoot%\system32;
%SystemRoot%;
%SystemRoot%\System32\Wbem;
%SYSTEMROOT%\System32\WindowsPowerShell\v1.0\;
C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;
C:\Program Files (x86)\Skype\Phone\;
C:\Program Files\doxygen\bin;%USERPROFILE%\.dnx\bin;
C:\Program Files\Microsoft DNX\Dnvm\;
C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;
C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;
C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;
C:\Program Files\010 Editor;C:\Program Files\Git\cmd;
C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;
C:\Program Files (x86)\Microsoft SDKs\TypeScript\1.0\;
C:\Program Files (x86)\WinMerge;C:\Program Files\TortoiseGit\bin;
C:\Program Files\Microsoft\Web Platform Installer\

rdacomp, 2016-09-16:
And where I read that and on what program?

arthurwolf, 2016-09-16:
@rdacomp : please stop posting about this problem here. This is not the
right place. You can try the Smoothie forums or mailing lists if you wish.

wolfmanjm, 2016-09-16:
use a clean path.. this is what is causing the problem and this is documented in the wiki.
C:\WinAVR-20100110\bin;C:\WinAVR0100110\utils\bin;

please read http://smoothieware.org/

If you continue to misuse the issues system we will unfortunately need to block you.