Smoothieware/Smoothieware #1032: variable acceleration discussion / doc ?

J-Dunn, 2016-09-24:
Hi, 

I just cloned edge to have a look at single step accel in relation to variable acceleration. 

'make all ' went like a dream  very nice. 

All you seem to have documented so far is a link to TinyG . Is this documented or discussed anywhere where I lurk or chip in ?

wolfmanjm, 2016-09-24:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/HowToFileIssues.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

closed 2016-09-24 by wolfmanjm

J-Dunn, 2016-09-24:
Well I can't help much if you have a dumb robot blocking all communication. 

Good-bye. !

arthurwolf, 2016-09-24:
It's not blocking communication, it's saying we reserve github issues for
actual issues.
We have a forum, mailing lists ( including one specifically to discuss this
exact sort of thing ), IRC, social media ... all those are good places for
this sort of discution, and github issues are the only one we reserve
specifically for bugs.
You are welcome to discuss this anywhere else.

Cheers.

J-Dunn, 2016-09-24:
Thanks Arthur. 

If this is reserved for bug reports only that maybe should be the first line in all the text that gets thrown up.  Your developers' page says nothing about any of this nor anything about how to contact or discuss. 

Secondly, perhaps you need to consider a better structured forum. Dumping everything in 'General' with no structure and no search is so basic I get the feeling I'm back in 1995. 

At least try to help people contribute, rather than making it difficult and throwing up automatic replies saying "casses-toi".

arthurwolf, 2016-09-24:
The forum is generally more used for support. For the kind of discution you
were looking for, the smoothie-dev mailing list is probably the most
adequate place.