Smoothieware/Smoothieware #1044: Extruding directly after homing results in extruder stepper not moving correctly

pyjamasam, 2016-10-17:
#### What were you doing?
Attempting to extrude any length/volume of filament after homing.

#### What did you expect to happen?
Filament to be extruded normally.  Geared extruder to move and push the filament to the hot end.

#### What happened instead?
The stepper for the extruder makes a high pitch "buzz/tweet" and the filament doesn't extrude.  Gears of the extruder twitch but don't move as expected.  A second extrude happens correctly.  Its only the first extrude after a home.  If I move the print head at all (I.e. send any other G0/G1 as long as it doesn't try to move the extruder stepper) and then try to extrude everything works fine as well.

#### Branch & Commit or Version of Smoothieware
Build version: edge-3563d9b, Build date: Oct 14 2016 00:40:16, MCU: LPC1769, System Clock: 120MHz
   
#### Hardware model you are using (Make, and version)
Azteeg X5 Mini V1.1

#### Host and Operating System running the Host
Ponterface/Printrun 2014.08.01 on macOS 10.12
Also Octoprint 1.2.17rc3 on a RPI2

#### Link to and console logs
SENT: G28
SENT: G92 E0
SENT: M114
RECV: ok
RECV: ok
RECV: ok C: X:0.8977 Y:1.7339 Z:267.3729 E:0.000
Extruding 5.000000mm of filament.
SENT: G91
SENT: G1 E5.0 F100
SENT: G90
    
#### Link to running config and M503 results
[Config](https://gist.github.com/pyjamasam/b3b719518d0c5388a3a3a1df6b49b628)
[M503](https://gist.github.com/pyjamasam/37562e794d3011fd357e4f8cd98879d8)

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Not sure what section to put this under, but this only occurs when I have a loaded compensation grid.

If I run M370 (or M561) and then repeat the homeing/extruding test I get the expected results.  If I then run M375 and try the test again I get the tweeting stepper results.

If any other information is required or any clarification on any points is needed please let me know.  I am going to try and dig into the code around the compensation, but I am not up to snuff on the robot and motion planning parts as I'd like to be.

wolfmanjm, 2016-10-17:
Are you doing a G92 E0 after you home but before you try to extrude? As extruders are not involved in homing they need to be homed manually, unfortunately reprap decided to use G92 to do this (which is incorrect) however we handle G92 E0 specially and reset the extruder when we get it.
Please try that.

pyjamasam, 2016-10-17:
I don't do a G92 E0 after I do a G28.  Its nothing something that I have ever done in my workflow of homing the printer.  Either with a Ramps board or with my Smoothieboard.

That being said I gave it a try:
```
Send: G28
Recv: ok
Send: G92 E0
Recv: ok
Send: G91 G1 E5 F100 G90
Recv: ok
Send: G91 G1 E5 F100 G90
Recv: ok
```
First G1 E5 results is the chrip from the extruder stepper, 2nd G1 E5 works correctly.

If I run though:
```
Send: M370
Recv: grid cleared and disabled
Recv: ok
Send: G28
Recv: ok
Send: G92 E0
Recv: ok
Send: G91 G1 E5 F100 G90
Recv: ok
Send: G91 G1 E5 F100 G90
Recv: ok
```
Both G1 E5's run produce the expected stepper motion.

Running the following produces the erroneous results:
```
Send: M375
Recv: grid loaded, radius: 50.000000, size: 9
Recv: ok
Send: G28
Recv: ok
Send: G92 E0
Recv: ok
Send: G91 G1 E5 F100 G90
Recv: ok
Send: G91 G1 E5 F100 G90
Recv: ok
```

wolfmanjm, 2016-10-17:
I think I know what the issue is, and it should be fixed by Pull Request #1037 Before you tty that though can you try setting your max E speed much lower, In my experience max extruder speeds are in the order of 50mm/sec not 118mm/sec so try setting M203 E50 and see if that helps. If not then please try the Pull request.
One of the fixes in that pull request is that the XY positions are set correctly after homing when ABL is active.
Thank you (also thank you for filling out the issue form correctly :)

pyjamasam, 2016-10-17:
After setting M203 E50 I no longer get the chirping noise, BUT the Extruder still isn't moving correctly.
It is twitching, at a slower speed now, but still it's not the correct motion.

I'll snag that PR and give things another test and report in.

wolfmanjm, 2016-10-17:
Please note that that PR is currently under testing and there may be other issues. You may need to run the grid and trim stuff again (but try without first).
I think it solves your issue though.
FWIW I would set your max E speed much lower as it appears 50mm/sec maybe too fast for it as well.

wolfmanjm, 2016-10-17:
Fixed in PR #1037

closed 2016-10-17 by wolfmanjm