Smoothieware/Smoothieware #1051: Implement Feedhold and Resume for CNC Routers

xpix, 2016-10-26:
Hello, i'm wondering its not able to feedhold (Pause the machine) and resume in Smoothiware?

Please can u implement this, that's very(!) important for CNC Machines. I.e. you wuold change a tool or something is happend and you want just stop the machine w/o reset.

I.e.:
https://github.com/synthetos/TinyG/wiki/TinyG-Feedhold-and-Resume

arthurwolf, 2016-10-26:
We have the suspend and resume commands

wolfmanjm, 2016-10-26:
We do have feed hold in grbl mode over USB, (use the firmware-cnc.bin) it is ! and ~ as per GRBL.
It must allow the current buffered and planned commands complete first so can take a while, but it works.

closed 2016-10-26 by wolfmanjm

xpix, 2016-11-22:
That's not a workable solution. It must stop immediately. if i send this command the machine has to stop in this moment! In example, i start a job and see that i forgot a tool on my work surface (just an example) then i need a solution to stop/pause the machine, reove the tool and resume my job. If this possible?

wolfmanjm, 2016-11-22:
This will not be implemented any time soon. You are welcome to take a shot at it. But stopping immediately is very hard if not impossible given it needs to decelerate to remain mechanically sound.
The planner has already been planned and the acceleration/deceleration curves already calculated.
You need to recalculate all the stored moves from the current one up to find a place where it can decelerate. This needs to be done fast enough so as not to disrupt the current move. Also the entry and exit speeds need to be such that it can in fact decelerate to zero in the given distance that is stored in the planner queue. If this were easy it would have been done.

For tool change it is not needed as you know where tool changes are in the gcode and you just stop sending until you have changed the tool. Emergency stop does not need to decelerate and that is already implemented with the kill button. So really the use cases you have proposed do not demand this be implemented. There are use cases where it would be nice to stop as soon as possible without losing position, however it has not been so important that anyone has tried to implement it.
So unless you want to do it, I suspect it will not get done any time soon. Sorry for the bad news.

xpix, 2016-11-22:
Ok, no Problem i'll test this on my side, but i'm wondering that's so hard to implement this ... it's just a pause of controller. GRBL and also TinyG has this feedhold/pause functions and they stop immediately in this moment. Ok a kill/reset button is also possible, but you have to home the machine again ... not nice, but possible :)

triffid, 2016-11-22:
It should be possible to set a flag somewhere that causes all motors to
decelerate at max, but what happens then to the queued moves? You could add
up all the step counters, subtract from last milestone and then FK to find
the current position, but people are gonna expect to be able to resume at
the paused point which requires recalculation of the whole queue, a tricky
proposition given that we don't store the start or end point of each queued
block..

From a code architecture perspective, how would you expect this to work
exactly?

xpix, 2016-11-22:
Maybe this help's:
https://github.com/synthetos/TinyG/wiki/TinyG-Feedhold-and-Resume

The Feedhold command it's a pause button, stop all moving of steppers but don't forget position. The commands in queued moves has to wait. After resume '~' the machine starts moving at this point. A very important command is the Queue Flush, after a pause you can remove all waiting commands and stop the machine.

GorillaMachines, 2017-10-30:
Is this still an open issue? I am certainly having the same issue on my end. The need for a proper pause/resume code is massive, especially for high load tools that could seriously hurt someone. It seems from the surface to not be too involved. I can understand the need for a deceleration and acceleration and how that would be the issue, but could it just simply be implemented as a hard stop with no deceleration. All that would have to be done is make it an optional feature and inform the user that it is possible to have a loss of position. You could even have a max feed setting where the machine at a certain feed rate (max) would require rehoming and would inform the user of loss of position and go into an alarm state, but below the max feedrate (tuned by the user depending on the machine just as jerk is) it keeps the position and simply triggers a hold on the queue. Is this possible?

wolfmanjm, 2017-10-30:
No this issue is closed. 

@GorillaMachines  there is already a hard abort either use the kill button or send a control X, this stops the machine instantly, however good practice would say that you must NEVER rely on firmware for emergency stop, you need a real e stop button that cuts power.

There is also already a pause and resume !/~ in grbl mode, and if you set segmentation to 0.1mm it will pause within 3.2mm, this is adequate for most people who need pause as opposed to emergency stop.

It is VERY hard to implement in immediate pause as stated above, which is why it has not been done.