Smoothieware/Smoothieware #1056: Include Math.h in /src/modules/tools/zprobe/DeltaCalibrationStrategy.cpp

Salamandar, 2016-11-04:
I tried to compile Smoothieware with my distro-provided arm-none-eabi-gcc, which is Gcc6.
```
modules/tools/zprobe/DeltaCalibrationStrategy.cpp: In member function 'bool DeltaCalibrationStrategy::probe_delta_points(Gcode*)':
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:99:19: error: 'isnan' was not declared in this scope
     if(isnan(bedht)) return false;
                   ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:117:19: error: 'NAN' was not declared in this scope
     float last_z= NAN;
                   ^~~
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:137:24: error: 'isnan' was not declared in this scope
         if(isnan(last_z)) {
                        ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:140:58: error: 'fabsf' was not declared in this scope
             max_delta= std::max(max_delta, fabsf(z-last_z));
                                                          ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp: In member function 'float DeltaCalibrationStrategy::findBed()':
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:156:30: error: 'NAN' was not declared in this scope
     zprobe->coordinated_move(NAN, NAN, -deltaz, zprobe->getFastFeedrate(), true);
                              ^~~
modules/tools/zprobe/DeltaCalibrationStrategy.cpp: In member function 'bool DeltaCalibrationStrategy::calibrate_delta_endstops(Gcode*)':
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:214:19: error: 'isnan' was not declared in this scope
     if(isnan(bedht)) return false;
                   ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:222:16: error: 'fabsf' was not declared in this scope
     if(fabsf(dz) > target) {
                ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:262:34: error: 'NAN' was not declared in this scope
         zprobe->coordinated_move(NAN, NAN, -bedht, zprobe->getFastFeedrate(), true); // do a relative move from home to the point above the bed
                                  ^~~
modules/tools/zprobe/DeltaCalibrationStrategy.cpp: In member function 'bool DeltaCalibrationStrategy::calibrate_delta_radius(Gcode*)':
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:321:19: error: 'isnan' was not declared in this scope
     if(isnan(bedht)) return false;
                   ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:329:16: error: 'fabsf' was not declared in this scope
     if(fabsf(dz) > target) {
                ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:369:19: error: 'fabsf' was not declared in this scope
         if(fabsf(d) <= target){
                   ^
modules/tools/zprobe/DeltaCalibrationStrategy.cpp:384:34: error: 'NAN' was not declared in this scope
         zprobe->coordinated_move(NAN, NAN, -bedht, zprobe->getFastFeedrate(), true); // needs to be a relative coordinated move

```
The problem was solved by adding line 17 in src/modules/tools/zprobe/DeltaCalibrationStrategy.cpp this :
`#include <math.h>
`
Indeed, some things changed in the Gcc headers. Anyways, that works fine now !

wolfmanjm, 2016-11-04:
Please use the gcc provided with smoothie. We do not support other compilers so if you insist on using a different compiler then you will need to do whatever is needed on you own branch. Thank you.

closed 2016-11-04 by wolfmanjm

Salamandar, 2016-11-05:
Well I don't understand that reaction. You're still using Gcc4.8, while Gcc5 is already 1.5 years old. Gcc6 is out for some months now.
I'm not using SW (still on an arduino) right now, I just wanted to see how things work. This is a ridiculous fix and would help people test SW with Gcc6.

wolfmanjm, 2016-11-19:
The reason is if everyone used a different compiler it would make testing impossible and make it impossible to ship a stable binary.

We use a stable compiler so the binaries are stable. You cannot possibly fully test smoothie with every version of every compiler you could use.

FWIW this issue was solved anyway with PR #1065