Smoothieware/Smoothieware #1057: Compiling fail if a signed commit is present

nmaggioni, 2016-11-08:
#### What were you doing?

Compiling the firmware as per the instructions, after having committed a modification. The commit was signed with my PGP key.

#### What did you expect to happen?

Compilation to succeed.

#### What happened instead?

Compilation failed with these errors - I suspect that signed commits are not parsed correctly somewhere:

```
...
Deploying vendor/NXP/cmsis/LPC11U24/core_cmFunc.h to drop
Deploying vendor/NXP/cmsis/LPC11U24/cmsis.h to drop
Exporting linker script to ../drop/LPC11U24/GCC_ARM/LPC11U24.ld
make[3]: uscita dalla directory "/home/nmaggioni/Scaricati/Smoothieware/mbed/src"
make[2]: uscita dalla directory "/home/nmaggioni/Scaricati/Smoothieware/mbed/src"
make[1]: uscita dalla directory "/home/nmaggioni/Scaricati/Smoothieware/mbed"
Building Smoothie
make[1]: ingresso nella directory "/home/nmaggioni/Scaricati/Smoothieware/src"
Building for device lpc1768
make[2]: ingresso nella directory "/home/nmaggioni/Scaricati/Smoothieware/src"
Compiling libs/Network/uip/clock-arch.c
arm-none-eabi-gcc: error: Signature: No such file or directory
arm-none-eabi-gcc: error: made: No such file or directory
arm-none-eabi-gcc: error: mar: No such file or directory
arm-none-eabi-gcc: error: 08: No such file or directory
arm-none-eabi-gcc: error: nov: No such file or directory
arm-none-eabi-gcc: error: 2016: No such file or directory
arm-none-eabi-gcc: error: 20:00:07: No such file or directory
arm-none-eabi-gcc: error: CET: No such file or directory
arm-none-eabi-gcc: error: gpg:: No such file or directory
arm-none-eabi-gcc: error: using: No such file or directory
arm-none-eabi-gcc: error: RSA: No such file or directory
arm-none-eabi-gcc: error: key: No such file or directory
arm-none-eabi-gcc: error: 4874B0C841E33264: No such file or directory
arm-none-eabi-gcc: error: gpg:: No such file or directory
arm-none-eabi-gcc: error: Good: No such file or directory
arm-none-eabi-gcc: error: signature: No such file or directory
arm-none-eabi-gcc: error: from: No such file or directory
arm-none-eabi-gcc: error: Niccolò Maggioni <nicco.maggioni@gmail.com>: No such file or directory
arm-none-eabi-gcc: error: [ultimate]: No such file or directory
arm-none-eabi-gcc: error: gpg:: No such file or directory
arm-none-eabi-gcc: error: aka: No such file or directory
arm-none-eabi-gcc: error: [jpeg image of size 3709]: No such file or directory
arm-none-eabi-gcc: error: [ultimate]: No such file or directory
arm-none-eabi-gcc: error: 1e65d04": No such file or directory
make[2]: *** [../build/common.mk:307: ../LPC1768/./libs/Network/uip/clock-arch.o] Error 1
make[2]: uscita dalla directory "/home/nmaggioni/Scaricati/Smoothieware/src"
make[1]: *** [../build/build.mk:95: lpc1768] Error 2
make[1]: uscita dalla directory "/home/nmaggioni/Scaricati/Smoothieware/src"
make: *** [makefile:9: all] Error 2
```

#### Branch & Commit or Version of Smoothieware

edge @ 9e0f7a5

#### Host and Operating System running the Host

    Linux x86_64

wolfmanjm, 2016-11-08:
looks like you corrupted a file, the build has nothing to do with git or signing.

closed 2016-11-08 by wolfmanjm

nmaggioni, 2016-11-08:
Could be, but I don't see how: without modifying any file, amending the commit and not signing it again gets rid of the problem.

arthurwolf, 2016-11-08:
Can we see

libs/Network/uip/clock-arch.c

?

wolfmanjm, 2016-11-08:
The make system neither uses nor requires git to even be available. Except when it tries to create the version string then it asks for the checksum and name of the current branch.

arhi, 2016-11-08:
@nmaggioni you pushed the diff trough email?

nmaggioni, 2016-11-08:
@arthurwolf Nothing fancy, it seems untouched: http://pastebin.com/zWaMgkn4

@arhi Negative.

@wolfmanjm That's the curlprit indeed! The command that the versioning script uses does not play nice with PGP signing.

Here is the output of a non-signed commit: `feature/decoupled-hotend-bed-preheating-417f227`, and here is the one of a signed commit: `feature/decoupled-hotend-bed-preheating-gpg: Signature made mar 08 nov 2016 23:55:17 CET gpg: using RSA key 4874B0C841E33264 gpg: Good signature from "Niccolò Maggioni <nicco.maggioni@gmail.com>" [ultimate] gpg: aka "[jpeg image of size 3709]" [ultimate] 8d8a6ab`.

nmaggioni, 2016-11-08:
Might `git rev-parse --short=7 HEAD` be a good replacement for `git log --pretty=format:%h -1`? It shows no problems on my end with both signed and non-signed commits.

wolfmanjm, 2016-11-08:
i'd suggest simply not signing your commits.

nmaggioni, 2016-11-08:
Sure, that's a way. But why leaving the chance to incur in the [git horror story](https://mikegerwitz.com/papers/git-horror-story)?

I usually sign all my commits, I noticed this problem only because I made git sign them all by default global configuration. I promptly turned it off for this repo, but I thought that you might have wanted to know about this "issue".

Great job anyway, I love Smoothieware :)

triffid, 2016-11-09:
if git rev-parse --short=7 HEAD has a more consistent format in various
situations, it would be wise to patch the version generator, no?

wolfmanjm, 2016-11-09:
It doesn't return the information we need.

nmaggioni, 2016-11-10:
The output seems quite the same to me... What are the needed informations?

wolfmanjm, 2016-11-11:
the script needs to be updated to use this and tested. As I do not sign stuff I can't test it. If you want this fixed you  need to modify the script and submit a pull request.

commit 4e9d251a (2016-11-11): Support for signed commit in versioning script

Closes #1057