Smoothieware/Smoothieware #1066: Support TMC2130

DaGov, 2016-11-18:
Are you have plans to support Watterott TMC2130 via SPI?

wolfmanjm, 2016-11-18:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

closed 2016-11-18 by wolfmanjm

wolfmanjm, 2016-11-19:
there is support for TMC2660, most TMC26x SPI chips are the same. Compare the TMC2130 to a TMC 26x and see what the differences are, and either add a new motordriver controller or modify the TMC26x accordingly.
There are no plans by the core team to add TMC2130.
(PS in future please do not ask questions in the issues section, it is not for questions, use one of the many forums available).

LichtiMC, 2018-01-31:
Has anybody already created a motordriver for TMC2130 controller?
Thank you very much...

wolfmanjm, 2018-01-31:
if they had it would be listed in the wiki. 
again this is for bug reports NOT asking questions.

wolfmanjm, 2018-05-22:
See PR #1315