Smoothieware/Smoothieware #1067: Cannot delete file on /ext using rm shell command

luc-github, 2016-11-26:
Hi,
I have an Azteeg X5 and viki2 screen and an AZSMZ with AZSMZ lscreen with different SDCard and latest smoothie FW
Using shell command I can delete file on /sd
like: `rm /sd/file1.gcode`
but when I do :
`rm /ext/file2.gcode`
I got `Could not delete /ext/file2.gcode`
File can be play using `play  /ext/file2.gcode` and visible when I use `ls /ext` 

I do not see any limitation in http://smoothieware.org/console-commands
It is a bug or a limitation ?

thanks

wolfmanjm, 2016-11-26:
It maybe a bug. But as /ext is not really supported I suspect it will not get fixed.
If you read the wiki section on external sdcards you will see we HIGHLY recommend it NOT be used due to noise on the SPI. It is simply not designed to run over a cable of any length.

closed 2016-11-26 by wolfmanjm

luc-github, 2016-11-26:
If it is not recommended and not supported , it would make sense to disable it directly in FW,  no ?

wolfmanjm, 2016-11-26:
I would like to, however some people are able to use it without little problem, others it barely works at all.
So it is left in there with dire warnings about using it.
I can see no reason why rm /ext/file would not work unless the external is set to readonly, which would actually make sense. not sure if the fatlib does readonly. or not.

luc-github, 2016-11-26:
Ok Thanks

wolfmanjm, 2016-11-26:
I looked at the code if this is a bug it is deep within the Chans FATLIB code (which is really really ugly ;)
or withing the Mbed diskio stuff.
My suspicion is that the sdcard is set to readonly somehow, although there is no explicit way to do that in smoothie, maybe the card reader is reading it as readonly?

luc-github, 2016-11-26:
both readers I tested are micro SD with no way to set as read only - so I guess /ext is mount as read only by file system but did not found if can be specified in mbed documentation ...