Smoothieware/Smoothieware #1070: When bed heat with M190 there is no way to stop it

luc-github, 2016-12-10:
When wait and heating command is sent there is not way to stop it
I have tried to stop using `M140 S0` the off command, the shell command `set_temp bed 0`, even `M112` the emergency stop and nothing works, I did not checked for extruder neither M116.

Is that normal ?  
Is there no way to stop heating when wait command is active before temperature is reached ?

Thank you

wolfmanjm, 2016-12-10:
this does not conform to issues... however yes there is a way, use the kill button or send control X

closed 2016-12-10 by wolfmanjm

luc-github, 2016-12-10:
Ok so it is a "feature" and not an issue if no command can stop heating.

I forgot to mention the interface is the network one, so no access to kill button neither control X that work only on Serial

Thanks for your answer