Smoothieware/Smoothieware #1074: Delta Grid Compensation

kenoro, 2016-12-16:
#### What were you doing?

    Delta Grid Compensation

#### What did you expect to happen?

Make the probing and the save ALL the values to the file.

#### What happened instead?

Made the probing but some values are left out and not included in the grid. And more I can't save the grid. 

Probe completed
>>> g374
SENDING:G374
>>> g375.1
SENDING:G375.1
>>> @version
SENDING:version
Build version: edge-dcf0252, Build date: Nov 27 2016 14:17:25, MCU: LPC1769, System Clock: 120MHz
>>> M503

#### Branch & Commit or Version of Smoothieware

    Build version: edge-dcf0252, Build date: Nov 27 2016 14:17:25, MCU: LPC1769, System Clock: 120MHz

#### Hardware model you are using (Make, and version)

    Smoothieboard 5X V1b

#### Host and Operating System running the Host

    Pronterface, win 8.1

#### Link to and console logs

https://gist.github.com/kenoro/d0e767957beed062d7e679a4b7916a51

    
#### Link to running config and M503 results

https://gist.github.com/kenoro/643f61adae072540d035f11a1dae67f6

https://gist.github.com/kenoro/dfe70a07936277d98fdcf118ad1d30c5

wolfmanjm, 2016-12-16:
The grid size must be odd numbers, try 9x9 or 11x11. Reopen if that does not fix it.

closed 2016-12-16 by wolfmanjm

wolfmanjm, 2016-12-16:
It doesn't save because you did not configure it to save.
Please carefully read the wiki documentation...  http://smoothieware.org/zprobe#toc11

kenoro, 2016-12-16:
Made the modifications to the config to save after the probing is finished but still doesnt save/ load the full grid.

> Delta Grid Probe...
Probe start ht is 34.885002 mm, probe radius is 75.000000 mm, grid size is 11x11
probe at 0,0 is 0.007501 mm
DEBUG: X75.0000, Y-75.0000, Z0.0850
DEBUG: X60.0000, Y-75.0000, Z0.0675
DEBUG: X45.0000, Y-75.0000, Z0.0975
DEBUG: X30.0000, Y-75.0000, Z0.0700
DEBUG: X15.0000, Y-75.0000, Z0.0625
DEBUG: X0.0000, Y-75.0000, Z0.0825
DEBUG: X-15.0000, Y-75.0000, Z0.0775
DEBUG: X-30.0000, Y-75.0000, Z0.0775
DEBUG: X-45.0000, Y-75.0000, Z0.1025
DEBUG: X-60.0000, Y-75.0000, Z0.0725
DEBUG: X-75.0000, Y-75.0000, Z0.0875
DEBUG: X-75.0000, Y-60.0000, Z0.0325
DEBUG: X-60.0000, Y-60.0000, Z0.0300
DEBUG: X-45.0000, Y-60.0000, Z0.0525
DEBUG: X-30.0000, Y-60.0000, Z0.0400
DEBUG: X-15.0000, Y-60.0000, Z0.0325
DEBUG: X0.0000, Y-60.0000, Z0.0450
DEBUG: X15.0000, Y-60.0000, Z0.0300
DEBUG: X30.0000, Y-60.0000, Z0.0350
DEBUG: X45.0000, Y-60.0000, Z0.0600
DEBUG: X60.0000, Y-60.0000, Z0.0400
DEBUG: X75.0000, Y-60.0000, Z0.0725
DEBUG: X75.0000, Y-45.0000, Z0.0525
DEBUG: X60.0000, Y-45.0000, Z0.0300
DEBUG: X45.0000, Y-45.0000, Z0.0450
DEBUG: X30.0000, Y-45.0000, Z0.0150
DEBUG: X15.0000, Y-45.0000, Z0.0175
DEBUG: X0.0000, Y-45.0000, Z0.0275
DEBUG: X-15.0000, Y-45.0000, Z0.0125
DEBUG: X-30.0000, Y-45.0000, Z0.0275
DEBUG: X-45.0000, Y-45.0000, Z0.0275
DEBUG: X-60.0000, Y-45.0000, Z0.0075
DEBUG: X-75.0000, Y-45.0000, Z0.0125
DEBUG: X-75.0000, Y-30.0000, Z0.0275
DEBUG: X-60.0000, Y-30.0000, Z0.0125
DEBUG: X-45.0000, Y-30.0000, Z0.0425
DEBUG: X-30.0000, Y-30.0000, Z0.0275
DEBUG: X-15.0000, Y-30.0000, Z0.0175
DEBUG: X0.0000, Y-30.0000, Z0.0325
DEBUG: X15.0000, Y-30.0000, Z0.0250
DEBUG: X30.0000, Y-30.0000, Z0.0175
DEBUG: X45.0000, Y-30.0000, Z0.0425
DEBUG: X60.0000, Y-30.0000, Z0.0275
DEBUG: X75.0000, Y-30.0000, Z0.0575
DEBUG: X75.0000, Y-15.0000, Z0.0350
DEBUG: X60.0000, Y-15.0000, Z0.0075
DEBUG: X45.0000, Y-15.0000, Z0.0275
DEBUG: X30.0000, Y-15.0000, Z0.0025
DEBUG: X15.0000, Y-15.0000, Z0.0025
DEBUG: X0.0000, Y-15.0000, Z0.0075
DEBUG: X-15.0000, Y-15.0000, Z-0.0125
DEBUG: X-30.0000, Y-15.0000, Z-0.0050
DEBUG: X-45.0000, Y-15.0000, Z-0.0025
DEBUG: X-60.0000, Y-15.0000, Z-0.0275
DEBUG: X-75.0000, Y-15.0000, Z-0.0150
DEBUG: X-75.0000, Y0.0000, Z-0.0325
DEBUG: X-60.0000, Y0.0000, Z-0.0500
DEBUG: X-45.0000, Y0.0000, Z-0.0200
DEBUG: X-30.0000, Y0.0000, Z-0.0275
DEBUG: X-15.0000, Y0.0000, Z-0.0350
DEBUG: X0.0000, Y0.0000, Z-0.0125
DEBUG: X15.0000, Y0.0000, Z-0.0200
DEBUG: X30.0000, Y0.0000, Z-0.0175
DEBUG: X45.0000, Y0.0000, Z0.0075
DEBUG: X60.0000, Y0.0000, Z-0.0125
[... 62 lines cut ...]
-0.0750 -0.0900 -0.0650 -0.0650 -0.0700 -0.0425 -0.0575 -0.0375 -0.0200 -0.0125  0.0225
-0.0900 -0.1000 -0.1000 -0.0750 -0.0950 -0.0675 -0.0375 -0.0450 -0.0075
-0.1550 -0.1500 -0.1450 -0.1375 -0.1425 -0.1300 -0.1225 -0.1200 -0.1000 -0.0875 -0.0475
-0.1175 -0.1050 -0.0975 -0.0550
-0.2050 -0.2100 -0.1900 -0.1925 -0.1875 -0.1725 -0.1825 -0.1625 -0.1425 -0.1450 -0.0800
Probe completed

After that I send 

> >>> m375.1
SENDING:M375.1
 0.0875  0.0725  0.1025  0.0775  0.0775  0.0825  0.0625  0.0700  0.0975  0.0675  0.0850
 0.0325  0.0300  0.0525  0.0400  0.0325  0.0450  0.0300  0.0350  0.0600  0.0400  0.0725
 0.0275  0.0175  0.0150  0.0450  0.0300  0.0525
 0.0275  0.0125  0.0425  0.0275  0.0175  0.0325  0.0250  0.0175  0.0425  0.0275  0.0575
-0.0150 -0.0275 -0.0025 -0.0050 -0.0125  0.0075  0.0025  0.0025  0.0275  0.0075  0.0350
-0.0325 -0.0500 -0.0200 -0.0275 -0.0350 -0.0125 -0.0200 -0.0175  0.0075 -0.0125  0.0250
-0.0750 -0.0900 -0.0650 -0.0650 -0.0700 -0.0425 -0.0575 -0.0375 -0.0200 -0.0125  0.0225
-0.1000 -0.1000 -0.0750 -0.0950 -0.0675 -0.0375 -0.0450 -0.0075
-0.1550 -0.1500 -0.1450 -0.1375 -0.1425 -0.1300 -0.1225 -0.1200 -0.1000 -0.0875 -0.0475
-0.1050 -0.0975 -0.0550
-0.2050 -0.2100 -0.1900 -0.1925 -0.1875 -0.1725 -0.1825 -0.1625 -0.1425 -0.1450 -0.0800

I attach also the grid file maybe you can view it. Is it possible that the values are there and available but for some god known reason are not displayed? 
[delta.zip](https://github.com/Smoothieware/Smoothieware/files/656640/delta.zip)

wolfmanjm, 2016-12-16:
It looks to me like it is working fine. I know this works, so you must simply have a configuration error. Please use the IRC or other help channels as this is not a bug.

kenoro, 2016-12-16:
It can be a configuration error, because the probing works! 

The saving and retrieving the values from the file isn't

kenoro, 2016-12-16:
How am i supposed to reopen the thread/issue?

kenoro, 2016-12-16:
Firmaware version:

SENDING:version
Build version: edge-d792543, Build date: Dec 15 2016 13:37:18, MCU: LPC1769, System Clock: 120MHz
5 axis

Made the modifications to the config to save after the probing is finished but still doesnt save/ load the full grid.

Delta Grid Probe...
Probe start ht is 34.885002 mm, probe radius is 75.000000 mm, grid size is 11x11
probe at 0,0 is 0.007501 mm
DEBUG: X75.0000, Y-75.0000, Z0.0850
DEBUG: X60.0000, Y-75.0000, Z0.0675
DEBUG: X45.0000, Y-75.0000, Z0.0975
DEBUG: X30.0000, Y-75.0000, Z0.0700
DEBUG: X15.0000, Y-75.0000, Z0.0625
DEBUG: X0.0000, Y-75.0000, Z0.0825
DEBUG: X-15.0000, Y-75.0000, Z0.0775
DEBUG: X-30.0000, Y-75.0000, Z0.0775
DEBUG: X-45.0000, Y-75.0000, Z0.1025
DEBUG: X-60.0000, Y-75.0000, Z0.0725
DEBUG: X-75.0000, Y-75.0000, Z0.0875
DEBUG: X-75.0000, Y-60.0000, Z0.0325
DEBUG: X-60.0000, Y-60.0000, Z0.0300
DEBUG: X-45.0000, Y-60.0000, Z0.0525
DEBUG: X-30.0000, Y-60.0000, Z0.0400
DEBUG: X-15.0000, Y-60.0000, Z0.0325
DEBUG: X0.0000, Y-60.0000, Z0.0450
DEBUG: X15.0000, Y-60.0000, Z0.0300
DEBUG: X30.0000, Y-60.0000, Z0.0350
DEBUG: X45.0000, Y-60.0000, Z0.0600
DEBUG: X60.0000, Y-60.0000, Z0.0400
DEBUG: X75.0000, Y-60.0000, Z0.0725
DEBUG: X75.0000, Y-45.0000, Z0.0525
DEBUG: X60.0000, Y-45.0000, Z0.0300
DEBUG: X45.0000, Y-45.0000, Z0.0450
DEBUG: X30.0000, Y-45.0000, Z0.0150
DEBUG: X15.0000, Y-45.0000, Z0.0175
DEBUG: X0.0000, Y-45.0000, Z0.0275
DEBUG: X-15.0000, Y-45.0000, Z0.0125
DEBUG: X-30.0000, Y-45.0000, Z0.0275
DEBUG: X-45.0000, Y-45.0000, Z0.0275
DEBUG: X-60.0000, Y-45.0000, Z0.0075
DEBUG: X-75.0000, Y-45.0000, Z0.0125
DEBUG: X-75.0000, Y-30.0000, Z0.0275
DEBUG: X-60.0000, Y-30.0000, Z0.0125
DEBUG: X-45.0000, Y-30.0000, Z0.0425
DEBUG: X-30.0000, Y-30.0000, Z0.0275
DEBUG: X-15.0000, Y-30.0000, Z0.0175
DEBUG: X0.0000, Y-30.0000, Z0.0325
DEBUG: X15.0000, Y-30.0000, Z0.0250
DEBUG: X30.0000, Y-30.0000, Z0.0175
DEBUG: X45.0000, Y-30.0000, Z0.0425
DEBUG: X60.0000, Y-30.0000, Z0.0275
DEBUG: X75.0000, Y-30.0000, Z0.0575
DEBUG: X75.0000, Y-15.0000, Z0.0350
DEBUG: X60.0000, Y-15.0000, Z0.0075
DEBUG: X45.0000, Y-15.0000, Z0.0275
DEBUG: X30.0000, Y-15.0000, Z0.0025
DEBUG: X15.0000, Y-15.0000, Z0.0025
DEBUG: X0.0000, Y-15.0000, Z0.0075
DEBUG: X-15.0000, Y-15.0000, Z-0.0125
DEBUG: X-30.0000, Y-15.0000, Z-0.0050
DEBUG: X-45.0000, Y-15.0000, Z-0.0025
DEBUG: X-60.0000, Y-15.0000, Z-0.0275
DEBUG: X-75.0000, Y-15.0000, Z-0.0150
DEBUG: X-75.0000, Y0.0000, Z-0.0325
DEBUG: X-60.0000, Y0.0000, Z-0.0500
DEBUG: X-45.0000, Y0.0000, Z-0.0200
DEBUG: X-30.0000, Y0.0000, Z-0.0275
[... 67 lines cut ...]
-0.0325 -0.0500 -0.0200 -0.0275 -0.0350 -0.0125 -0.0200 -0.0175 0.0075 -0.0125 0.0250
-0.0750 -0.0900 -0.0650 -0.0650 -0.0700 -0.0425 -0.0575 -0.0375 -0.0200 -0.0125 0.0225
-0.0900 -0.1000 -0.1000 -0.0750 -0.0950 -0.0675 -0.0375 -0.0450 -0.0075
-0.1550 -0.1500 -0.1450 -0.1375 -0.1425 -0.1300 -0.1225 -0.1200 -0.1000 -0.0875 -0.0475
-0.1175 -0.1050 -0.0975 -0.0550
-0.2050 -0.2100 -0.1900 -0.1925 -0.1875 -0.1725 -0.1825 -0.1625 -0.1425 -0.1450 -0.0800
Probe completed
After that I send

m375.1
SENDING:M375.1
0.0875 0.0725 0.1025 0.0775 0.0775 0.0825 0.0625 0.0700 0.0975 0.0675 0.0850
0.0325 0.0300 0.0525 0.0400 0.0325 0.0450 0.0300 0.0350 0.0600 0.0400 0.0725
0.0275 0.0175 0.0150 0.0450 0.0300 0.0525
0.0275 0.0125 0.0425 0.0275 0.0175 0.0325 0.0250 0.0175 0.0425 0.0275 0.0575
-0.0150 -0.0275 -0.0025 -0.0050 -0.0125 0.0075 0.0025 0.0025 0.0275 0.0075 0.0350
-0.0325 -0.0500 -0.0200 -0.0275 -0.0350 -0.0125 -0.0200 -0.0175 0.0075 -0.0125 0.0250
-0.0750 -0.0900 -0.0650 -0.0650 -0.0700 -0.0425 -0.0575 -0.0375 -0.0200 -0.0125 0.0225
-0.1000 -0.1000 -0.0750 -0.0950 -0.0675 -0.0375 -0.0450 -0.0075
-0.1550 -0.1500 -0.1450 -0.1375 -0.1425 -0.1300 -0.1225 -0.1200 -0.1000 -0.0875 -0.0475
-0.1050 -0.0975 -0.0550
-0.2050 -0.2100 -0.1900 -0.1925 -0.1875 -0.1725 -0.1825 -0.1625 -0.1425 -0.1450 -0.0800

I attach also the grid file maybe you can view it. Is it possible that the values are there and available but for some god known reason are not displayed?
delta.zip

wolfmanjm, 2016-12-16:
It looks like it is working properly but your display is dropping characters,  nothing at all wrong with smoothie.

kenoro, 2016-12-17:
Pronterface es connect to smoothie by lan. I will try again by usb to see what's happening.
I don't understand how my display cab drop characters, it's a laptop not the display of the printer so it could be the case in doesn't have enough ram.

Anyway I will be back after I check symptoms over the usb connections

callericaz, 2017-06-17:
I have the same issue as Kenoro, but I have some additional information and a temporary workaround.
I am currently running upstreamedge-01cacfc, Build date: May 31 2017 16:35:23, MCU: LPC1769
Hardware revision is 5XC v1.0a

This issue manifests with any grid density greater than 15 (225 probe points)
The delta grid will appear to save correctly after probing with G31,
Sending M374 gives output: "grid saved to /sd/delta.grid"
However, when M375 is issued, an error is output: "error:grid size is different read 60 - config 17"
This output was from a configured probe density of 17, and with the config setting:
"leveling-strategy.delta-grid.save false" and the M375 command removed from config-override.
I am relatively certain that I have avoided errors in my configuration file, but will provide it on request if needed.

Delta grid probing with higher density, such as 29 (841 probe points) CAN currently be used,
but ONLY UNTIL a power cycle or M561, as the probe data refuses to save to SD in a readable format.
This is annoying, as I power cycle my unit to remove the SD card, so I can load large gcode files without the huge bottleneck of the web interface.

I don't use pronterface to load or stream gcode due to unrelated problems regarding "relative" extruder mode, which is required for my use case due to an unrelated "wrap-around" issue causing the extruder to reverse after it "wraps around" smoothieware's maximum axis value.

wolfmanjm, 2017-06-17:
using grids of 15 or more probably runs out of memory which is why it cannot be saved, there is nothing we can do about that I am afraid.

callericaz, 2017-06-18:
That makes sense, although I did successfully use a delta grid size of 
29 until the power was turned off.
Could you add that to the documentation and save a bunch of other smart 
people from the mild frustration I dealt with regarding this issue?

Thank you very much for your response.

callericaz, 2017-06-18:
By the way, I used your design, I built a Wolfstock Delta!

wolfmanjm, 2017-06-18:
Different setups will run out of memory at different times so it is not possible to say how big a grid you can use for any specific setup/config. 
One way is to issue the mem command and see how much free memory is in use in the AHB0 section then make sure you do not entirely use that up with the grid which will be n*n*4 bytes in size.

wolfmanjm, 2017-06-18:
It may not be the memory either, there is an issue with writing big files in smoothie to sdcard, and the grid is quite big, there is a potential issue with the write getting corrupted. I think that is more likely what you are running into.
I can do a work around for that though.

callericaz, 2017-06-18:
With the size 15 grid loaded, I have 10112 bytes free in AHB0,
Math seems to suggest that I'm well within the memory limit for my crazy 
use case.

Again, I was able to run the printer with a newly probed Size 29 grid, 
but the save file always seems to be corrupted and unloadable with 
anything over 15.
Is the grid supposed to save in ASCII text format?

wolfmanjm, 2017-06-18:
no the grid is saved in binary

callericaz, 2017-06-18:
Cool,
Let me know when more information is required.
Thank you again for taking the time to go through this for me!

callericaz, 2017-06-20:
Got it,
Let me know if any more information is needed.

wolfmanjm, 2017-06-20:
@callericaz can you try the pull request  #1200 and see if it fixes your issue.
Thanks

callericaz, 2017-06-22:
Hack works great!
I was able to successfully probe, save, and load a Size 33 grid.
MEM command shows 6656 bytes free in AHB0 with grid loaded.