Smoothieware/Smoothieware #1092: Temperature-controlled switches are turned off on HALT

Fulg, 2017-01-11:
#### What were you doing?

- Set up a hotend fan to be temperature-controlled:

```
    switch.misc.enable                           true
    switch.misc.input_on_command                 M42
    switch.misc.input_off_command                M43
    switch.misc.output_pin                       1.19

    temperatureswitch.hotend.enable              true
    temperatureswitch.hotend.designator          T
    temperatureswitch.hotend.switch              misc
    temperatureswitch.hotend.threshold_temp      50.0
    temperatureswitch.hotend.heatup_poll         5
    temperatureswitch.hotend.cooldown_poll       15
```

- Heat up the hotend past the trigger temperature.
- Fan turns on automatically as expected.
- Click Emergency Stop in Simplify3D (or issue M112 manually) to trigger a HALT condition.

#### What did you expect to happen?

Temperature-controlled switches should continue to be monitored and stay active while the temperature is above the trigger point, even after a HALT condition.

#### What happened instead?

The temperature-controlled fan turned off! I had to enable it manually with M42 to prevent my hotend from overheating and cause a jam.

#### Branch & Commit or Version of Smoothieware

```
SENT: version
READ: Build version: edge-6ce309b, Build date: Jan  2 2017 23:50:04, MCU: LPC1768, System Clock: 100MHz
READ: 5 axis
```

#### Hardware model you are using (Make, and version)

Re-ARM v1 (from Panucatt)

#### Host and Operating System running the Host

Simplify3D on Windows x64

#### Link to and console logs

http://pastebin.com/SFyMcwJA

#### Link to running config and M503 results

http://pastebin.com/JipAe8KW

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)...

Zaaphod, 2017-01-11:
Did you try setting switch.misc.ignore_on_halt to true?

Zaaphod, 2017-01-11:
I'm not sure if temperature monitoring would still happen during a halt condition, I'm not that familiar with temperature monitoring, but you could set switch.misc.fail_safe_set_to  to 1 then during halt it would at least turn the fan on which is better than turning it off. 

http://smoothieware.org/switch

Fulg, 2017-01-11:
Well that was easy. Triggering a HALT with `switch.misc.ignore_on_halt` set to `true` fixes the problem, and the fan is even turned off during HALT once the temperature drops below the trigger (like I initially expected).

It could be argued that this should be the default condition for temperature-controlled switches...

Zaaphod, 2017-01-11:
Glad it worked.. the ignore_on_halt option should at least be shown in examples of temperature controlled switch configuration.

Fulg, 2017-01-11:
Is it worth keeping the issue opened for this? http://smoothieware.org/switch says that `ignore_on_halt` prevents the **fail safe value** from being set on HALT, but the *real* source of my issue is that the fail safe value should be the **triggered** state of the temperatureswitch, not just "off".

Zaaphod, 2017-01-11:
In my opinion it's working the way it should.
I don't think there is any way for fail safe value to be previously triggered state, because then it's not really a failsafe value anymore. If you need it to remain the same then you disable loading the fail safe value with ignore_on_halt. I think what you really want is for temperature control to continue during halt, so ignore_on_halt is the best way to accomplish that. 

If there were some way to load the current triggered state as during halt, if the fan was on it would be forced to stay on during halt, and if it was off it would be forced to stay off during halt, regardless of what the temperature section was doing. If you don't want halt to affect the switch, then ignore_on_halt takes care of that for you.

Fulg, 2017-01-11:
It's just a bit weird that this is needs to be set explicitly, I think as a safety precaution any temperatureswitch should implicitly inherit `ignore_on_halt` by default.

The only reason I saw this issue (and subsequently fixed my config) was because I triggered a HALT explicitly; if this had happened during printing I would not have seen it.

Fulg, 2017-01-11:
BTW I am only realizing now that I forgot to thank you for your answers. I certainly don't want to seem ungrateful for the support :)

Zaaphod, 2017-01-11:
I agree, but the way it works is you set the swtich module, then the temperature control module calls that particular switch with temperatureswitch.hotend.switch the ignore_on_halt is part of the switch module not the temperature control module, so I don't think the temperature module can't set the default for the switch module.  Since switches are also used for other things like spindle on/off,  where you would want it to default to shutting the spindle off during halt, it's difficult to determine what the default should be.     I think at least better examples or maybe including the ignore_on_halt option should be in the sample config.

Zaaphod, 2017-01-11:
You're welcome :)  I'm glad to be able to help

closed 2017-01-11 by Fulg