Smoothieware/Smoothieware #1096: Smoothie problem with streaming raster gcode over USB

DarklyLabs, 2017-01-20:
When streaming raster gcode to smoothie over USB, the motion is not smooth. 

Any feed-rate over 30-40mm/sec for raster type gcode results in jerky and stop-start motion from smoothie.

A simple test to replicate this is the following:
1: Create the gcode for a raster image using any method of your choice (LW3, rasterizer, etc).
Set the feedrate for the raster at a low level for most machines (30-40+mm/sec).
2: Run the gcode program on the smoothie over USB using any streamer of your choice (LW3, Pronterface, etc...)
3: You will see the machine judder and stall, struggling to create a smooth movement.

wolfmanjm, 2017-01-20:
We need a lot more info as this could be caused by a lot of things.
1. does this same thing happen with LW4 that uses a better rasterizer?
2. does this happen if playing from sdcard?
3. does this happen if the g code is sent as 64 byte blocks as was discussed in a different thread?

It really is not clear that this is an issue that is simply that the gcode is being consumed faster than it can be sent in which case there is no cure except send it faster, and there have been several recommendations for fixing that in the Host.

I have a test that interleaves the reception of packets over USB with the planning, this should help a bit, in conjunction with sending gcode as at least 64 byte blocks. I'll post a PR for that for testing, but it won;t help much if gcodes are sent one at a time as USB serial simply is not efficient for sending small buffers, plus each ok that comes back has to be sent as a 64 byte buffer.
@arthurwolf has reported that they solved this issue by fixing how the raster is sent, and that that was implemented in LW4 so that needs to be explored too.

wolfmanjm, 2017-01-20:
Try PR #1097 See if that helps, this needs testing.

wolfmanjm, 2017-01-20:
FYI my tests showed that we can process about 1,000 gcodes/sec (on average) and if you were running at 30mm/sec and the pixel was 0.1mm that would take about 3ms per pixel to consume, so the planner in this case could keep the buffer full, so if it was stalling it means the host was not able to send a line (or gcode) at a rate of one every 2-3ms (it needs to be faster than it is consumed). We determined that USB serial is capable of around 10,000 gcodes/sec so it isn't the communication bandwidth (although one has to take into account that USB may not be full duplex and a small buffer containing the ok needs to be sent back, plus the overhead of USB handshaking and protocol... but I guess at worst that would be half the bandwidth). Given those estimates I think this may put the ball back in the hosts court. Can someone confirm how many gcodes the host can actually send? are my estimates of 0.1mm pixels correct? Have I made an order of magnitude error somewhere?

wolfmanjm, 2017-01-21:
Also FWIW I did a raster streaming with bCNC and had no pauses at all and it was running at around 50-100mm/sec on my custom cnc/laser rig.
I was running on a mediocre speed Linux box.
Are you running on Windows? How fast is the PC?

arthurwolf, 2017-01-21:
I think it's very important, to move forward, that we figure out a
*precise* way of reproducing the issue. This is a standard and necessary
part of the resolution of any firmware issue.

What I mean is no "using any method of your choice"-style stuff.

But rather :
* A specific G-code file you know causes the issue. Ideally does on several
different machines that would be even better.
* A specific firmware version ( ideally that being the very latest edge )
* A specific configuration file ( so if we can't reproduce the issue we can
compare it to our config for clues )

This will help reproduce and pinpoint the issue.

Thanks !

arthurwolf, 2017-01-21:
@DarklyLabs Out of topic : By the way, you did not follow the rules for posting an issue : https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md
Generally, @wolfmanjm just closes automatically any issue that doesn't conform. I'm glad he did not in this case, I know how frustrating this issue has been for you, and you probably would have gone on a killing rampage if he had done that.
What I'm trying to say is : in the future make sure you follow the issues rules, they actually help resolve your issue faster.

arthurwolf, 2017-01-21:
@DarklyLabs : Don't ignore the questions and things to try @wolfmanjm mentionned, they are much more important than anything I could babble, but : can you try the exact same thing you are doing now, but on a different OS ( linux ideally ) ? Depending on results we could learn something about whether this has anything to do with USB drivers.

DarklyLabs, 2017-01-21:
Here is some more information as requested:

**does this same thing happen with LW4 that uses a better rasterizer?**
Yes. We have tested many iterations of rasterizer gcode generators. All the way from the original LW3 generator through to Sebastian's latest and greatly optimised version.

**does this happen if playing from sdcard?**
SDCard playback is much better. We had originally considered this a viable workaround for the USB issues but unfortunately the writing to the SD card is unusable because of its extremely slow speed.

**does this happen if the g code is sent as 64 byte blocks as was discussed in a different thread?**
This has a small improvement. Our tests showed that it allowed us to increase the feedrate by approximately 5-10mm/sec. We achieved smooth movement at 25mm/sec as opposed to 20mm/sec. It was also heavily dependant on the size and composition of the image.

**What is the pixel size?**
We are using 0.2mm at the moment because it is the maximum size we can use to achieve a reasonable engraving. 0.1mm is what we really need to achieve.

**what is the acceleration?**
We have tested various acceleration values ranging from 5000 down to 500. The slower accelerations produce smoother movement.

**Did you set queue_delay_time_ms to 0 and test it?**
Yes. When we have experimented with this value in the past it has caused our board to not boot correctly.

wolfmanjm, 2017-01-21:
setting queue_delay_time_ms to 0 would not affect boot at all so I suspect you have a different problem there. Try removing all lines in your config that are not used or obsolete, as large config files can stop smoothie booting.

DarklyLabs, 2017-01-21:
We do not run Linux as it is not what our customers use.

We have tested for a number of months on both OSX and Windows.

OSX: 10.12.2
Windows: 7 / 10

Hardware: Intel I7, I5 and Core2Duo (various different laptops and workstations)

DarklyLabs, 2017-01-21:
@wolfmanjm 

1: Can you post the gcode you had running at 50-100mm/sec. I will run it here and let you know the results.

2: Can you recommend a preferred method of streaming on either win or osx that we can use to help standardise our results for this problem resolution?

3: Would you like us to post a generic image as well as the raster gcode? If so, can you specify the rasterizer you would like us to use and the feed-rate / pixel size you would like?

wolfmanjm, 2017-01-21:
Ok so the fact it works better from sdcard shows it is NOT a smoothie issue I am afraid, it is a host issue not being able to send fast enough. This is because the planning would be the same from sdcard so the planning time (which would be over 1,000 pixles/sec) is not the limiting factor.

I did some special timing tests on the communications only just now and was surprised at how slow the host can send over the USB it is an order of magnitude slower than the theoretical speed. So I suspect this has something to do with the USB drivers on the hosts.
My results showed I can send 2474 lines/sec (these are short 5 character lines) from a simple python streamer that does simple ping pong. This is a lot less than I would expect, but still exceeds the 1,000 pixels/sec.
I was able to nearly double that by using a special streaming technique that relies on the inherent flow control built in to the USB CDC protocol. However I am not sure LW can use it as it requires genuine multi threading to read and send in separate threads, I wrote the streamer in c++ which has genuine threads. This may work in python which does have a global lock though, so I'll try it and see if it still gets the same fast results.
The basic idea is to send as fast as it can and read the replies in a separate thread, basically counting  and consuming the ok replies.

So basically this looks a lot like the host is at fault here and there may simply not be a good workaround unless the host can send faster. I will continue to investigate and test different ideas.

DarklyLabs, 2017-01-21:
@wolfmanjm I will refrain from comparing the results to GRBL. I have used it because it is the closest direct comparison to our hardware and performance achievable. It shows that what we are trying achieve is not out of the realm of possibility.

wolfmanjm, 2017-01-21:
You need to test my PR #1097 see if that helps some. That is the only real fix I can think of in smoothie.
If you are using the latest LW3 which for smoothie will pack multiple gcodes into lines it should help some.

wolfmanjm, 2017-01-21:
As I said comparison to GRBL is not helpful at all it is a totally different driver on the host and simply is not something that smoothie can use.
As my current tests show this issue is in the USB CDC/ACM performance on the host and there are limited things I can do in smoothie to mitigate that.
As you use windows you will will not be able to use my new streamer as that is written in C++ and would only be compilable on linux or mac, and I do not use mac. Which is a shame as I think it may prove once and for all it is not a smoothie issue. However if I cobble a python version and it has similar performance then you may be able to use that to stream on windows. I'm not very good at python and I hate it with a vengeance, but I'll give it a go just for you ;)

DarklyLabs, 2017-01-21:
@wolfmanjm When you designed the smoothie you would have surely tested it under the normal conditions and hardware that customers would be using it with?
A piece of technology that doesn't work with standard hardware is not very useful. How do we now proceed with pushing the problem back to the host?

wolfmanjm, 2017-01-21:
This really is not an issue for any other people using smoothie. It seems to be uniquely yours and for others using similar LW3 and raster, most of our users are using smoothie for 3d printing so this never came up before. As I said this really is starting to look like an issue with the streamer on the host and not a smoothie issue at all.

DarklyLabs, 2017-01-21:
@wolfmanjm Thanks for the help. we will look at your suggestions and let you know if they make any difference.

DarklyLabs, 2017-01-21:
Is there a recommended driver that may improve things for us?

wolfmanjm, 2017-01-21:
the driver is what is built into windows unfortunately we have no control there.
I am experimenting though with my new laptop, it seems a USB3 port does not work at all well, also if you are plugged into a USB1 hub it seems to reduce the comms to around 1000 lines/sec
so make sure you have a USB2 socket or hub and not plugged into a USB3, I was able to get it to work as expected with a USB2 hub plugged into a USB3 socket. This was on linux though.

wolfmanjm, 2017-01-21:
Also it appears that if the host is accessing or scanning the sdcard it drops to 100lines/sec so make sure the sdcard is not mounted.

DarklyLabs, 2017-01-22:
Will test this now.
If the SDcard is unmounted, does than mean users would have to remove the card and place it into a computer to upgrade their firmware?

wolfmanjm, 2017-01-22:
no you can unmount it (actually try to disable auto mount), then just manually mount it when needed. But even when it is unmounted I do see occasional attempts by the host to scan it so not sure if that is happening or not. That is why we have a nomsd version of the firmware. But in that case you would need to remove the sdcard to write to it.

DarklyLabs, 2017-01-22:
**Test Results:**
Firmware = Build Date: Thursday 19 January 2017 17:05 UTC
OS = OSX 10.12.2
Processor: 1.7GHz Intel I7
SDCard Unmounted
Image: J-Lo.jpg (see attached)

Test 1:
Raster GCode generated with latest LW3 master.
File: Test1.gcode
Streamed through LW3.
Image Size: 160DPI, 208mm x 143.83mm
Pixel Size: 0.2mm
Max smooth feedrate: 45-50mm/sec

Test 2:
Rater GCode generated with latest optimised rasterizer-to-gcode.js
File: Test2.gcode
Streamed through LW3.
Image Size: 160DPI, 208mm x 143.83mm
Pixel Size: 0.2mm
Max smooth feedrate: 60-65mm/sec

Test 3:
Rater GCode generated with latest optimised rasterizer-to-gcode.js
File: Test3.gcode
Streamed through LW3.
Image Size: 80DPI, 416mm x 288mm
Pixel Size: 0.2mm
Max smooth feedrate: 15-20mm/sec

Notes:
1: The exact same settings on a larger version of the same image caused a significant reduction in max feedrate
2: Mouse movements on computer cause stuttering in feedrate.

[image]
[Test1.gcode.zip](https://github.com/Smoothieware/Smoothieware/files/721635/Test1.gcode.zip)
[Test2.gcode.zip](https://github.com/Smoothieware/Smoothieware/files/721634/Test2.gcode.zip)
[Test3.gcode.zip](https://github.com/Smoothieware/Smoothieware/files/721633/Test3.gcode.zip)

DarklyLabs, 2017-01-22:
@wolfmanjm What is the process for us to test 'PR #1097 '?

wolfmanjm, 2017-01-22:
very odd that image size can affect the results, as smoothie has no indication of image size.

I did find that my serial throughput tests on windows does indeed show that the windows drivers are crap. the throughput is abysmal.. around 900 lines/sec so the limitation is indeed windows, and to be honest there is very little we can do about it, I suggest you find a Linux box and try your tests to confirm, I know your end users use windows and that is a shame but at least we can confirm where the problem lies. FWIW my windows is win7 x64, on a gen3 I7 core CPU.

This also explains why no one else is reporting the problems as I think most people are using linux.

I'll try to write this improved python streamer and see if it helps on windows as I really do not want to have to learn how to compile c++ on windows (again I did it 15 years ago). If the improved python streamer works at least you will have a workaround. It is possible it could even be launched from LW3, not sure about that.

wolfmanjm, 2017-01-22:
can you also try to stream using bCNC? I never used the streaming on LW3 and do not plan to either. I'd be interested to see if bCNC is any better.
I suspect that as my simple python streamer has bad performance that LW3 streamer is probably even worse given the javscript stuff is probably less stable than the python.
In grbl there is a simple-stream.py script you could also try to stream using that and again see if there is any improvement. I'd really like to focus on where the problem lies.

DarklyLabs, 2017-01-22:
Thanks for the attention to this and help.
The tests above were OSX, not Windows, but our customers will be using one or the other mostly.

I will try bCNC and the grbl simple stream and get back to you.

wolfmanjm, 2017-01-22:
To test the PR you need to compile smoothie yourself. and use git to merge the PR in to test. That is a process that is too involved for me to explain here, you will need to go on github where it tells you how to do that.

wolfmanjm, 2017-01-22:
I am wondering if as the windows (and presumably mac) cdc drivers are so bad, that maybe you could buy an ftdi and hook it to the smoothie uart port and try that? it certainly couldn't be any slower ;) and that would match pretty much what you would see when connecting to grbl. The arduino has an ftdi chip hooked up to the 386 serial uart. I wouldn't be surprised if the FTDI drivers were much better on windows.

DarklyLabs, 2017-01-22:
Testing with bCNC. Running into some trouble getting gcode to actually start the machine moving. All connected and homed etc. Job stated and time counting but no progress or movement. We will spend some more time on this and let you know how the results turn out.

wolfmanjm, 2017-01-22:
Did you select smoothie as the "flavor" in bCNC? also you need formware-cnc bin or set the grbl_mode true in config. bCNC requoires grbl_mode and to be told it is talking to smoothie not grbl.

wolfmanjm, 2017-01-22:
Ok try streaming using this script..

https://github.com/Smoothieware/Smoothieware/blob/edge/fast-stream.py

usage is...

`> python fast-stream.py time-test.g /dev/ttyACM0 -q`

`time-test.g` would be the name of the gcode file you want to send.

FWIW on my linux system it streams twice as fast as the simple streamer.
Haven't tested it on windows yet.

DarklyLabs, 2017-01-22:
Just ran it on OSX.
After running the script, the response was 'Streaming file etc'. The motors made a small move and then sat there.
I am using /dev/cu.usbmodem14211 as this is where the smoothie connects.
Tested with and without grbl_enable true/false.

wolfmanjm, 2017-01-22:
can you leave off the -q and see what it prijnts out? that will be the verbose mode and we can see where it is stopping
I suspect the mac does not allow full duplex serial read/write
so it deadlocks... that is just a guess

DarklyLabs, 2017-01-22:
ok, here is a dump of the returned info. Just stops at SND 89

Lab$ python fast-stream.py JLO_LW3_A.gcode /dev/cu.usbmodem14211
Streaming JLO_LW3_A.gcode to /dev/cu.usbmodem14211
SND 1:  - 0
SND 2: G0 F6000 - 0
SND 3: G0 Y143.827 - 0
SND 4: G0 X0. S0 - 0
SND 5: G1 X0.2 S0.85 F3000 - 0
SND 6: G1 X0.4 S0.85 - 0
SND 7: G1 X0.6 S0.84 - 0
SND 8: G1 X0.8 S0.85 - 0
SND 9: G1 X1. S0.86 - 0
SND 10: G1 X1.2 S0.85 - 0
SND 11: G1 X1.4 S0.85 - 0
SND 12: G1 X1.6 S0.86 - 0
SND 13: G1 X1.8 S0.84 - 0
SND 14: G1 X2. S0.85 - 0
SND 15: G1 X2.2 S0.85 - 0
SND 16: G1 X2.4 S0.85 - 0
SND 17: G1 X2.6 S0.85 - 0
SND 18: G1 X2.8 S0.86 - 0
SND 19: G1 X3. S0.85 - 0
SND 20: G1 X3.2 S0.85 - 0
SND 21: G1 X3.4 S0.86 - 0
SND 22: G1 X3.6 S0.84 - 0
SND 23: G1 X3.8 S0.85 - 0
SND 24: G1 X4. S0.83 - 0
SND 25: G1 X4.2 S0.85 - 0
SND 26: G1 X4.4 S0.84 - 0
SND 27: G1 X4.6 S0.84 - 0
SND 28: G1 X4.8 S0.85 - 0
SND 29: G1 X5. S0.84 - 4
SND 30: G1 X5.2 S0.84 - 4
SND 31: G1 X5.4 S0.85 - 5
SND 32: G1 X5.6 S0.83 - 5
SND 33: G1 X5.8 S0.83 - 5
SND 34: G1 X6. S0.84 - 5
SND 35: G1 X6.2 S0.84 - 5
SND 36: G1 X6.4 S0.84 - 6
SND 37: G1 X6.6 S0.85 - 6
SND 38: G1 X6.8 S0.85 - 6
SND 39: G1 X7. S0.83 - 6
SND 40: G1 X7.2 S0.83 - 6
SND 41: G1 X7.4 S0.85 - 6
SND 42: G1 X7.6 S0.85 - 6
SND 43: G1 X7.8 S0.84 - 7
SND 44: G1 X8. S0.85 - 7
SND 45: G1 X8.2 S0.85 - 7
SND 46: G1 X8.4 S0.85 - 7
SND 47: G1 X8.6 S0.84 - 7
SND 48: G1 X8.8 S0.84 - 7
SND 49: G1 X9. S0.84 - 7
SND 50: G1 X9.2 S0.84 - 7
SND 51: G1 X9.4 S0.84 - 7
SND 52: G1 X9.6 S0.84 - 7
SND 53: G1 X9.8 S0.82 - 7
SND 54: G1 X10. S0.83 - 7
SND 55: G1 X10.2 S0.84 - 7
SND 56: G1 X10.4 S0.84 - 7
SND 57: G1 X10.6 S0.85 - 7
SND 58: G1 X10.8 S0.85 - 7
SND 59: G1 X11. S0.84 - 7
SND 60: G1 X11.2 S0.84 - 17
SND 61: G1 X11.4 S0.84 - 17
SND 62: G1 X11.6 S0.83 - 18
SND 63: G1 X11.8 S0.84 - 18
SND 64: G1 X12. S0.84 - 18
SND 65: G1 X12.2 S0.84 - 18
SND 66: G1 X12.4 S0.84 - 18
SND 67: G1 X12.6 S0.82 - 18
SND 68: G1 X12.8 S0.82 - 18
SND 69: G1 X13. S0.83 - 18
SND 70: G1 X13.2 S0.84 - 18
SND 71: G1 X13.4 S0.84 - 18
SND 72: G1 X13.6 S0.84 - 18
SND 73: G1 X13.8 S0.84 - 18
SND 74: G1 X14. S0.84 - 19
SND 75: G1 X14.2 S0.83 - 19
SND 76: G1 X14.4 S0.84 - 19
SND 77: G1 X14.6 S0.84 - 19
SND 78: G1 X14.8 S0.83 - 19
SND 79: G1 X15. S0.84 - 19
SND 80: G1 X15.2 S0.84 - 19
SND 81: G1 X15.4 S0.83 - 19
SND 82: G1 X15.6 S0.83 - 19
SND 83: G1 X15.8 S0.83 - 19
SND 84: G1 X16. S0.83 - 19
SND 85: G1 X16.2 S0.84 - 19
SND 86: G1 X16.4 S0.84 - 19
SND 87: G1 X16.6 S0.84 - 19
SND 88: G1 X16.8 S0.83 - 19
SND 89: G1 X17. S0.84 - 19

wolfmanjm, 2017-01-22:
can you post the file  JLO_LW3_A.gcode somewhere? what is on line 90?

DarklyLabs, 2017-01-22:
Sorry, it's the same file as previously uploaded called: Test1.gcode

Line 90 is: 
G1 X15.4 S0.83

wolfmanjm, 2017-01-22:
ok I guess I need to send you my test firmware and have you run my test file that times the pure communication rate, although it looks like mac does not support full duplex threaded operations as it appears to be deadlocking after 90 lines.

DarklyLabs, 2017-01-22:
I will test on Windows and let you know how it goes.
Thx for the help!

wolfmanjm, 2017-01-22:
Not sure this is relevant (probably not) but `G1 X17. S0.84` is illegal gcode

it should be X17.0

someone needs to fix that.

wolfmanjm, 2017-01-22:
I just tried your file here on linux and it streamed perfectly.... not hooked up to a machine so no ideas if it is jerky or not.

DarklyLabs, 2017-01-22:
Thx for the update.
I'll run the same tests on our windows machines and see whether they reveal anything different.

wolfmanjm, 2017-01-22:
are you running on a K40? if so what is the results of M503, I'll set mine up here and see how that file performs.

DarklyLabs, 2017-01-22:
We are running it on our machine, the Emblaser.

Here are the results:
Printer is now online.
>>> M503
SENDING:M503
; No config override
;Steps per unit:
M92 X200.00000 Y200.00000 Z1200.00000
;Acceleration mm/sec^2:
M204 S1500.00000 Z500.00000
;X- Junction Deviation, Z- Z junction deviation, S - Minimum Planner speed mm/sec:
M205 X0.05000 Z-1.00000 S0.00000
;Max cartesian feedrates in mm/sec:
M203 X333.33334 Y333.33334 Z33.33333
;Max actuator feedrates in mm/sec:
M203.1 X333.33334 Y333.33334 Z33.33333
;WCS settings
G54
;Digipot Motor currents:
M907 X1.20000 Y1.20000 Z0.35000 A0.80000
;Home offset (mm):
M206 X0.00 Y0.00 Z0.00

wolfmanjm, 2017-01-22:
ok I ran it at F5000 on my K40 was smooth as silk, of course that was from Linux I don;t have a windows machine setup there I can try at the moment.
F6000 was pretty smooth with a little rattling but more due to I think getting close to K40 max speed.

wolfmanjm, 2017-01-23:
So out of curiosity I wanted to see at what point that image would start to get jerky... Was not able to get it to behave that way at all. (On my linux system).
I set the G1 to F20000 and acceleration to 5000 and using the fast-stream.py it ran very smooth.

Of course it is not getting upto F20000 (333mm/sec) as there is only 6.4mm of travel in the planner queue when it is full, but still as the acceleration is increased it gets faster. The max speed at 5000 acceleration it can get to is about 150mm/sec.

So I have to wonder what is going on in your case.
could it be noise on the USB cable forcing retries in the USB protocol? (try a really short good quality cable with ferrites).
It could be the O/S limiting the timeouts on the communications, at least in windows there are settings you can tweak to eliminate those.
I am really at a loss to explain this, and I cannot duplicate the issue which makes it even harder to debug.

DarklyLabs, 2017-01-23:
It looks like it becomes rough after the very first batch of data has passed through the buffer. So basically immediately.

We thought the same thing about USB cables and have tested various options. All include ferrites. They made no difference.

It would be really good if you or Arthur could try this on a non-linux machine to see whether it is just something at my end.

wolfmanjm, 2017-01-23:
I don't have any non-linux machines, I discarded windows a long time ago :) and I don't think I can run this on my laptop which is dual boot but usually boots into linux ;)

wolfmanjm, 2017-01-23:
Could it be something odd about your laser machines? I am testing on a K40, the steps/mm are not the same but close. Could there be a lot of EMF in your machines? although I find it hard to believe they could be worse than these chinese abominations :)
It really has me baffled.
To make matters worse when I tried running my speed tests again today on my laptop booted in win7, the speed jumped to 1500lines/sec nearly double what O was getting yesterday, and I haven't changed anything!
One thing that came to mind is maybe the mac and windows do not like the composite USB device we present. You could try flashing the firmware-disablemsd.bin on github see if that makes any difference, at this point it is worth trying everything :)

DarklyLabs, 2017-01-23:
It's definitely not EMF. We have been testing our machines and electronics at an EMC lab and they are well within limits. We are also running tests without the laser driver enabled, which basically means its a smoothieboard.

When you mention lines/sec, how does this translate to feedrate? 

I will run a test with the disabled composite usb device.
Thanks for the assistance.

wolfmanjm, 2017-01-23:
lines/sec are basically short 5 character lines, so that would translate to a line of gcode, if there was one gcode per line it would be the same as pixels/sec.
So for 0.2mm pixels at 50mm/sec that is 250 pixels/sec being consumed at 100mm/sec it is 500 pixels/sec being consumed, so feeding any faster than that would keep the buffer full. So really even at last nights speeds of 900lines/sec on my windows it should be able to keep up.
On average the planner calculations seems to be able to do on average 1500 pixels/sec which would keep up with speeds up around 300mm/sec.

wolfmanjm, 2017-01-23:
Ok it was hard to setup ;) but I ran the same test1.gcode from my laptop booted in win7 pro x64, and set G1 F10000 and it ran smoothly no hiccups at all... using the fast-stream as I don't have a regular streamer on that laptop.

DarklyLabs, 2017-01-23:
Could the fact we are running coreXY be causing an issue for us?

wolfmanjm, 2017-01-23:
I don't think it should, corexy adds very little overhead, but I suppose anything could be possible and I have no other explanations right now.
Try setting it to regular cartesian and see what happens, it'll run just diagonally but you would see if it was stuttering or not.

wolfmanjm, 2017-01-23:
try the no msd firmware first, then try the non corexy just to eliminate that as a potential.
also please try the fast-stream on windows.

wolfmanjm, 2017-01-23:
Also if you are using a panel, try disabling that. At this point it s a process of elimination to see what may be at fault.

DarklyLabs, 2017-01-23:
coreXY had no effect. Will try the other ideas tonight and advise.

Here is a copy of our basic config being used.
Does something look incorrect in there?
[config.txt](https://github.com/Smoothieware/Smoothieware/files/723010/config.txt)

wolfmanjm, 2017-01-23:
you have tabs in the config which are not allowed, but I do not think they are causing this issue.
Please replace all tabs with spaces.

wolfmanjm, 2017-01-24:
So just to prove it, here is it doing test1 at a little over 100mm/sec actual speed (Set much higher but due to acceleration etc 100 is the max it can get to with such short segments).
https://flic.kr/p/RgN1xb
Any jerkiness is the video not the actual operation which was smooth.
(My grayscales are not very good on this K40)

[9 more replies]

closed 2017-01-25 by wolfmanjm