Smoothieware/Smoothieware #1098: Overextrusion when printing 20x20cm cube after new motion control updates.

dreamscape74, 2017-01-20:
Points to note :
1) When printing with old firmware before motion control update, there is no issue. I reloaded old config and firmware to confirm, the issue occurs only when using the new config or firmware.
2) At the start of each layer there is an over extrusion when printing 20x20 calibration box
3) Relative extrusion is enabled.
4) Over extrusion happens even when direct G-code is used as well, G92 E0 G1 E-0.9 F2400 G1 E0.8 F2400
5) If M83 is used prior to the code mentioned above there is no over extrude. (relative extrusion is triggered) 
6) But during print it has over extrude on the box at every retraction and immediately after Z-lift.

Attached within is the box gcode and my config file. 

[config.txt](https://github.com/Smoothieware/Smoothieware/files/719468/config.txt)

[20mm-box.txt](https://github.com/Smoothieware/Smoothieware/files/719470/20mm-box.txt)

wolfmanjm, 2017-01-20:
Did you reverse the e motors as required in the upgrade notes? (Did you read the upgrade notes?)
Did you try to re calibrate your extrusion with new edge?
This is NOT reproducible, and works as expected here, so I suspect something on your end but not exactly sure what it could be.
Please try to recalibrate your e steps.

One thing I noticed is your acceleration is WAY to high for an extruder, and this may not have been noticed with the old firmware. Try changing it to 500 instead of 5000

wolfmanjm, 2017-01-20:
FYI...

G28 Y ; Home Y first
G28 X ;Then home X
G28 Z ;then home Z

Are illegal gcodes
It must be Y0 X0 Z0 this will probably die in the future even as it may work now.

wolfmanjm, 2017-01-20:
It is probably NOT a good idea to put this in the gcode... M221 S100 it is NOT designed to be put in gcode and may well cause extrusion artifacts. Please remove them.
M221 is designed to be an override set manually for tunign extrusion, it is NOT a regular Gcode and MUST not be used as such.

wolfmanjm, 2017-01-20:
As a rule it is not recommended to use relative extrusion. there is absolutely no need for it and it can cause issues as it is never reset. I recommend you use absolute extrusion as the norm.

wolfmanjm, 2017-02-08:
closed due to lack of response

closed 2017-02-08 by wolfmanjm