Smoothieware/Smoothieware #1101: Web interface blocks printing

surdu, 2017-01-24:
I know this is a known issue, but I'm very interested in tracking it. Maybe even someone with deeper knowledge of the firmware provide more details so someone might try to fix it ?

#### What were you doing?

While printing, I issue a command using the web interface.

#### What did you expect to happen?

The command execute successfully on the printer and return the command result.

#### What happened instead?

The print stopped and the printer stopped responding.

#### Branch & Commit or Version of Smoothieware

edge

#### Hardware model you are using (Make, and version)

    MKS SBASE v1.3

#### Host and Operating System running the Host

    webif web interface

wolfmanjm, 2017-01-24:
This is very unlikely to get fixed on V1 smoothie sorry.

wolfmanjm, 2017-01-24:
FWIW if you are printing via sdcard it will not crash, it will just stutter while the web page loads.

If you are streaming over pronterface using the telnet port then you cannot use the web page, you cannot use web and telnet at the same time, this is not a bug but a limitation of the non threaded multitasking firmware, and cannot be fixed in V1.

I see no point in keeping this issue open as it cannot be fixed.

closed 2017-01-24 by wolfmanjm

arthurwolf, 2017-01-24:
This is a known limitation of the current firmware, version 2 will solve it, unless v2 never happens because MKS succeeds in destroying the Smoothie project by parasiting it's work and ressources.

qharley, 2017-01-25:
I think part of the problem with 3rd party manufacturers taking over is
that the smoothieboard development cycle is too long. That combined with
very long lead times on hardware forced me to look for an alternative here
in South Africa.

The world wants smoothieware. The only way to generate funds is to make
sure you have the preferred delivery system... and I fear you do not have
that currently. V2 sounded great two years ago already. Tired of waiting

arthurwolf, 2017-01-25:
> 274937959>,

Well Smoothie v2 is a colossal project. I think people who are "tired of
waiting" just don't understand how much we are trying to do ( and will do )
with it.
We do things right, that takes time.

qharley, 2017-01-25:
I did not say I am done waiting... just tired. What is most frustrating for
me is the fact that I can do nothing to help.

Fight the good fight.  We will see it when it is done eventually.