Smoothieware/Smoothieware #1103: Spindle pwm pin pw:period has no effect

xpix, 2017-01-29:
Hello,

i test for my XATC some boards to get some functions stable. During my test with the PWMSpindle i wonder that the change in pwm_period has no effect on pin 1.23 (PWM capable pin).

Here my configuration:
Build version: edge-75df940, Build date: Jan 18 2017 20:24:49, MCU: LPC1768, System Clock: 100MHz
  CNC Build 3 axis

```
switch.spindle.enable                            true             #
switch.spindle.pwm_period                        100000            # <---- default 1000, sets the PWM frequency
switch.spindle.input_on_command                  M3               #
switch.spindle.input_off_command                 M5               #
switch.spindle.output_pin                        1.23             # Here we are using the first big MOSFET
switch.spindle.output_type                       pwm            # pwm output settable with S parameter in the input_on_comand
switch.spindle.max_pwm                           255              # set max pwm for the pin default is 255

```
[image]

You see pic from my oszi at the command "M3 S100" ... (btw, can we put the max rpm/min in config and calculate the pwm value correct? i.e. "M3 S6000" max 12000rpm/min == 50% on pwm pin)

I changed for my DC Motor Controller the PWM Frequency to 100 kHz but the Frequency never changed. I check the code but find nothing:

https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/tools/spindle/PWMSpindleControl.cpp#L84

Please help :)

arthurwolf, 2017-01-29:
> spindle/PWMSpindleControl.cpp#L84

you want to be using hw_pwm for output-type, on a pwm-capable pin, not
"pwm".
See http://smoothieware.org/switch

xpix, 2017-01-29:
Thnaks for ur fast answer ok, i change it:

```
switch.spindle.enable                            true             #
switch.spindle.pwm_period                        100000            # default 1000, sets the PWM frequency
switch.spindle.input_on_command                  M3               #
switch.spindle.input_off_command                 M5               #
switch.spindle.output_pin                        1.23             # Here we are using the first big MOSFET
switch.spindle.output_type                       hwpwm            # pwm output settable with S parameter in the input_on_comand
switch.spindle.max_pwm                           255              # set max pwm for the pin default is 255
```

Same result at M3 S50 (it's just between 0 and 100), Frequency is about 0.050 kHz

[image]

arthurwolf, 2017-01-29:
The value you want to set is not pwm_period but pwm_period_ms ( period in
milliseconds ), and 10000 is a strange value to set there.

Also, next time you file an issue, make sure you strictly follow the rules
: https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md

xpix, 2017-01-29:
Ok, found the problem. You have another parameter:
switch.spindle.pwm_period_ms = 20 # 50Hz
change it to
switch.spindle.pwm_period_ms = 0.2 # 5 kHz
and it works, many thanks for ur help.

closed 2017-01-29 by xpix