Smoothieware/Smoothieware #1104: Board fully operable after M999 with KILL button still pressed

Meizirkki, 2017-01-30:
KILL button does not appear to keep the board dead. M999 wakes the board regardless of KILL button status. As a result, the board wakes before kill button is released. This is probably not a bug in code, just a thing that got overlooked, but the KILL button should always KILL the machine.

#### What were you doing?
    1. Press the KILL button
    2. Still holding down the KILL button, issue M999 from serial console
    3. Still holding down the KILL button, issue G28 (or any other command)

#### What did you expect to happen?
    The board to stay dead.

#### What happened instead?
    KILL button is ignored: machine starts moving.

#### Branch & Commit or Version of Smoothieware
    Bug affects all versions I've tested.
    I'm currently running latest EDGE, firmwarebin/firmware-disablemsd commit 564c791

#### Hardware model you are using (Make, and version)
    Smoothieboard 4XC

#### Host and Operating System running the Host
    Raspberry Pi 3 Linux

#### Link to and console logs
    
#### Link to running config and M503 results
    config http://pastebin.com/DBuU1eJY

#### Reasoning

In it's current state, KILL button can not be used for handling of any external hardware stop signals including emergency stop, where it's critical the machine does not start before the stop signal is removed. Were the KILL button to keep the machine dead as long as the stop signal exists it would be the perfect way of properly handling an emergency stop or any other external hardware stop situation.

```diff
# Example of an expected ESTOP functionality with kill button:
1. Emergency stop pressed.
2. ESTOP relay disconnects 24V power and triggers smoothieboard KILL signal.
3. Smoothieboard informs the host that the kill button was pressed.
4. Host attempts to send M999 and start program while emergency stop still pressed.
+ 5. Smoothieboard stays dead and informs the host that the kill button is still pressed.
6. Emergency stop is released -> ESTOP relay restores 24V power and removes KILL signal.
+ 7. Machine does not operate until M999 is sent and program restarted,
+    thus fulfilling the requirements of EU machinery directive.

# Actual behavior:
1. Emergency stop pressed.
2. ESTOP relay disconnects 24V power and triggers smoothieboard KILL signal.
3. Smoothieboard informs the host that the kill button was pressed.
4. Host attempts to send M999 and start program while emergency stop still pressed.
- 5. Smoothieboard wakes up and starts a program.
6. Emergency stop is released -> ESTOP relay restores 24V power and removes KILL signal.
- 7. Machine starts moving immediately upon releasing the emergency stop,
-    a dangerous behavior specifically prohibited by EU legislation.
```

wolfmanjm, 2017-01-30:
This is normal and is how it is supposed to be. The kill is a momentary switch.
If you want a permanent kil then install an E Stop switch on the main power.

Sorry it is not as you expect.

closed 2017-01-30 by wolfmanjm

arthurwolf, 2017-01-30:
The kill command isn't meant to implement an emergency stop in the sense the EU legislation mean it. It is only a feature that does what it currently does.
We recommend in the documentation that for emergency stop purposes, users wire a real estop button that cuts power to the machine.

Meizirkki, 2017-01-30:
Sorry for filing a bug when I should have filed a [feature request.](https://github.com/Smoothieware/Smoothieware/issues/1105)