Smoothieware/Smoothieware #1105: Feature request: Handling of external stop events

Meizirkki, 2017-01-30:
I've built a 3-axis cartesian robot, with a custom host software sending g-codes to smoothieboard which controls the motion and external hardware. I'm using light curtains and a mushroom E stop button with a redundant safety relay for disabling the machine.

The safety relay cuts off VBB, disabling motors and external hardware, but 5V power is left on to keep the smoothieboard alive and it's connection to host intact. The working area is accessed every other minute when a program routine in finished, which means that the light curtain goes off all the time. Such a stop event is part of the normal operation of the machine.

Naturally, the controlling host computer has to be constantly aware of the state of the hardware it's controlling. At the moment there appears to be no proper way of handling emergency stop or other external stop events with smoothieboard so additional I/O solution is needed that connects to host computer directly, bypassing the smoothieboard altogether.

While I understand this is not a flaw in smoothieboard/smoothieware and none of your concern I kindly ask you to considered this feature be implemented:

1. An input pin is specified for a stop signal (similar to the killbutton)
2. When a stop signal is received, smoothieboard stops all operations and ignores attempts to resume operation until the stop signal is cleared

This would make it possible to use automatically resetting light curtains and other modern safety devices with smoothieboard.

wolfmanjm, 2017-01-30:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2017-01-30:
FWIW this can currently be done with a switch module which turns your power relay off. once turned off it would stay off as all power is removed.

closed 2017-01-30 by wolfmanjm