Smoothieware/Smoothieware #1106: Hard Stall due to large gap in steps when transitioning from a diagonal to a straight line, but not a straight line to a diagonal

Zaaphod, 2017-02-07:
#### What were you doing?

Running programs on CNC Routers,.  These are proven programs that have run for years with my old motion control program.  I am performing a smoothie upgrade to 8 CNC routers.

#### What did you expect to happen?

I expected the programs to all run normally

#### What happened instead?

I encountered a hard stall running some programs, I found this stall was duplicatable and repeatable on all 8 machines.  There was never a problem with the old DOS based cnc controllers.  I narrowed the problem down to a short line of code that demonstrates the issue.

#### Branch & Commit or Version of Smoothieware

Build version: edge-75df940, Build date: Jan 18 2017 20:24:49, MCU: LPC1769, System Clock: 120MHz
  CNC Build 3 axis

#### Hardware model you are using (Make, and version)

 Smoothieboard V1.0a 0x

#### Host and Operating System running the Host

   putty terminal - running on windows 10

#### Link to and console logs
nothing much to show.. just normal responses to gcodes

G00 X32 Y66
ok
G01 X27 Y66.4
ok
G01 X24 
ok
G01 X27
ok
G01 X32 Y66
ok
    
#### Link to running config and M503 results

Config - http://pastebin.com/aCf37j46
M503 - http://pastebin.com/29bAX0HA

I have read the [Wiki](http://smoothieware.org) and Especially the [Trouble Shooting section](http://smoothieware.org/troubleshooting)... 

Here is a test program that demonstrates this behavior:

 G00 X32 Y66
 G01 X27 Y66.4
 G01 X24 
 G01 X27
 G01 X32 Y66

The G01 X24 line will ALWAYS cause a stall when transitioning from the diagonal move to the straight move.   However the reverse move, going from a straight line into a diagonal move works as expected,  The X axis continues to move smoothly while the Y axis accelerates into the diagonal move.  For some reason when moving from the diagonal to a straight line, the X axis is pausing, effectively missing many steps, then instantly re-starting without any acceleration causing the stall,  this gap in steps will always stall, and cannot be prevented.

Here is the logic analysis showing the gap in steps which happens when the Y axis is decelerated at the end of the diagonal move

[image]

Here is the logic analysis showing how there is no gap in X axis steps when the sequence is run in reverse,  when Y must be accelerated to match X and make the diagonal move, everything is perfect.
[image]

Zaaphod, 2017-02-07:
More testing shows that this does not always happen.. if the diagonal angle is sharp enough, it will ramp down to a complete stop, where X and Y stop at exactly the same instant,  then it will immediately ramp up the X axis move

wolfmanjm, 2017-02-07:
How are you feeding the gcode?
you are reversing direction in X and you have extremely small junction deviation setting so it will stop before changing direction that is to be expected. I'd suggest making JD a littl ebigger and see if it an issue.

Again this is not reproducible on a normal system, so I suspect it is due to the extremely odd settings you like to use ;)

wolfmanjm, 2017-02-07:
I am afraid I cannot debug this issue so I suspect this will be up to you to figure out. Sorry about that. I look forward to seeing what you discover.

Zaaphod, 2017-02-07:
I am just feeding them over serial line waiting for ok before each.  The stop for direction reversal is not the problem  it's the transition from the diagonal line to an x only move.   

I changed the junction deviation all the way to 0.1, the results are the same. should I go higher than that?

If I run the program 
G00 X32 Y66
G01 X27 Y66.4
G01 X24

at the point where G01 X24 starts,  My X axis will delay about 50 steps until the y finishes decelerating, then instantly resume X at the exact speed it was going, but since there it was already accelerated and it cannot start at that speed, it just stalls.  

If I run the reverse program:
G00 X24 Y66.4
G01 X27
G01 X32 Y66

The X axis moves smoothly as the y axis is accelerated when the diagonal line is encountered.  This seems to only produce the gap in steps when an axis is decelerated and the other keeps going straight.

wolfmanjm, 2017-02-07:
try running from the sdcard just to eliminate the planner queue getting starved

Zaaphod, 2017-02-07:
I tried the following:

G00 X32 Y66
G01 X27 Y66.4 F100
G01 X24 F200

It is doing the X27 Y66.4 move at 100, but not decelerating the x axis, it just stops at the full F100 before the Y finishes moving.  as soon as the y finishes it's last step, then X takes off at F200 with absolutely no acceleration.

When I try this program:
G00 X24 Y66.4
G01 X27 F100
G01 X32 Y66 F200
it behaves completely different.  It accelerates X to 100 and does the X move at F100 then when it gets to the X32 Y66 F200 move, it accelerates X as it accelerates y to F200

I will try testing everything from SD card as you suggest.

Zaaphod, 2017-02-07:
Playing from SD card produces identical results in both directions

wolfmanjm, 2017-02-07:
aren't you using a modified version of smoothie? or using stock version?

If using stock version them please try the master stock version.

wolfmanjm, 2017-02-07:
so sdcard works ok?

Zaaphod, 2017-02-07:
No, SD card has the same issue and stalls the machine in the same exact way as sending serial commands.

I am using a stock - unmodified smoothieboard wired to external stepper motor drivers.  
where can I find master stock version?

wolfmanjm, 2017-02-07:
in the master branch on github under Firmwares

wolfmanjm, 2017-02-07:
I suspect this is the same issue you originally had, some odd edge condition to do with the slow speeds and steps/mm you use. no one else has reported similar issues so I have no idea what the issue could be, but I'd look at the planner and start debugging to see where it is. I really doubt the stepticker is at fault here. I suggest turning on the dump block queue debug statement in the planner/block and see what entry and exit speeds it comes up with. The way the stepticker works is the steps have to end at the same time.

Zaaphod, 2017-02-07:
I tried firmware.bin here: 
https://github.com/Smoothieware/Smoothieware/blob/archive/old-motion-control/FirmwareBin/firmware.bin

and it does NOT have the issue.  It stops at the end of the diagonal and ramps up the X axis only move.

wolfmanjm, 2017-02-07:
That is totally irrelevant. I asked yo try the master stock version. and then you need to debug from there. I really will not be able to debug this for you sorry.

Zaaphod, 2017-02-07:
I will try master branch now...   Thanks for the help.  I'll try to figure out what's going on.   Just for fun I tried running my machine really fast, and it was able to get up enough momentum to carry it through the missing steps.. it had a loud thunk but it didn't stall, but the steps were still delayed on logic analyzer.

Zaaphod, 2017-02-07:
firmware from https://github.com/Smoothieware/Smoothieware/blob/master/FirmwareBin/firmware-cnc.bin 
is having the same issue.

laubed, 2017-02-07:
Going to try the gcode on my machine later this day. Seems like a strange thing thought.

Is mirroring the action does make a difference?
Daniel Laube | Harzblick 48 | 38122 Braunschweig | +49 176 46669865

wolfmanjm, 2017-02-07:
I suspect it is a rounding error in the integer step stuff, and probably only happens with extreme edge conditions, maybe due to very low acceleration and speeds. Or maybe an integer resolution issue in the stepticker code after all.

laubed, 2017-02-07:
> ![decel test
> ![accel test

I don't see any max acceleration settings for your axis in the config file.
Vould you add them and see how it performs then?

Daniel Laube | Harzblick 48 | 38122 Braunschweig | +49 176 46669865

Zaaphod, 2017-02-07:
I have more information.  

This program produces the gap shown in my top logic analyzer screen shot
G00 X32 Y66
G01 X27 Y66.4
G01 X24

however this program does NOT have the gap, and it runs perfectly.
G00 X32 Y66
G01 X27 Y66.7
G01 X24

I am still on master branch https://github.com/Smoothieware/Smoothieware/blob/master/FirmwareBin/firmware-cnc.bin

it does seem likely that it is something that shows up more with my low steps/mm and slow accelerations.

 I don't see max acceleration in the sample config, how do I add those?

laubed, 2017-02-07:
That with max acceleration was rubbish, there is no config for that because all accelerations are being done at acceleration set in config, my fault.

Daniel Laube | Harzblick 48 | 38122 Braunschweig | +49 176 46669865

laubed, 2017-02-07:
Your junction deviations seems way to low either change to 0 (dont know if that is possible) or to 0.01 or something like that.
Daniel Laube | Harzblick 48 | 38122 Braunschweig | +49 176 46669865

wolfmanjm, 2017-02-07:
I think the issue is your steps/mm are so low that many of these moves generate no steps at all, and it is hitting a weird edge condition. can you increase your microstepping on the driver and double or quadruple the steps/mm and see if that fixes the issue?

Zaaphod, 2017-02-07:
all of my machines all have 1/2step only drives on them.    I am getting plenty of steps for each move I want to make.  I do have my logic analyzer hooked up so I can do a test with any number of steps per mm and see if I get the same results... I don't need any stepper drivers hooked up to see the problem.

Here's an update on what I have figured out so far... 

I changed my config steps_per_mm to an even 40 instead of 41.58464566929134 and now I do not have the issue with 
G00 X32 Y66
G01 X27 Y66.4
G01 X24

OR 

G00 X32 Y66
G01 X27 Y66.7
G01 X24

however the problem appears again if I try
G00 X32 Y66
G01 X27 Y66.9
G01 X24

so making it an even number of steps just moved the issue a little.  

It seems this issue is related to step rounding error and/or resolution rounding maybe?  

Looking close at my logic analyzer, when the issue is present, the X step frequency before the gap is almost exact as the frequency after the gap, and I also notice that X steps resume at the same amount of time after the last y step as they would have been if the last x step happened at the same time as the last y step.  So the exit speeds of both X and Y on the diagonal move must be perfectly matched with the entry speed of the following X only move.. and the issue is that the either the X steps finished too soon, or the Y steps finished too late on the diagonal move, the X only move was just continuing as it should have if the previous move had ended both X and Y at the same time.    

Y is stopping over 29ms after X stopped on my capture above, which was enough time that there should have been 48 steps on the X axis.

junction deviation was the first thing wolfmanjm suggested, it's currently at 0.1

Zaaphod, 2017-02-07:
Here's some information:  Still have the config set to 40 steps/mm,  will try increasing this next.

Here's the results from the following program:
G00 X32 Y66
G01 X27 Y66.9
G01 X24
just a small gap
[image]

G00 X32 Y66
G01 X27 Y66.91
G01 X24
a larger gap
[image]

G00 X32 Y66
G01 X27 Y66.92
G01 X24
still a gap
[image]

G00 X32 Y66
G01 X27 Y66.93
G01 X24
no gap at all,  machine ran perfectly
[image]

Zaaphod, 2017-02-07:
Results from changing to 160 steps per mm are exactly the same,  I re-ran the above test and I show everything is perfect with Y=66.9 but gaps where either X finished early or Y finished late on the other 3.  they are not in the same proportions however,   for example Y=66.91 Y took 3 more steps and X was delayed about 20 steps, but with my slower step rate when Y took 3 more steps, X was delayed about 36 steps.  I did not change anything but the steps per mm.

arthurwolf, 2017-02-07:
Hey !

Jim identified another thing you can try.

You can go to
https://github.com/Smoothieware/Smoothieware/blob/edge/src/libs/StepTicker.h
And change the uint32 values to be floats.
And also comment out the 3 fixed point defines at the beginning of the file.
This should switch the step generation from fixed point to floating point,
which could solve this issue. It's slower but that's not a problem in your
setup.

Can you try and report the results here ?
Tell me if you need any kind of help.

Cheers.

wolfmanjm, 2017-02-07:
One thing to test is change the fixed point step stuff to float and try again. this would eliminate or verify it is a fixed point resolution or overflow issue.

wolfmanjm, 2017-02-07:
Actually the changed are needed here...
https://github.com/Smoothieware/Smoothieware/blob/edge/src/modules/robot/Block.h#L54

All the variables that say fixed point change to float.
and then make the
```
#define STEPTICKER_FPSCALE (1<<30)
#define STEPTICKER_TOFP(x) ((int32_t)roundf((float)(x)*STEPTICKER_FPSCALE))
#define STEPTICKER_FROMFP(x) ((float)(x)/STEPTICKER_FPSCALE)
``` 
just return what they get

Zaaphod, 2017-02-07:
Thank you for the advice,  I will try it and report back

Zaaphod, 2017-02-08:
I changed all the fixed point to floats and it improved dramatically, but still would have a small delay sometimes,  Here's the logic analysis of floats.
[image]

So for the fun of it, I changed them all to double and now every time the X and Y stop precisely on the same exact step.  So it would seem this solves the issue.
[image]

However as I'm sure wolfmanjm is just about to point out,  Doubles take way too long to compute, and yes, now with all these doubles, I'm ending up with some kind of issue with my usb serial not being able to always send data and getting into some kind of race condition.   I did manage to run my test program completely once and on each test there was always a step in X precisely at the last y step, and no interruption in the X steps.   Is there a way to define a variable half way between a float and a double?  maybe fixed to 32 bits or something?

maybe I just need a smoothie 2 with a 200MHz processor?

wolfmanjm, 2017-02-08:
Yes I can do 64bit fixed point. but in the meantime I suggest you use the float version.

wolfmanjm, 2017-02-08:
However that is going to significantly reduce the amount of memory available as the block size will increase significantly. this is a pain as most people will not benefit from this at all and I don't see a way to make it optional except for two different builds and using #ifdef

wolfmanjm, 2017-02-08:
Can you try this branch...
https://github.com/Smoothieware/Smoothieware/tree/stepticker/use-64bit-fixed-point
Be careful as it may not work at all ;)
compile with ```make CNC=1``` will help with the extra memory it uses.
--Thanks

arthurwolf, 2017-02-08:
I think having a separate branch, or a compile option, are both valid
solutions here. We definitely don't want to explode the queue's size for
everybody.

Zaaphod, 2017-02-08:
Here's an update.   First of all,  the new 64bit fixed point test works perfectly.  I performed logic analysis on it and the last y step is always perfectly aligned with an x step and X never has any delay.  The pulses look exactly the same as the ones where I used double.  The 64bit fixed point version has no more problems with sending even very large programs to Smoothie.    

Before I tested the 64 bit fixed point, I was trying it with float, and even though float improved the delay, it was still there, and I experienced a few stalls because of it.  

With the 64 bit fixed point version, I have run programs that were having issues several times in a row all the way through without any stalls at all, and no problems streaming through USB Serial.  

I really appreciate the help with this, I was really in a bit of a situation as I am on location at my customer over 1000 miles from home and have just upgraded 8 CNC routers with Smoothie based systems.  

I will let you decide how this is to be implemented,   Personally I do not like any delay in a step chain, especially after a motor has been accelerated, but I do understand that with microstep drives and fast accelerations it's a bit more forgiving.   I would think for CNC applications where machines tend to move more mass and require slower accelerations, and more memory is available, perhaps it would be good to make this the default for CNC=1 and an option for any other built?     

If there is a processing downside I should look for, I would be happy to do side by side comparison between 64 bit fixed and current edge.  I do have programs that run under both.   My opinion is that even if there is a processing slowdown, it's better to have a slowdown than a potential for a stall, but it would be nice to see exactly what the trade off is, if any.

arthurwolf, 2017-02-08:
I think at least in a first time we need to keep this as a compile option.
Could we have the choice between 32 and 64bit fp using a compile option ?
Or maybe we could choose where the comma is in the 32bit fp with a compile
option ? Tuning that would have worked in this case, right ?

wolfmanjm, 2017-02-08:
No 32 bit would not have worked, this seems to be a resolution issue, 2.30 vs 2.62 the latter working.

This could be a compile option but the downside is the max step rate will need to be lowered as this is significantly more processing per step and for the planner. I will have to measure the latency of this now though to see exactly what the speed penalty is. For now I'll maintain the separate branch rather than pollute the code with #ifdef.

wolfmanjm, 2017-03-27:
**UPDATE** This issue is more prevalent than originally suspected. I am not exactly sure what conditions cause it, but I suspect low acceleration coupled with low feed rates. In those cases the 32bit fixed point simply does not have enough resolution. In all cases 64bit fixed point seems to fix it, however this increases the block queue size dramatically to the point where many configurations will simply run out of memory. For now I recommend a CNC builld with 3 axis use the https://github.com/Smoothieware/Smoothieware/tree/stepticker/use-64bit-fixed-point branch.
In the long run we need to figure out how to put his in mainline code without running out of memory.

Zaaphod, 2017-04-04:
Here are the results of the test of firmware-testfp-2.bin

For comparison,  Here is the current edge version, showing a large gap in steps when transitioning from a slight diagonal into a single axis move
[image]

Here is the 64bit Fixed Point version showing no gap in steps, they are nice and uniform throughout the transition
[image]

Here are the results of firmware-testfp-2.bin - Still showing the gap
[image]

Zaaphod, 2017-04-04:
Configuration files and test program are here:
https://pastebin.com/tchvYDDe

Zaaphod, 2017-04-05:
Last Step differnce in rising edge

[image]

wolfmanjm, 2017-04-07:
The 64 bit has been merged into edge, so this issue is now closed

closed 2017-04-07 by wolfmanjm

qharley, 2017-04-08:
Is 64bit now standard, or do you need to do something compile time to
invoke 64bit?