Smoothieware/Smoothieware #1112: Error parsing switch.blah.input_pin_behavior momentary

DM-johnwiedey, 2017-02-08:
In Switch::on_config_reload(), I believe this line to be in error:

`this->input_pin_behavior = THEKERNEL->config->value(switch_checksum, this->name_checksum, input_pin_behavior_checksum )->by_default(momentary_checksum)->as_number();`

The value in the config file is a string ("momentary" or "toggle"), but the default value given is a checksum (uint16_t) and the result is parsed as_number(). This takes things right off the rails as "momentary" gets pared down to the hex value "ea"! Anyway, one possible fix is this:

`std::string input_pin_behavior = THEKERNEL->config->value(switch_checksum, this->name_checksum, input_pin_behavior_checksum )->by_default("momentary")->as_string();`

`this->input_pin_behavior = (input_pin_behavior == "momentary") ? momentary_checksum : toggle_checksum;`

or you could make `this->input_pin_behavior` a `std::string`, but other code in Switch.cpp would have to change as well.

wolfmanjm, 2017-02-08:
Hello
This is an automated message to tell you that your issue does not conform to the rules :
[How to file an issue](https://github.com/Smoothieware/Smoothieware/blob/edge/ISSUE_TEMPLATE.md)

 In order to help you as best as we can, please read those rules and modify your issue accordingly.

The issue has been closed to make sure only valid issues are opened at any one time, but it will be re-opened as soon as you fix it.

Thanks, the Smoothieware team.

wolfmanjm, 2017-02-08:
good catch, will fix some time.

closed 2017-02-08 by wolfmanjm

DM-johnwiedey, 2017-02-08:
Ha - I resubmitted as #1113 (following the rules this time ;-)

PR #1114 (2017-02-08): fix switch momentary/toggle config setting
fixes #1112